riscv/meta-riscv #5: Create a wic file template for SD-Card creation

kraj, 2018-03-20:
This would help in creating a bootable SD-Card images out of image creator. We need to create appropriate wic file.

mirzak, 2018-03-20:
Working on it :)

mirzak, 2018-03-20:
WIC does not support custom fstypes. 

And my understand is for the bootrom to pick the correct location one must set the "boot part" fstype to `2E54B353-1271-4842-806F-E436D6AF6985`.

I will continue to investigate and will ask some questions on appropriate mailing lists.

But it probably will be easier to create a custom image instead and do some parted magic and then we do not really have any limitations.

mirzak, 2018-03-20:
Actually confused `fstype` with `part-type`, it is `part-type` that we want to set and WIC supports this just fine.

kraj, 2018-03-21:
Right I would like to avoid custom sd card image creator scripts they are error prone and difficult to maintain

mirzak, 2018-03-21:
Yeah me to. But I think I will sort it out WIC, or I think I already did here https://github.com/mirzak/meta-riscv-1/commits/wic-image.

The output wic image has the same properties as the image created by the freedom-u-sdk, no way to test it thought.

I am having some issue with circular deps after enabling wic and hence no pull-request yet

kraj, 2018-03-21:
wic should not be enabled for initramfs images only when doing full SD-Card images.

mirzak, 2018-03-21:
Yeah you are right. Here is the error "for the record". 

```
ERROR: 2747 unbuildable tasks were found.##########################################################                                                                                                  | ETA:  0:00:00
These are usually caused by circular dependencies and any circular dependency chains found will be printed below. Increase the debug level to see a list of unbuildable tasks.

Identifying dependency loops (this may take a short while)...

ERROR: 
Dependency loop #1 found:
  Task /home/mirza/project/riscv/meta-riscv/recipes-core/images/riscv-initramfs-image.bb:do_image_complete (dependent Tasks ['riscv-initramfs-image.bb:do_image_cpio', 'riscv-initramfs-image.bb:do_image_tar', 'riscv-initramfs-image.bb:do_image', 'riscv-initramfs-image.bb:do_image_wic'])
  Task /home/mirza/project/riscv/meta-riscv/recipes-kernel/linux/linux-riscv_4.15.bb:do_bundle_initramfs (dependent Tasks ['linux-riscv_4.15.bb:do_install', 'riscv-initramfs-image.bb:do_image_complete'])
  Task /home/mirza/project/riscv/meta-riscv/recipes-kernel/linux/linux-riscv_4.15.bb:do_deploy (dependent Tasks ['linux-riscv_4.15.bb:do_populate_sysroot', 'linux-riscv_4.15.bb:do_packagedata', 'linux-riscv_4.15.bb:do_bundle_initramfs', 'depmodwrapper-cross_1.0.bb:do_populate_sysroot'])
  Task /home/mirza/project/riscv/meta-riscv/recipes-core/images/riscv-initramfs-image.bb:do_image_wic (dependent Tasks ['riscv-initramfs-image.bb:do_image', 'mtools_4.0.18.bb:do_populate_sysroot', 'dosfstools_4.1.bb:do_populate_sysroot', 'parted_3.2.bb:do_populate_sysroot', 'gptfdisk_1.0.3.bb:do_populate_sysroot', 'linux-riscv_4.15.bb:do_deploy', 'riscv-initramfs-image.bb:do_rootfs_wicenv'])

ERROR: Command execution failed: 1
```

Would still be nice if we could have initramfs + wic initially, since you could still boot the initramfs and chroot to the ext4 part. Until we change for it to boot to SD/MMC completely.

kraj, 2018-03-21:
thats possible too, you have to use kernel+initramfs as your kernel file and rest remaining same. In other words we alsway boot into a small initrd and then switch_root to ext4 partition

closed 2018-03-27 by kraj