riscv/meta-riscv #28: Build instructions (README.md) fail to checkout master

jleidel, 2018-08-29:
The build instructions detailed in the top-level README.md file fail to checkout the master repository.  

$> mkdir riscv-yocto
$> repo init -u git://github.com/riscv/meta-riscv  -b master -m tools/manifests/riscv-yocto.xml
Traceback (most recent call last):
  File "/hdd/home/jleidel/dev/working/yocto/.repo/repo/main.py", line 531, in <module>
    _Main(sys.argv[1:])
  File "/hdd/home/jleidel/dev/working/yocto/.repo/repo/main.py", line 507, in _Main
    result = repo._Run(argv) or 0
  File "/hdd/home/jleidel/dev/working/yocto/.repo/repo/main.py", line 180, in _Run
    result = cmd.Execute(copts, cargs)
  File "/hdd/home/jleidel/dev/working/yocto/.repo/repo/subcmds/init.py", line 399, in Execute
    self._SyncManifest(opt)
  File "/hdd/home/jleidel/dev/working/yocto/.repo/repo/subcmds/init.py", line 245, in _SyncManifest
    m.MetaBranchSwitch()
  File "/hdd/home/jleidel/dev/working/yocto/.repo/repo/project.py", line 2913, in MetaBranchSwitch
    self.Sync_LocalHalf(syncbuf)
  File "/hdd/home/jleidel/dev/working/yocto/.repo/repo/project.py", line 1363, in Sync_LocalHalf
    lost = self._revlist(not_rev(revid), HEAD)
  File "/hdd/home/jleidel/dev/working/yocto/.repo/repo/project.py", line 2501, in _revlist
    return self.work_git.rev_list(*a, **kw)
  File "/hdd/home/jleidel/dev/working/yocto/.repo/repo/project.py", line 2700, in rev_list
    (self._project.name, str(args), p.stderr))
error.GitError: manifests rev-list (u'^1d3ade5e42f03b6cce64b6313d37791b69075441', 'HEAD', '--'): fatal: bad revision 'HEAD'

kraj, 2018-08-29:
repo tool needs python2, can you check if you default python install /usr/bin/python is python2 and not python3. if not then open the repo script and change it

```
#!/usr/bin/env python2
```

jleidel, 2018-08-29:
This isn't a python problem.  The scripts worked yesterday morning and they're broken this morning following the recent commits.  I'm running python 2.7

kraj, 2018-08-29:
For tests I just created a fresh workspace and it worked ok. I wonder if you have .repo directory lying around in the tree heirarchy where you are trying to create it.

I have also noticed that the instructions should have said to cd into riscv-yocto folder after creating it.
This has been fixed with https://github.com/riscv/meta-riscv/commit/64b2b1317be909a79ac680af2423ea8af2e5bc36

jleidel, 2018-08-29:
Confirmed the fix is in the README.md in b9351d1e364687791ca871a37e9fffde95d7d912

closed 2018-08-29 by jleidel