riscv/meta-riscv #29: Duplicate with #26. But attach log.

pino-kim, 2018-08-29:
[rootfs_log.txt](https://github.com/riscv/meta-riscv/files/2332140/rootfs_log.txt)

It's duplicate with #26 but I want to share  the error logfile.
I build 5525260 and get same error.

and 

build e64b2ee I meet same log too.

thanks.

kraj, 2018-08-29:
I think the problem is this

```
/home/pino/workspace/yocto_build/riscv-yocto/build/tmp/work/qemuriscv64-poky-linux/core-image-full-cmdline/1.0-r0/recipe-sysroot/usr/bin/crossscripts/qemuwrapper: 9: /home/pino/workspace/yocto_build/riscv-yocto/build/tmp/work/qemuriscv64-poky-linux/core-image-full-cmdline/1.0-r0/recipe-sysroot/usr/bin/crossscripts/qemuwrapper: qemu-riscv64: not found
```

we are probably not building qemu user binary or its called differently

kraj, 2018-08-29:
I have pushed changes to master to use oe-core instead of poky, where riscv64 support is enabled by
default, can you retry with latest master ?

closed 2018-09-06 by pino-kim