riscv/meta-riscv #34: report  build #33

pino-kim, 2018-09-22:
@kraj 

I open new issue about #33 .

start run build with command 
$ bitbake core-image-full-cmdline

I entered below error log

pino@pino-VirtualBox:~/workspace/opensource_work/riscv-yocto/build$ bitbake core-image-full-cmdline
ERROR: Unable to start bitbake server
ERROR: Server log for this session (/home/pino/workspace/opensource_work/riscv-yocto/build/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 4836 at 2018-09-22 23:46:00.420139 ---
ERROR: ParseError at /home/pino/workspace/opensource_work/riscv-yocto/build/conf/auto.conf:12: unparsed line: 'USER_CLASSES ?= "buildstats buildhistory buildstats-summary image-mklibs image-p'

pino-kim, 2018-09-22:
bitbake-cookerdaemon.log say
bash: syntax error near unexpected token `/home/pino/workspace/opensource_work/riscv-yocto/build/bitbake-cookerdaemon.log'

pino-kim, 2018-09-22:
Dear @kraj
I guess ....
I chagne          image-p' -> image-prelink" 
am I right? Anyway build is start now.

kraj, 2018-09-22:
my bad, Fixed now

pino-kim, 2018-09-22:
@kraj 
I encounter same end error.
So I change shell dash (ubuntu's default shell) to bash and build again.

I asked your distro linux yester.
What's your distro linux when build test?

kraj, 2018-09-22:
Ah it might be better to use bash for now although using dash is a supported usecase in OE. but use bash can make you proceed.

My distro is here

https://github.com/YoeDistro/yoe-distro

Go through readme and you should be able to build for risvc64 qemu

pino-kim, 2018-09-23:
Thanks @kraj 
I can build meta-riscv with yoe-distro.
But I have other issue. 

I can't see command line start end of qemu boot.
It's other issue just little bit qemu build option in ubuntu.
Anyway rottfs is made and I can see boot start. 

We can find #26 's solution from this thread and yoe-distro.

kraj, 2018-09-23:
Can you add

```
DISTRO_FEATURES_append = " systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
```

in local.conf and rebuild.

Then launch with
```
runqemu nographic
```

wait a while give it atleast 1-2 minutes when booting for first time.

pino-kim, 2018-09-23:
Ok
I will follow your guide both(yoe-linux and only meta-riscv)
and report here again.

pino-kim, 2018-09-24:
yoe-linux with meta-riscv build
------------------------------------

 
Build Configuration:
BB_VERSION           = "1.39.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "riscv64-yoe-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "yoe"
DISTRO_VERSION       = "2.6"
TUNE_FEATURES        = "riscv64 littleendian"
meta-yoe             = "HEAD:d7d236f9edea9e42cbe70a277ab670a464a37c96"
meta-qcom            = "HEAD:f1ced39e1d89874c78097009df122d45b5c26b0b"
meta-raspberrypi     = "HEAD:9da0723f4c808f8158c7770cb4125f515e5a663c"
meta-96boards        = "HEAD:80c26c1e41c3a963c59d135beffea520a3d7744f"
meta-riscv           = "HEAD:cb2be0259ee329138b17a443d795fa01e7b9d6fc"
meta-browser         = "HEAD:e4a667deaaf9a26a3a1aeb355770d1f29da549ad"
meta-qt5             = "HEAD:66c49f8a1d4ab5ab37a977c4234f58ea8ceb6053"
meta-filesystems     
meta-gnome           
meta-initramfs       
meta-multimedia      
meta-networking      
meta-oe              
meta-perl            
meta-python          
meta-webserver       
meta-xfce            = "HEAD:a06c29be2f5bc204f965ee07298c34232c5c39fc"
meta                 = "HEAD:92c266fd9b1c4034a13bd5b102d1817df388a7b5"

Initialising tasks: 100% |#######################################| Time: 0:00:02
Sstate summary: Wanted 1006 Found 1003 Missed 3 Current 0 (99% match, 0% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Tasks Summary: Attempted 2948 tasks of which 2698 didn't need to be rerun and all succeeded.
NOTE: Writing buildhistory
NOTE: Build completion summary:
NOTE:   do_populate_sysroot: 100.0% sstate reuse(226 setscene, 0 scratch)
NOTE:   do_package_qa: 100.0% sstate reuse(134 setscene, 0 scratch)
NOTE:   do_packagedata: 100.0% sstate reuse(134 setscene, 0 scratch)
NOTE:   do_package_write_ipk: 100.0% sstate reuse(134 setscene, 0 scratch)
NOTE:   do_populate_lic: 100.0% sstate reuse(233 setscene, 0 scratch)

kraj, 2018-09-24:
yay so it seems all works now?

pino-kim, 2018-09-24:
#33  meta-riscv only build
-------------------------------------
Build Configuration:
BB_VERSION           = "1.39.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64 littleendian"
meta                 = "work:e3f5ceec690379d64bd38330111a32f6c8fc6353"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      = "work:a06c29be2f5bc204f965ee07298c34232c5c39fc"
meta-riscv           = "work:cb2be0259ee329138b17a443d795fa01e7b9d6fc"

NOTE: Fetching uninative binary shim from http://downloads.yoctoproject.org/releases/uninative/2.2/x86_64-nativesdk-libc.tar.bz2;sha256sum=e3b77208169bf1ac4e89496f3cdbf27695f5b18a2694a908a793390f28b67f83
Initialising tasks: 100% |#######################################| Time: 0:00:03
Sstate summary: Wanted 1054 Found 0 Missed 1054 Current 0 (0% match, 0% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: core-image-full-cmdline-1.0-r0 do_rootfs: The postinstall intercept hook 'update_font_cache' failed, details in /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/core-image-full-cmdline/1.0-r0/temp/log.do_rootfs
ERROR: core-image-full-cmdline-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/core-image-full-cmdline/1.0-r0/temp/log.do_rootfs.10408
ERROR: Task (/home/pino/workspace/opensource_work/riscv-yocto/openembedded-core/meta/recipes-extended/images/core-image-full-cmdline.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2702 tasks of which 5 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Build completion summary:
NOTE:   do_populate_sysroot: 0.0% sstate reuse(0 setscene, 196 scratch)
NOTE:   do_package_qa: 0.0% sstate reuse(0 setscene, 33 scratch)
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 139 scratch)
NOTE:   do_packagedata: 0.0% sstate reuse(0 setscene, 139 scratch)
NOTE:   do_package_write_ipk: 0.0% sstate reuse(0 setscene, 139 scratch)
NOTE:   do_populate_lic: 0.0% sstate reuse(0 setscene, 247 scratch)

Summary: 1 task failed:
  /home/pino/workspace/opensource_work/riscv-yocto/openembedded-core/meta/recipes-extended/images/core-image-full-cmdline.bb:do_rootfs
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

pino-kim, 2018-09-24:
@kraj
I can runqemu
yoe-distro with meta-riscv build

but we still have issue meta-rscv only build.

I think.... we should compare both rootfs log build

I hang both file here just wailt ...

kraj, 2018-09-24:
can you login using 'root' with no password ?

pino-kim, 2018-09-24:
@kraj
yepp.
yoe with meta-riscv is fine in ubuntu 16.04.

[image]

pino-kim, 2018-09-24:
## only meta-riscv rootfs
1373 NOTE: Exit code 1. Output:
1374 + '[' True = False ']'
1375 + qemu-riscv64 -r 4.15 -E LD_LIBRARY_PATH=/home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-li     nux/core-image-full-cmdline/1.0-r0/rootfs/usr/lib:/home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/qemuriscv     64-oe-linux/core-image-full-cmdline/1.0-r0/rootfs/lib -L /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/qe     muriscv64-oe-linux/core-image-full-cmdline/1.0-r0/rootfs -E /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work     /qemuriscv64-oe-linux/core-image-full-cmdline/1.0-r0/rootfs/usr/libexec/fc-cache --sysroot=/home/pino/workspace/opensource_work/r     iscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/core-image-full-cmdline/1.0-r0/rootfs --system-only
1376 usage: qemu-riscv64 [options] program [arguments...]
1377 Linux CPU emulator (compiled for riscv64 emulation)
1378 
1379 Options and associated environment variables:
1380 
1381 Argument             Env-variable      Description
1382 -h                                     print this help
1383 -help
1384 -g port              QEMU_GDB          wait gdb connection to 'port'
1385 -L path              QEMU_LD_PREFIX    set the elf interpreter prefix to 'path'
1386 -s size              QEMU_STACK_SIZE   set the stack size to 'size' bytes
1387 -cpu model           QEMU_CPU          select CPU (-cpu help for list)
1388 -E var=value         QEMU_SET_ENV      sets targets environment variable (see below)
1389 -U var               QEMU_UNSET_ENV    unsets targets environment variable (see below)
1390 -0 argv0             QEMU_ARGV0        forces target process argv[0] to be 'argv0'
1391 -r uname             QEMU_UNAME        set qemu uname release string to 'uname'
1392 -B address           QEMU_GUEST_BASE   set guest_base address to 'address'
1393 -R size              QEMU_RESERVED_VA  reserve 'size' bytes for guest virtual address space
1394 -d item[,...]        QEMU_LOG          enable logging of specified items (use '-d help' for a list of items)
1395 -dfilter range[,...] QEMU_DFILTER      filter logging based on address range
1396 -D logfile           QEMU_LOG_FILENAME write logs to 'logfile' (default stderr)
1397 -p pagesize          QEMU_PAGESIZE     set the host page size to 'pagesize'
1398 -singlestep          QEMU_SINGLESTEP   run in singlestep mode
1399 -strace              QEMU_STRACE       log system calls
1400 -seed                QEMU_RAND_SEED    Seed for pseudo-random number generator
1401 -trace               QEMU_TRACE        [[enable=]<pattern>][,events=<file>][,file=<file>]
1402 -version             QEMU_VERSION      display version information and exit
1403 
1404 Defaults:
1405 QEMU_LD_PREFIX  = /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/core-image-full-cmdl     ine/1.0-r0/rootfs
1406 QEMU_STACK_SIZE = 8388608 byte
1407 
1408 You can use -E and -U options or the QEMU_SET_ENV and
1409 QEMU_UNSET_ENV environment variables to set and unset
1410 environment variables for the target process.
1411 It is possible to provide several variables by separating them
1412 by commas in getsubopt(3) style. Additionally it is possible to
1413 provide the -E and -U options multiple times.
1414 The following lines are equivalent:
1415     -E var1=val2 -E var2=val2 -U LD_PRELOAD -U LD_DEBUG
1416     -E var1=val2,var2=val2 -U LD_PRELOAD,LD_DEBUG
1417     QEMU_SET_ENV=var1=val2,var2=val2 QEMU_UNSET_ENV=LD_PRELOAD,LD_DEBUG
1418 Note that if you provide several changes to a single variable
1419 the last change will stay in effect.
1420 
1421 See <https://qemu.org/contribute/report-a-bug> for how to report bugs.
1422 More information on the QEMU project at <https://qemu.org>.
1423 
1424 ERROR: The postinstall intercept hook 'update_font_cache' failed, details in /home/pino/workspace/opensource_work/riscv-yocto/bui     ld/tmp-glibc/work/qemuriscv64-oe-linux/core-image-full-cmdline/1.0-r0/temp/log.do_rootfs
1425 DEBUG: Python function do_rootfs finished
1426 ERROR: Function failed: do_rootfs

------------------------------

## yoe with meta-riscv rootfs
1483 NOTE: + '[' True = False ']'
1484 + qemu-riscv64 -r 4.15 -E LD_LIBRARY_PATH=/home/pino/workspace/opensource_work/yoe-distro/build/tmp/work/qemuriscv64-yoe-linux/co     re-image-full-cmdline/1.0-r0/rootfs/usr/lib:/home/pino/workspace/opensource_work/yoe-distro/build/tmp/work/qemuriscv64-yoe-linux/     core-image-full-cmdline/1.0-r0/rootfs/lib -L /home/pino/workspace/opensource_work/yoe-distro/build/tmp/work/qemuriscv64-yoe-linux     /core-image-full-cmdline/1.0-r0/rootfs /home/pino/workspace/opensource_work/yoe-distro/build/tmp/work/qemuriscv64-yoe-linux/core-     image-full-cmdline/1.0-r0/rootfs/usr/libexec/gio-querymodules /home/pino/workspace/opensource_work/yoe-distro/build/tmp/work/qemu     riscv64-yoe-linux/core-image-full-cmdline/1.0-r0/rootfs/usr/lib/gio/modules/
1485 

......

I guess this point little bit difference and occur error.

pino-kim, 2018-09-25:
#35 build success in ubuntu cmd mode.

closed 2018-09-25 by pino-kim