riscv/meta-riscv #57: qemuriscv32 : core-image-full-cmdline build fail

pino-kim, 2018-11-30:
**Description**
Thanks to @alistair23  support qemuriscv32.
I builded it today. I met some error in do_compile ```libgpg-error```
I guess current build fail is occured by has no  ```lock-obj-pub.riscv32-unknown-linux-gnu.h``` in libgpg-error
(Only ```lock-obj-pub.riscv64-unknown-linux-gnu.h```)

**Steps to reproduce the issue:**
1. ```. ./meta-riscv/setup.sh```
2. replace ```MACHINE="qemuriscv64"``` to ```MACHINE="qemuriscv32"```
3. bitbakae core-image-full-cmdline -k

**Describe the results you received:**
> ERROR: libgpg-error-1.32-r0 do_compile: Function failed: do_compile (log file is located at /home/pino/workspace/opensource/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libgpg-error/1.32-r0/temp/log.do_compile.130692)
ERROR: Logfile of failure stored in: /home/pino/workspace/opensource/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libgpg-error/1.32-r0/temp/log.do_compile.130692
Log data follows:
| DEBUG: SITE files ['endian-little', 'bit-32', 'riscv-common', 'common-linux', 'common-glibc', 'riscv32-linux', 'riscv32-linux', 'common']
| DEBUG: Executing shell function do_compile
| cp: cannot stat '/home/pino/workspace/opensource/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libgpg-error/1.32-r0/libgpg-error-1.32/src/syscfg/lock-obj-pub.riscv32-unknown-linux-gnu.h': No such file or directory
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/pino/workspace/opensource/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libgpg-error/1.32-r0/temp/log.do_compile.130692)
ERROR: Task (/home/pino/workspace/opensource/riscv-yocto/openembedded-core/meta/recipes-support/libgpg-error/libgpg-error_1.32.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 970 tasks of which 969 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory

>Summary: 1 task failed:
  /home/pino/workspace/opensource/riscv-yocto/openembedded-core/meta/recipes-support/libgpg-error/libgpg-error_1.32.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

**Describe the results you expected:**
 - build fine, runqemu fine

**Additional information you deem important (e.g. issue happens only occasionally):**
>Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv32-oe-linux"
MACHINE              = "qemuriscv32"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv32 littleendian"
meta                 = "work:21387613fec1a8c142ed48d7a74d587e205b0c98"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      = "work:9a64b3805de82c813d89de92163129a84c2ebf97"
meta-riscv           = "HEAD:51ebe0bf47c95933481c371ae66dde6ad46977f6"

>Initialising tasks: 100% |############################################################| Time: 0:00:01
Sstate summary: Wanted 5 Found 0 Missed 5 Current 188 (0% match, 97% complete)

alistair23, 2018-11-30:
Yeah, I can reproduce this error. It looks like libgpg isn't building:

```
DEBUG: SITE files ['endian-little', 'bit-32', 'riscv-common', 'common-linux', 'common-glibc', 'riscv32-linux', 'riscv32-linux', 'common']
DEBUG: Executing shell function do_compile
cp: cannot stat '/scratch/alistair/yocto/oe-master/build/tmp-glibc/work/riscv32-oe-linux/libgpg-error/1.32-r0/libgpg-error-1.32/src/syscfg/lock-obj-pub.riscv32-unknown-linux-gnu.h': No such file or directory
WARNING: /scratch/alistair/yocto/oe-master/build/tmp-glibc/work/riscv32-oe-linux/libgpg-error/1.32-r0/temp/run.do_compile.22455:1 exit 1 from 'cp /scratch/alistair/yocto/oe-master/build/tmp-glibc/work/riscv32-oe-linux/libgpg-error/1.32-r0/libgpg-error-1.32/src/syscfg/lock-obj-pub.$TUPLE.h /scratch/alistair/yocto/oe-master/build/tmp-glibc/work/riscv32-oe-linux/libgpg-error/1.32-r0/libgpg-error-1.32/src/syscfg/lock-obj-pub.$TARGET_FILE.h'
ERROR: Function failed: do_compile (log file is located at /scratch/alistair/yocto/oe-master/build/tmp-glibc/work/riscv32-oe-linux/libgpg-error/1.32-r0/temp/log.do_compile.22455)

```

pino-kim, 2018-11-30:
Oh good day?
I eagg to build libgpg.
I eager to get  ```lock-obj-pub.riscv32-unknown-linux-gnu.h``` file in the qemu risv32 env.
Maybe ... We can get ```lock-obj-pub.riscv32-unknown-linux-gnu.h``` run ```gen-posix-lock-obj```
in the qemuriscv32 env.

alistair23, 2018-11-30:
I haven't been able to get to a login prompt on 32-bit RISC-V yet, so I can't run it on the guest.

Manually editing the file gave me this, which at least compiles: https://github.com/alistair23/meta-riscv/tree/alistair/libgpg

alistair23, 2018-11-30:
Do you mind testing that branch and let me know if it works for you?

pino-kim, 2018-12-01:
@alistair23 Oh test your branch ?
 https://github.com/alistair23/meta-riscv/tree/alistair/libgpg 
sure why not

pino-kim, 2018-12-01:
```Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv32-oe-linux"
MACHINE              = "qemuriscv32"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv32 littleendian"
meta                 = "work:21387613fec1a8c142ed48d7a74d587e205b0c98"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      = "work:ebc7b9e20ac22f6f2ad373621917f53e8a9af81c"
meta-riscv           = "alistair/libgpg:19c8c3531b69f07c32ea540b84441b2088f81857"

Initialising tasks: 100% |######################################################################################################################| Time: 0:00:02
Sstate summary: Wanted 2 Found 0 Missed 2 Current 1094 (0% match, 99% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: core-image-full-cmdline-1.0-r0 do_rootfs: The postinstall intercept hook 'update_udev_hwdb' failed, details in /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/temp/log.do_rootfs
ERROR: core-image-full-cmdline-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/temp/log.do_rootfs.92002
ERROR: Task (/home/pino/workspace/opensource_work/riscv-yocto/openembedded-core/meta/recipes-extended/images/core-image-full-cmdline.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3218 tasks of which 3217 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory

pino-kim, 2018-12-01:
```Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "riscv32-oe-linux"
MACHINE              = "qemuriscv32"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv32 littleendian"
meta                 = "work:21387613fec1a8c142ed48d7a74d587e205b0c98"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      = "work:ebc7b9e20ac22f6f2ad373621917f53e8a9af81c"
meta-riscv           = "alistair/libgpg:19c8c3531b69f07c32ea540b84441b2088f81857"

Initialising tasks: 100% |######################################################################################################################################################################################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 1096 Found 1093 Missed 3 Current 0 (99% match, 0% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: core-image-full-cmdline-1.0-r0 do_rootfs: The postinstall intercept hook 'update_udev_hwdb' failed, details in /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/temp/log.do_rootfs
ERROR: core-image-full-cmdline-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/temp/log.do_rootfs.117583
ERROR: Task (/home/pino/workspace/opensource_work/riscv-yocto/openembedded-core/meta/recipes-extended/images/core-image-full-cmdline.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3218 tasks of which 2953 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Build completion summary:
NOTE:   do_populate_sysroot: 100.0% sstate reuse(249 setscene, 0 scratch)
NOTE:   do_package_qa: 100.0% sstate reuse(145 setscene, 0 scratch)
NOTE:   do_packagedata: 100.0% sstate reuse(145 setscene, 0 scratch)
NOTE:   do_package_write_ipk: 100.0% sstate reuse(145 setscene, 0 scratch)
NOTE:   do_populate_lic: 100.0% sstate reuse(256 setscene, 0 scratch)

Summary: 1 task failed:
  /home/pino/workspace/opensource_work/riscv-yocto/openembedded-core/meta/recipes-extended/images/core-image-full-cmdline.bb:do_rootfs
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

pino-kim, 2018-12-01:
[build_error_20181201.log](https://github.com/riscv/meta-riscv/files/2635845/build_error_20181201.log)

alistair23, 2018-12-01:
Hmm... This looks like a separate problem. So hopefully that branch does fix the libgpg-error problem. Either way it looks like 32-bit support needs more work

pino-kim, 2018-12-02:
I cross compile ```gen-posix-lock-obj``` file to get header.
yepp.... I run it riscv32 tinyemu(https://bellard.org/tinyemu/) but stuck.
Because system dosent have glibc 2.27
Glibc 2.27 suport riscv-32?
[image]

pino-kim, 2018-12-03:
Hi good news.
get correct header on riscv32

1.build riscv32 compiler
https://github.com/riscv/riscv-gnu-toolchain

2. build gen-opsix-lock-obj for riscv32
$ cd libgpg-error
$ export CC=riscv32-unknown-linux-gnu-gcc 
$ ./configure --host=riscv32-unknown-linux-gnu --build=build=x86_64-linux-gnu  --enable-static=yes --enable-shared=no --sys=/opt/riscv/lib
$ cd src
$ make gen-posix-lock-obj

4. set riscv32 emulate system.
set tinyemu riscv32 : https://bellard.org/tinyemu/buildroot.html

5. run gpgeror gen-posix-lock-obj

[image]

alistair23, 2018-12-03:
Awesome! Thanks for doing that.

I have updated my branch to reflect what you have there. I will send a pull request later today.

If you have a SoB line I can add it to the libgpg-error patch if you want?

alistair23, 2018-12-03:
Also, my 32-bit RISC-V kernel doesn't run on QEMU, what kernel are you using?

pino-kim, 2018-12-03:
you can see kernel 4.15 above pic. I prepare 

> Awesome! Thanks for doing that.
> 
> I have updated my branch to reflect what you have there. I will send a pull request later today.
> 
> If you have a SoB line I can add it to the libgpg-error patch if you want?

Q/A time plz
1. hmm? what is Sob line? 
- like this ? 

2. Tinyemu use 4.15 kernel riscv

alistair23, 2018-12-03:
Just bitbake core-image-minimal and then runqemu

pino-kim, 2018-12-03:
@kraj , @alistair23 
yepp I check bootup core-image-minimal.
let close it.

closed 2018-12-03 by pino-kim