riscv/meta-riscv #64: qemuriscv32 : core-image-full-cmdline build fail(last commit : 0742b61)

pino-kim, 2018-12-10:
**Description**
Build test last  0742b61. 
riscv 64bit build fine. but I met build error last commit  0742b61 when I build riscv 32bit.
let's digging.

**Steps to reproduce the issue:**
1. . ./meta-riscv/setup.sh
2. replace MACHINE="qemuriscv64" to MACHINE="qemuriscv32"
3. bitbakae core-image-full-cmdline -k

**Describe the results you received:**

> ERROR: core-image-full-cmdline-1.0-r0 do_rootfs: The postinstall intercept hook 'update_udev_hwdb' failed, details in /home/pino-yocto/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/temp/log.do_rootfs
ERROR: core-image-full-cmdline-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/pino-yocto/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/temp/log.do_rootfs.5405
ERROR: Task (/home/pino-yocto/workspace/opensource_work/riscv-yocto/openembedded-core/meta/recipes-extended/images/core-image-full-cmdline.bb:do_rootfs) failed with exit code '1'

**Describe the results you expected:**
- build fine, runqemu fine.

**Additional details (revisions used, host distro, etc.):**
>Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv32-oe-linux"
MACHINE              = "qemuriscv32"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv32 littleendian"
meta                 = "HEAD:924d6956fb5bef3effe8914fe61e8cad6f71a231"
meta-oe
meta-multimedia
meta-python
meta-networking      = "HEAD:ebc7b9e20ac22f6f2ad373621917f53e8a9af81c"
meta-riscv           = "HEAD:0742b61e65e9f1388c6a8d5cecd0b337951290b4"

pino-kim, 2018-12-10:
attach logfile here :-) [log.do_rootfs.5405.txt](https://github.com/riscv/meta-riscv/files/2663709/log.do_rootfs.5405.txt)

alistair23, 2018-12-10:
I can't reproduce this, it works for me. Can you share more information?

kraj, 2018-12-11:
`update_udev_hwdb` uses qemu usermode. It seems its dying on that.  @alistair23 are you using poky ? or something else ? its possible its not triggerring for distro you are using ?

alistair23, 2018-12-11:
Ah, I'm not using poky. So it sounds like we need riscv32 user mode support. That should be there though

pino-kim, 2018-12-11:
@kraj, @alistair23  
hmm.... I see ....
I will test attch ```qemu-usermode``` at 
meta-riscv/conf/machine/include/qemuriscv.inc
```MACHINE_FEATURES = "screen keyboard ext2 ext3 serial"```

pino-kim, 2018-12-11:
@kraj , @alistair23 
thanks to give me a hint.
I can build riscv32  attach below line.
I will make pr to review it.
```# QEMU usermode fails with invalid instruction error (YOCTO: #10304)
MACHINE_FEATURES_BACKFILL_CONSIDERED_append_riscv64 = "${@bb.utils.contains('TUNE_FEATURES', 'riscv64', ' qemu-usermode', '', d)}"
MACHINE_FEATURES_BACKFILL_CONSIDERED_append_riscv32 = "${@bb.utils.contains('TUNE_FEATURES', 'riscv32', ' qemu-usermode', '', d)}"
```

kraj, 2018-12-13:
This should be fixed now

closed 2018-12-13 by kraj