riscv/meta-riscv #69: Enable experimental libstdc++ in Yocto Toolchain

alistair23, 2018-12-14:
I'm using Yocto to generate a baremetal tool chain with the following command:
`MACHINE=baremetal-riscv64 bitbake meta-toolchain`

I'm then trying to use that toolchain to build the [Kendryte FreeRTOS port](https://github.com/kendryte/kendryte-freertos-sdk).

I source the toolchain
`. /usr/local/oecore-x86_64/environment-setup-riscv64-oe-elf`
`. /usr/local/oecore-x86_64/environment-setup-riscv64-oe-linux `

Then I fix their cmake file (see diff at end) and disable reent as that seems to only work with newlib. After that I try and build FreeRTOS but I see this error:

```
kendryte-freertos-sdk/lib/freertos/include/kernel/object.hpp:46:41: error: expected ‘)’ before ‘in’
     constexpr object_ptr(std::in_place_t in, T *obj) noexcept
                         ~               ^~~
                                         )
```

After grepping through the GCC code it looks like in_place_t is an experimental feature. Is it possible to enable experimental libstdc++ features in the Yocto toolchain?

Fix CMake
```
diff --git a/cmake/toolchain.cmake b/cmake/toolchain.cmake
index 9d0afd4..359c7d6 100644
--- a/cmake/toolchain.cmake
+++ b/cmake/toolchain.cmake
@@ -4,32 +4,6 @@ else ()
     set(EXT "")
 endif ()
 
-message(STATUS "Check for RISCV toolchain ...")
-
-if(NOT TOOLCHAIN)
-       find_path(_TOOLCHAIN riscv64-unknown-elf-gcc${EXT})
-    global_set(TOOLCHAIN "${_TOOLCHAIN}")
-elseif(NOT "${TOOLCHAIN}" MATCHES "/$")
-    global_set(TOOLCHAIN "${TOOLCHAIN}")
-endif()
-
-if (NOT TOOLCHAIN)
-    message(FATAL_ERROR "TOOLCHAIN must be set, to absolute path of kendryte-toolchain dist/bin folder.")
-endif ()
-
-message(STATUS "Using ${TOOLCHAIN} RISCV toolchain")
-
-global_set(CMAKE_C_COMPILER "${TOOLCHAIN}/riscv64-unknown-elf-gcc${EXT}")
-global_set(CMAKE_CXX_COMPILER "${TOOLCHAIN}/riscv64-unknown-elf-g++${EXT}")
-global_set(CMAKE_LINKER "${TOOLCHAIN}/riscv64-unknown-elf-ld${EXT}")
-global_set(CMAKE_AR "${TOOLCHAIN}/riscv64-unknown-elf-ar${EXT}")
-global_set(CMAKE_OBJCOPY "${TOOLCHAIN}/riscv64-unknown-elf-objcopy${EXT}")
-global_set(CMAKE_SIZE "${TOOLCHAIN}/riscv64-unknown-elf-size${EXT}")
-global_set(CMAKE_OBJDUMP "${TOOLCHAIN}/riscv64-unknown-elf-objdump${EXT}")
-if (WIN32)
-    global_set(CMAKE_MAKE_PROGRAM "${TOOLCHAIN}/mingw32-make${EXT}")
-endif ()
-
 execute_process(COMMAND ${CMAKE_C_COMPILER} -print-file-name=crt0.o OUTPUT_STRIP_TRAILING_WHITESPACE OUTPUT_VARIABLE CRT0_OBJ)
 execute_process(COMMAND ${CMAKE_C_COMPILER} -print-file-name=crtbegin.o OUTPUT_STRIP_TRAILING_WHITESPACE OUTPUT_VARIABLE CRTBEGIN_OBJ)
 execute_process(COMMAND ${CMAKE_C_COMPILER} -print-file-name=crtend.o OUTPUT_STRIP_TRAILING_WHITESPACE OUTPUT_VARIABLE CRTEND_OBJ)
```

Disable reent
```
diff --git a/lib/freertos/conf/FreeRTOSConfig.h b/lib/freertos/conf/FreeRTOSConfig.h
index 47136e2..bb613e8 100644
--- a/lib/freertos/conf/FreeRTOSConfig.h
+++ b/lib/freertos/conf/FreeRTOSConfig.h
@@ -62,7 +62,7 @@
 #define configTICK_RATE_HZ					( ( TickType_t ) 100 )
 
 /* multithreading */
-#define configUSE_NEWLIB_REENTRANT				1
+#define configUSE_NEWLIB_REENTRANT				0
 
 #define configUSE_PREEMPTION					1
 #define configUSE_PORT_OPTIMISED_TASK_SELECTION	0
```

pino-kim, 2018-12-14:
Hello @alistair23 
Can I join your issue?
First, To get toolchain, build it by command  ```bitbake meta-toolchain``` ?

and to build ```Kendryte FreeRTOS port```
Type below command?
>mkdir build && cd build
cmake .. -DPROJ=<ProjectName> -DTOOLCHAIN=/opt/riscv-toolchain/bin && make

alistair23, 2018-12-14:
Ah, it looks like the experimental features are enabled by the command line when calling GCC. I had tried -std-gnu++14 previously but apparently -std=gnu++17 is required. I'll close this issue as I don't think changes to Yocto are required.

Now, the next error of a missing diskio.h file.

closed 2018-12-14 by alistair23

kraj, 2018-12-14:
right c++17 is sort of complete in gcc 8, so the project should explicitly ask for it since the default for gcc is gnu++14 standard but you can demand a different std in CMakeLists.txt add
```
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)
```

and that should help.

alistair23, 2018-12-15:
Ok, progress! Now it looks like the Linux toolchain is missing "machine/syscall.h" which is available in the elf toolchain but that causes all sorts of conflicts.

kraj, 2018-12-15:
`machine/syscall.h` should be provided by C library, and in this case where bare-metal app is being built I expect the app to provide it.

alistair23, 2018-12-15:
This isn't really baremetal as it relied on a lot of libraries and uses the Linux toolchain. The problem is that the Linux toolchain doesn't include `machine/syscall.h` and the one in the elf toolchain conflicts with other Linux toolchain includes.

kraj, 2018-12-15:
I should have said, stand-alone app ( which is not hosted on an OperatingSystem ), this header is part of the runtime ABI of a platform. If it is expecting this from toolchain then toolchain has to contain the target ABI support like Linux/Freebsd/ etc. OE builds either toolchain for building applications for Linux or free-standing ( which provide their own runtime), hope that makes it a bit clear.

alistair23, 2018-12-17:
Ok, I almost have it working. Now I see this weird error of missing standard libraries:

```
/usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/usr/bin/riscv64-oe-linux/riscv64-oe-linux-gcc   --sysroot=/usr/local/oecore-x86_64/sysroots/riscv64-oe-linux --sysroot=/usr/local/oecore-x86_64/sysroots/riscv64-oe-elf   -O2 -pipe -g -feliminate-unused-debug-types -mcmodel=medany -march=rv64imafdc -fno-common -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -fno-zero-initialized-in-bss -ggdb -std=gnu11 -L"/scratch/alistair/software/tier3/kendryte/kendryte-freertos-sdk/include/" -g   -nostartfiles -static -Wl,--gc-sections -Wl,-static -Wl,--start-group -Wl,--whole-archive -Wl,--no-whole-archive -Wl,--end-group -Wl,-EL -T "/scratch/alistair/software/tier3/kendryte/kendryte-freertos-sdk/lds/kendryte.ld" -rdynamic "/usr/local/oecore-x86_64/sysroots/riscv64-oe-linux/usr/lib/riscv64-oe-linux/8.2.0/crti.o" "/usr/local/oecore-x86_64/sysroots/riscv64-oe-linux/usr/lib/riscv64-oe-linux/8.2.0/crtbegin.o" CMakeFiles/Project.dir/src/hello_world/main.c.o "/usr/local/oecore-x86_64/sysroots/riscv64-oe-linux/usr/lib/riscv64-oe-linux/8.2.0/crtend.o" "/usr/local/oecore-x86_64/sysroots/riscv64-oe-linux/usr/lib/riscv64-oe-linux/8.2.0/crtn.o" -o Project -Wl,--start-group -lm SDK/freertos/libfreertos.a -latomic SDK/bsp/libbsp.a -lc -lstdc++ SDK/drivers/libdrivers.a -Wl,--end-group SDK/freertos/libfreertos.a SDK/hal/libhal.a -lm -lfatfs -lstdc++ -lm
/usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/usr/libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/8.2.0/real-ld: cannot find -lm
/usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/usr/libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/8.2.0/real-ld: cannot find -latomic
/usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/usr/libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/8.2.0/real-ld: cannot find -lc
/usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/usr/libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/8.2.0/real-ld: cannot find -lstdc++
collect2: error: ld returned 1 exit status
```

Any ideas why GCC thinks they are missing?

kraj, 2018-12-28:
there was a bug in gcc multilib in OE which is fixed now via [patch](https://patchwork.openembedded.org/patch/157481/)
Can you see if this helps ?

alistair23, 2018-12-29:
Thanks, I'll check the patch and let you know