riscv/meta-riscv #75: [Question] I stcck get meta-toolchain package.

pino-kim, 2018-12-25:
**Description**
This Question may be issue. I Stuck to get meta-toolchain package when I follow #69 .
So I type ```bitbake meta-toolchain``` how can I get meta-toolchain package?
And happy X-mas.

**Steps to reproduce the issue:**
1. run setup.sh script.
2. bitbake meta-toolchain
3. stuck at gdb-cross-canadian-riscv64

**Describe the results you received:**
>/home/pino/workspace/Opensource/riscv-yocto/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/gdb-cross-canadian-riscv64/riscv-r0/git/missing: line 81: bison: command not found
WARNING: 'bison' is missing on your system.
         You should only need it if you modified a '.y' file.
         You may want to install the GNU Bison package:
         <http://www.gnu.org/software/bison/>

...

>Makefile:849: recipe for target 'all' failed
make: *** [all] Error 2
ERROR: oe_runmake failed
WARNING: /home/pino/workspace/Opensource/riscv-yocto/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/gdb-cross-canadian-riscv64/riscv-r0/temp/run.do_compile.21026:1 exit 1 from 'exit 1'
ERROR: Function failed: do_compile (log file is located at /home/pino/workspace/Opensource/riscv-yocto/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/gdb-cross-canadian-riscv64/riscv-r0/temp/log.do_compile.21026)

I attach error log ```log.do_compile.21026``` here
[meta-toolchain_buildlog.txt](https://github.com/riscv/meta-riscv/files/2708450/meta-toolchain_buildlog.txt)

**Describe the results you expected:**
get toolchain packaget

**Additional details (revisions used, host distro, etc.):**
>NOTE: Resolving any missing task queue dependencies

>Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64 littleendian"
meta                 = "work:95659bed3f6f3216b346f70cfc9ffae9788c0fc1"
meta-oe    
meta-multimedia    
meta-python    
meta-networking      = "work:e052ea1b7494715d95c88c81746a987452707282"
meta-riscv           = "HEAD:5b1e9f2fe01430891b94bd43e914397680d0c3d6"

kraj, 2018-12-25:
@pino-kim can you add `bison-native` to `DEPENDS` in `meta/recipes-devtools/gdb/gdb-cross-canadian.inc` from openembedded-core repo and see if that fixes problem

pino-kim, 2018-12-26:
@kraj Oh Build success. should we send a patch to OE?
I add a ```bision-native``` at  ```meta/recipes-devtools/gdb/gdb-cross-canadian.inc```  and get package.

kraj, 2018-12-26:
yes

pino-kim, 2018-12-26:
Ok I prepare patch for OE. Thanks.

pino-kim, 2018-12-26:
reopen it.
build fail current OE : 7bbc453ac2b8d63680855789948a145fc448017f

closed 2018-12-26 by pino-kim