riscv/meta-riscv #76: Missing DEPENDS when building SDK

tuanha-viosoft, 2018-12-26:
**Description**

**Steps to reproduce the issue:**
1. Build SDK
build$ bitbake core-image-full-cmdline -c populate_sdk

**Describe the results you received:**
| /home/ubuntu/riscv-yocto/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/gdb-cross-canadian-riscv
64/riscv-r0/git/missing: 81: /home/ubuntu/riscv-yocto/build/tmp-glibc/work/x86_64-nativesdk-oesdk-lin
ux/gdb-cross-canadian-riscv64/riscv-r0/git/missing: bison: not found
| Makefile:2483: recipe for target 'rust-exp.c' failed
| make[2]: *** [rust-exp.c] Error 1
| Makefile:2483: recipe for target 'm2-exp.c' failed
| make[2]: *** [m2-exp.c] Error 1
| WARNING: 'bison' is missing on your system.
|          You should only need it if you modified a '.y' file.
|          You may want to install the GNU Bison package:
|          <http://www.gnu.org/software/bison/>
| Makefile:2483: recipe for target 'p-exp.c' failed

**Describe the results you expected:**
Build successfully

**Additional details (revisions used, host distro, etc.):**
This bug can be fixed by adding dependencies in recipe:
diff --git a/recipes-devtools/gdb/gdb-cross-canadian_riscv.bb b/recipes-devtools/gdb/gdb-cross-canadian_riscv.bb
index 5c79fdd..48daaea 100644
--- a/recipes-devtools/gdb/gdb-cross-canadian_riscv.bb
+++ b/recipes-devtools/gdb/gdb-cross-canadian_riscv.bb
@@ -1,3 +1,4 @@
 require recipes-devtools/gdb/gdb-common.inc
 require recipes-devtools/gdb/gdb-cross-canadian.inc
 require gdb-${PV}.inc
+DEPENDS += " bison-native "

kraj, 2018-12-26:
please prepare a patch to add it to `gdb-cross-canadian_riscv.bb` while we work to get it into oe-core

kraj, 2018-12-26:
btw. can you test it with `nativesdk-bison` instead of `bison-native`

pino-kim, 2018-12-26:
@tuanha-viosoft 
Oh can you check my previous QA #75 ?
I get sdk ```bitbake meta-toolchain``` cmd. 

I'll reproduce it add ```nativesdk-bison``` and ```native-bison```

tuanha-viosoft, 2018-12-26:
> Oh can you check my previous QA #75 ?

Sorry, I don't see it. It should be the same issue.

tuanha-viosoft, 2018-12-26:
#75

closed 2018-12-26 by tuanha-viosoft

pino-kim, 2018-12-26:
@kraj, @tuanha-viosoft
I decide to reopen issue #75.
before send my PR to OE 
(https://github.com/pino-kim/openembedded-core/commit/03df3dfac5f788defe02a14fc730514c10891cc4 )
I. run build test it . I success build it on OE : 95659bed3f6f3216b346f70cfc9ffae9788c0fc1
but I can't build it OE : 7bbc453ac2b8d63680855789948a145fc448017f.

So we need to figure out again.