riscv/meta-riscv #77: SDK doesn't work

tuanha-viosoft, 2018-12-27:
**Description**

**Steps to reproduce the issue:**
1. Build SDK
`~/riscv-yocto/build$ bitbake core-image-full-cmdline -c populate_sdk`
2. Extract SDK
```
~/riscv-yocto/build$ cd tmp-glibc/deploy/sdk
~/riscv-yocto/build/tmp-glibc/deploy/sdk$ ./oecore-x86_64-riscv64-toolchain-nodistro.0.sh
```
3. Build a Debian package by SDK (Ex: flex package)
```
$ source /opt/x86_64-riscv64-toolchain/environment-setup-riscv64-oe-linux
$ apt-get source flex
$ cd flext-2.5.39
$ ./configure --host=riscv64-unknown-linux CC=riscv64-oe-linux-gcc BUILD_CC=riscv64-oe-linux-gcc CXX=riscv64-oe-linux-g++ BUILD_CXX=riscv64-oe-linux-g++ AR=riscv64-oe-linux-ar RANLIB=riscv64-oe-linux-ranlib LD=riscv64-oe-linux-ld ARCH=riscv64
```

**Describe the results you received:**
Configuration failed.

> configure: loading site script /opt/x86_64-riscv64-toolchain/site-config-riscv64-oe-linux
> checking for a BSD-compatible install... /usr/bin/install -c
> checking whether build environment is sane... yes
> checking for riscv64-unknown-linux-strip... riscv64-oe-linux-strip
> checking for a thread-safe mkdir -p... /bin/mkdir -p
> checking for gawk... gawk
> checking whether make sets $(MAKE)... yes
> checking whether NLS is requested... yes
> checking for msgfmt... /usr/bin/msgfmt
> checking for gmsgfmt... /usr/bin/msgfmt
> checking for xgettext... /usr/bin/xgettext
> checking for msgmerge... /usr/bin/msgmerge
> checking for style of include used by make... GNU
> checking for riscv64-unknown-linux-gcc... riscv64-oe-linux-gcc
> checking whether the C compiler works... no
> configure: error: in `/home/viosoft/test/flex-2.5.39':
> configure: error: C compiler cannot create executables
> See `config.log' for more details

**Describe the results you expected:**
Run configuration successfully

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

kraj, 2018-12-28:
@tuanha-viosoft  Just use

```
./configure --host=riscv64-unknown-linux
```

SDK env script should have set the needed cross tools in environment for you already

tuanha-viosoft, 2018-12-28:
I had the same issue.

`flex-2.5.39$ ./configure --host=riscv64-unknown-linuxconfigure: loading site script /opt/x86_64-riscv64-toolchain/site-config-riscv64-oe-linux`
> checking for a BSD-compatible install... /usr/bin/install -c
> checking whether build environment is sane... yes
> checking for riscv64-unknown-linux-strip... riscv64-oe-linux-strip
> checking for a thread-safe mkdir -p... /bin/mkdir -p
> checking for gawk... gawk
> checking whether make sets $(MAKE)... yes
> checking whether NLS is requested... yes
> checking for msgfmt... /usr/bin/msgfmt
> checking for gmsgfmt... /usr/bin/msgfmt
> checking for xgettext... /usr/bin/xgettext
> checking for msgmerge... /usr/bin/msgmerge
> checking for style of include used by make... GNU
> checking for riscv64-unknown-linux-gcc... riscv64-oe-linux-gcc  --sysroot=/opt/x86_64-riscv64-toolchain/sysroots/riscv64-oe-linux
> checking whether the C compiler works... no
> configure: error: in `/home/viosoft/test/flex-2.5.39':
> configure: error: C compiler cannot create executables
> See `config.log' for more details

kraj, 2018-12-28:
OK, I will try to reproduce it here

tuanha-viosoft, 2018-12-28:
> It is working for me, can you build SDK from latest master ?
> 
> ```
> % ./configure --host=riscv64-linux
> configure: loading site script /opt/yoe/2.6/site-config-riscv64-yoe-linux
> checking for a BSD-compatible install... /usr/bin/install -c
> checking whether build environment is sane... yes
> checking for riscv64-linux-strip... riscv64-yoe-linux-strip
> checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
> checking for gawk... gawk
> checking whether make sets $(MAKE)... yes
> checking whether NLS is requested... yes
> checking for msgfmt... /usr/bin/msgfmt
> checking for gmsgfmt... /usr/bin/msgfmt
> checking for xgettext... /usr/bin/xgettext
> checking for msgmerge... /usr/bin/msgmerge
> checking for style of include used by make... GNU
> checking for riscv64-linux-gcc... riscv64-yoe-linux-gcc  --sysroot=/opt/yoe/2.6/sysroots/riscv64-yoe-linux
> checking whether the C compiler works... yes
> checking for C compiler default output file name... a.out
> checking for suffix of executables... 
> checking whether we are cross compiling... yes
> checking for suffix of object files... o
> checking whether we are using the GNU C compiler... yes
> checking whether riscv64-yoe-linux-gcc  --sysroot=/opt/yoe/2.6/sysroots/riscv64-yoe-linux accepts -g... yes
> checking for riscv64-yoe-linux-gcc  --sysroot=/opt/yoe/2.6/sysroots/riscv64-yoe-linux option to accept ISO C89... none needed
> checking dependency style of riscv64-yoe-linux-gcc  --sysroot=/opt/yoe/2.6/sysroots/riscv64-yoe-linux... gcc3
> checking build system type... x86_64-unknown-linux-gnu
> ...
> ```

I got new issue when make:
```
$ make
make[1]: Entering directory '/home/viosoft/runtime-EclipseApplication/fda/Release-riscv64-oe-linux-/userlevel'
  FINDELEMENTS elements.conf
  CREATE elements.mk
  CXX ../../teranium_click/lib/string.cc
In file included from /opt/oecore-x86_64/sysroots/riscv64-oe-linux/usr/include/c++/8.2.0/stdlib.h:36,
                 from ../../teranium_click/include/click/glue.hh:80,
                 from ../../teranium_click/include/click/straccum.hh:4,
                 from ../../teranium_click/lib/string.cc:24:
/opt/oecore-x86_64/sysroots/riscv64-oe-linux/usr/include/c++/8.2.0/cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:58: recipe for target 'string.o' failed
make[1]: *** [string.o] Error 1
make[1]: Leaving directory '/home/viosoft/runtime-EclipseApplication/fda/Release-riscv64-oe-linux-/userlevel'
Makefile:57: recipe for target 'userlevel' failed
make: *** [userlevel] Error 2
```

kraj, 2018-12-28:
Posted a [patch](https://patchwork.openembedded.org/patch/157481/) to fix it in OE-Core

Please apply it to openembedded-core repo rebuild the SDK and see if this helps.

tuanha-viosoft, 2019-01-01:
> 
> 
> Posted a [patch](https://patchwork.openembedded.org/patch/157481/) to fix it in OE-Core
> 
> Please apply it to openembedded-core repo rebuild the SDK and see if this helps.

@kraj It works.

kraj, 2019-01-01:
Thanks, it will be merged after work resumes in new year

kraj, 2019-01-03:
it has been merged here http://git.openembedded.org/openembedded-core/commit/?id=badbeb01dd642d9d647d3760a3ce2aa39c3567da

closed 2019-01-03 by kraj