riscv/meta-riscv #91: why there is no <image>.wic.gz file exists after bitbake the console image finished?

13824125580, 2019-01-29:
i want to flash the sd card to boot my unleased board of riscv64, but i cant find the image file on in the top yocto build directiry hierachy.  so am i missed something?
thanks for your help.

alistair23, 2019-01-29:
The files are in the tmp-glibc/deploy/images/${MACHINE} directory

13824125580, 2019-01-30:
that is my file list in the directory, still cant find the wic file the read me said.
-rwxr-xr-x 1 czl czl   12601144  1月 29 14:39 bbl
-rw-rw-r-- 2 czl czl       1243  1月 29 17:44 core-image-full-cmdline-qemuriscv64-20190129071611.qemuboot.conf
-rw-r--r-- 2 czl czl  127620096  1月 29 17:44 core-image-full-cmdline-qemuriscv64-20190129071611.rootfs.ext4
-rw-r--r-- 2 czl czl      12102  1月 29 17:44 core-image-full-cmdline-qemuriscv64-20190129071611.rootfs.manifest
-rw-r--r-- 2 czl czl   26284198  1月 29 17:44 core-image-full-cmdline-qemuriscv64-20190129071611.rootfs.tar.bz2
-rw-r--r-- 2 czl czl     197608  1月 29 17:44 core-image-full-cmdline-qemuriscv64-20190129071611.testdata.json
lrwxrwxrwx 2 czl czl         62  1月 29 17:44 core-image-full-cmdline-qemuriscv64.ext4 -> core-image-full-cmdline-qemuriscv64-20190129071611.rootfs.ext4
lrwxrwxrwx 2 czl czl         66  1月 29 17:44 core-image-full-cmdline-qemuriscv64.manifest -> core-image-full-cmdline-qemuriscv64-20190129071611.rootfs.manifest
lrwxrwxrwx 2 czl czl         64  1月 29 17:44 core-image-full-cmdline-qemuriscv64.qemuboot.conf -> core-image-full-cmdline-qemuriscv64-20190129071611.qemuboot.conf
lrwxrwxrwx 2 czl czl         65  1月 29 17:44 core-image-full-cmdline-qemuriscv64.tar.bz2 -> core-image-full-cmdline-qemuriscv64-20190129071611.rootfs.tar.bz2
lrwxrwxrwx 2 czl czl         64  1月 29 17:44 core-image-full-cmdline-qemuriscv64.testdata.json -> core-image-full-cmdline-qemuriscv64-20190129071611.testdata.json
-rw-rw-r-- 2 czl czl       1223  1月 29 21:46 core-image-lsb-sdk-qemuriscv64-20190129103108.qemuboot.conf
-rw-r--r-- 2 czl czl 1524891648  1月 29 21:49 core-image-lsb-sdk-qemuriscv64-20190129103108.rootfs.ext4
-rw-r--r-- 2 czl czl     129572  1月 29 21:46 core-image-lsb-sdk-qemuriscv64-20190129103108.rootfs.manifest
-rw-r--r-- 2 czl czl  298490997  1月 29 21:49 core-image-lsb-sdk-qemuriscv64-20190129103108.rootfs.tar.bz2
-rw-r--r-- 2 czl czl     198278  1月 29 21:46 core-image-lsb-sdk-qemuriscv64-20190129103108.testdata.json
lrwxrwxrwx 2 czl czl         57  1月 29 21:49 core-image-lsb-sdk-qemuriscv64.ext4 -> core-image-lsb-sdk-qemuriscv64-20190129103108.rootfs.ext4
lrwxrwxrwx 2 czl czl         61  1月 29 21:46 core-image-lsb-sdk-qemuriscv64.manifest -> core-image-lsb-sdk-qemuriscv64-20190129103108.rootfs.manifest
lrwxrwxrwx 2 czl czl         59  1月 29 21:46 core-image-lsb-sdk-qemuriscv64.qemuboot.conf -> core-image-lsb-sdk-qemuriscv64-20190129103108.qemuboot.conf
lrwxrwxrwx 2 czl czl         60  1月 29 21:50 core-image-lsb-sdk-qemuriscv64.tar.bz2 -> core-image-lsb-sdk-qemuriscv64-20190129103108.rootfs.tar.bz2
lrwxrwxrwx 2 czl czl         59  1月 29 21:46 core-image-lsb-sdk-qemuriscv64.testdata.json -> core-image-lsb-sdk-qemuriscv64-20190129103108.testdata.json
-rw-rw-r-- 2 czl czl    2801833  1月 29 12:48 modules--4.19.14+git0+70d33ded25_eebb51300a-r0-qemuriscv64-20190129043355.tgz
lrwxrwxrwx 2 czl czl         77  1月 29 12:48 modules-qemuriscv64.tgz -> modules--4.19.14+git0+70d33ded25_eebb51300a-r0-qemuriscv64-20190129043355.tgz
lrwxrwxrwx 2 czl czl         33  1月 29 14:38 u-boot.bin -> u-boot-qemuriscv64-2018.11-r0.bin
lrwxrwxrwx 2 czl czl         33  1月 29 14:38 u-boot.elf -> u-boot-qemuriscv64-2018.11-r0.elf
-rw-r--r-- 2 czl czl     300560  1月 29 14:38 u-boot-qemuriscv64-2018.11-r0.bin
-rw-r--r-- 2 czl czl    3366664  1月 29 14:38 u-boot-qemuriscv64-2018.11-r0.elf
lrwxrwxrwx 2 czl czl         33  1月 29 14:38 u-boot-qemuriscv64.bin -> u-boot-qemuriscv64-2018.11-r0.bin
lrwxrwxrwx 2 czl czl         33  1月 29 14:38 u-boot-qemuriscv64.elf -> u-boot-qemuriscv64-2018.11-r0.elf
lrwxrwxrwx 2 czl czl         77  1月 29 12:48 vmlinux -> vmlinux--4.19.14+git0+70d33ded25_eebb51300a-r0-qemuriscv64-20190129043355.bin
-rw-r--r-- 2 czl czl   12689720  1月 29 12:48 vmlinux--4.19.14+git0+70d33ded25_eebb51300a-r0-qemuriscv64-20190129043355.bin
lrwxrwxrwx 2 czl czl         77  1月 29 12:48 vmlinux-qemuriscv64.bin -> vmlinux--4.19.14+git0+70d33ded25_eebb51300a-r0-qemuriscv64-20190129043355.bin

kraj, 2019-01-30:
We do not generate wic images for QEMU machines, But it could be added just add
`IMAGE_FSTYPES += "wic.xz"` in `local.conf` or alternatively use freedom-u board as baseline

closed 2019-07-08 by kraj