riscv/meta-riscv #98: Cross compile fortran failed

tuanha-viosoft, 2019-02-18:
**Description**
I tried to bitbake lapack, it depends on libgfortran, but nothing provides 'libgfortran'

**Steps to reproduce the issue:**
1. build$ bitbake lapack

**Describe the results you received:**
ERROR: Nothing PROVIDES 'libgfortran' (but /home/ubuntu/riscv-yocto/meta-openembedded/meta-oe/recipes-devtools/lapack/lapack_3.8.0.bb DEPENDS on or otherwise requires it)
libgfortran was skipped: libgfortran needs fortran support to be enabled in the compiler
ERROR: Required build target 'lapack' has no buildable providers.
Missing or unbuildable dependency chain was: ['lapack', 'libgfortran']

**Describe the results you expected:**
Bitbake successfully

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

pino-kim, 2019-02-18:
@tuanha-viosoft 

I repoduce your issue below ENV. Is it same?

-------------------------------------------------------
Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64 littleendian"
meta                 = "work:c118705f1d4c8f4f04a7ba517b74c90829fdd81b"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      = "work:c06db080f2331d98fb5877531baf7d0da95fba55"
meta-riscv           = "HEAD:2ec848f8f37ea3c0d3538f32a59133294a9234cc"

tuanha-viosoft, 2019-02-18:
This is my Build Configuration:

BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64 littleendian"
meta                 = "work:c118705f1d4c8f4f04a7ba517b74c90829fdd81b"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      = "work:c06db080f2331d98fb5877531baf7d0da95fba55"
meta-riscv           = "work:2ec848f8f37ea3c0d3538f32a59133294a9234cc"

kraj, 2019-02-18:
@tuanha-viosoft we do not build fortran support in gcc by default, so please add a 
`FORTRAN = ",fortran"` to your local.conf

tuanha-viosoft, 2019-02-18:
> @tuanha-viosoft we do not build fortran support in gcc by default, so please add a
> `FORTRAN = ",fortran"` to your local.conf

I tried. But it doesn't work. I still met this error:

build$ bitbake libgfortran
Loading cache: 100% |#################################| Time: 0:00:00
Loaded 3239 entries from dependency cache.
Parsing recipes: 100% |###############################| Time: 0:00:01
Parsing of 2198 .bb files complete (2197 cached, 1 parsed). 3240 targets, 135 skipped, 0 masked, 0 errors.
ERROR: Nothing PROVIDES 'libgfortran'
libgfortran was skipped: libgfortran needs fortran support to be enabled in the compiler

Summary: There was 1 ERROR message shown, returning a non-zero exit code.

pino-kim, 2019-02-18:
Hello @kraj @tuanha-viosoft 

I attach below line at auto.conf(local.conf)
and bit bitbake lapack success. 

+# Enable fortran
+FORTRAN_forcevariable = ",fortran"

tuanha-viosoft, 2019-02-18:
> Hello @kraj @tuanha-viosoft
> 
> I attach below line at auto.conf(local.conf)
> and bit bitbake lapack success.
> 
> +# Enable fortran
> +FORTRAN_forcevariable = ",fortran"

@pino-kim It works. Thanks a lot

closed 2019-02-18 by tuanha-viosoft

hinar24, 2019-06-27:
Hi 
after adding # Enable fortran and FORTRAN_forcevariable = ",fortran" in local.conf, I am facing below error. Attached log file.
ERROR: Function failed: do_configure (log file is located at /home/heena/project/source/apps_proc/poky/build/tmp-glibc/work/aarch64-oe-linux/libgfortran/5.2.0-r0/temp/log.do_configure.24296)

[log_fortan.txt](https://github.com/riscv/meta-riscv/files/3334221/log_fortan.txt)

kraj, 2019-06-27:
Can you paste ‘config.log’ from libgfortran build tree

hinar24, 2019-07-05:
Hi 
please can you help me to find out config.log file for libgfortran.
I have attached log file from /project/source/apps_proc/poky/meta/recipes-devtools/gcc
[tree.log](https://github.com/riscv/meta-riscv/files/3362579/tree.log)

kraj, 2019-07-08:
I see this

```
../../../../../../../../work-shared/gcc-5.2.0-r0/gcc-5.2.0/libgfortran/configure: line 1899: aarch64-oe-linux-gfortran: command not found
configure:15686: $? = 127
configure: failed program was:
| 
|       program foo
|       real, parameter :: bar = sin (12.34 / 2.5)
|       end program foo
configure:15690: result: no
configure:15692: error: GNU Fortran is not working; please report a bug in http://gcc.gnu.org/bugzilla, attaching /home/heena/project/source/apps_proc/poky/build/tmp-glibc/work/aarch64-oe-linux/libgfortran/5.2.0-r0/gcc-5.2.0/build.aarch64-oe-linux.aarch64-oe-linux/aarch64-oe-linux/libgfortran/config.log
```

I wonder how is it getting to aarch64 as target for libgfortan, thats the obvious error