riscv/meta-riscv #99: Compile libmysqlclient failed

tuanha-viosoft, 2019-02-20:
**Description**
I build libmysqlclient and got the error while compiling.

**Steps to reproduce the issue:**
1. build$ bitbake libmysqlclient

**Describe the results you received:**
Loading cache: 100% |############################################################################################################| Time: 0:00:00
Loaded 3243 entries from dependency cache.
Parsing recipes: 100% |##########################################################################################################| Time: 0:00:01
Parsing of 2200 .bb files complete (2199 cached, 1 parsed). 3244 targets, 133 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64 littleendian"
meta                 = "work:c118705f1d4c8f4f04a7ba517b74c90829fdd81b"
meta-oe
meta-multimedia
meta-python
meta-networking      = "work:c06db080f2331d98fb5877531baf7d0da95fba55"
meta-riscv           = "work:2ec848f8f37ea3c0d3538f32a59133294a9234cc"

Initialising tasks: 100% |#######################################################################################################| Time: 0:00:01
Sstate summary: Wanted 8 Found 0 Missed 8 Current 439 (0% match, 98% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: mariadb-5.5.62-r0 do_compile: Function failed: do_compile (log file is located at /home/ubuntu/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/mariadb/5.5.62-r0/temp/log.do_compile.19190)
ERROR: Logfile of failure stored in: /home/ubuntu/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/mariadb/5.5.62-r0/temp/log.do_compile.19190
...
| make[2]: Leaving directory '/home/ubuntu/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/mariadb/5.5.62-r0/build'
| [ 76%] Built target mysqlbinlog
| make[1]: Leaving directory '/home/ubuntu/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/mariadb/5.5.62-r0/build'
| Makefile:165: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 2 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/ubuntu/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/mariadb/5.5.62-r0/temp/log.do_compile.19190)
ERROR: Task (/home/ubuntu/riscv-yocto/meta-openembedded/meta-oe/recipes-dbs/mysql/mariadb_5.5.62.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1877 tasks of which 1876 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory

Summary: 1 task failed:
  /home/ubuntu/riscv-yocto/meta-openembedded/meta-oe/recipes-dbs/mysql/mariadb_5.5.62.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

**Describe the results you expected:**
Run bitbake successfully

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

[log.do_compile.19190.txt](https://github.com/riscv/meta-riscv/files/2884044/log.do_compile.19190.txt)

pino-kim, 2019-02-20:
Dear @tuanha-viosoft 
I will reproduce your issue.
So did you add other meta-? for build libmysqlclient ?

tuanha-viosoft, 2019-02-20:
@pino-kim No, I don't

pino-kim, 2019-02-20:
i get some error log here.

> collect2: error: ld returned 1 exit status
cd /home/pino/workspace/Opensource/RISCV/riscv-yocto-64/build/tmp-glibc/work/riscv64-oe-linux/mariadb/5.5.62-r0/build/libmysql && rm /home/pino/workspace/Opensource/RISCV/riscv-yocto-64/build/tmp-glibc/work/riscv64-oe-linux/mariadb/5.5.62-r0/build/libmysql/libmysqlclient.a
libmysql/CMakeFiles/libmysql.dir/build.make:96: recipe for target 'libmysql/libmysqlclient.so.18..0' failed
make[2]: *** [libmysql/libmysqlclient.so.18..0] Error 1
make[2]: Leaving directory '/home/pino/workspace/Opensource/RISCV/riscv-yocto-64/build/tmp-glibc/work/riscv64-oe-linux/mariadb/5.5.62-r0/build'
CMakeFiles/Makefile2:6287: recipe for target 'libmysql/CMakeFiles/libmysql.dir/all' failed
make[1]: *** [libmysql/CMakeFiles/libmysql.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

kraj, 2019-03-04:
@tuanha-viosoft @pino-kim  Can you try this patch and see if it helps your case ?

pino-kim, 2019-03-04:
@kraj @tuanha-viosoft
I rebuild it with current head OE package after your bug label.
If compile fail I will test with your patch soon.
Thanks.

tuanha-viosoft, 2019-03-04:
@kraj @pino-kim It works

pino-kim, 2019-03-04:
@kraj @tuanha-viosoft   
Me too. Patch is work.

closed 2019-03-04 by kraj