riscv/meta-riscv #103: run setup.sh and bitbake-layer crash

pino-kim, 2019-02-25:
**Description**
When run setup.sh script print error rlog like below at bblayer-  

when script run and do 'bitbake-layers add-layer' series to add layer. return error message like below
>NOTE: Starting bitbake server...
Traceback (most recent call last):
  File "/home/pino/workspace/opensource/riscv-yocto/openembedded-core/bitbake/bin/bitbake-layers", line 103, in <module>
    ret = main()
  File "/home/pino/workspace/opensource/riscv-yocto/openembedded-core/bitbake/bin/bitbake-layers", line 71, in main
    tinfoil.prepare(True)
  File "/home/pino/workspace/opensource/riscv-yocto/openembedded-core/bitbake/lib/bb/tinfoil.py", line 418, in prepare
    self.run_command('parseConfiguration')
  File "/home/pino/workspace/opensource/riscv-yocto/openembedded-core/bitbake/lib/bb/tinfoil.py", line 476, in run_command
    raise TinfoilCommandFailed(result[1])
bb.tinfoil.TinfoilCommandFailed: Traceback (most recent call last):
  File "/home/pino/workspace/opensource/riscv-yocto/openembedded-core/bitbake/lib/bb/command.py", line 84, in runCommand
    result = command_method(self, commandline)
  File "/home/pino/workspace/opensource/riscv-yocto/openembedded-core/bitbake/lib/bb/command.py", line 285, in parseConfiguration
    command.cooker.parseConfiguration()
  File "/home/pino/workspace/opensource/riscv-yocto/openembedded-core/bitbake/lib/bb/cooker.py", line 437, in parseConfiguration
    self.handleCollections(self.data.getVar("BBFILE_COLLECTIONS"))
  File "/home/pino/workspace/opensource/riscv-yocto/openembedded-core/bitbake/lib/bb/cooker.py", line 1236, in handleCollections
    raise CollectionError("Errors during parsing layer configuration")
bb.cooker.CollectionError: Errors during parsing layer configuration

**Steps to reproduce the issue:**

>mkdir riscv-yocto && cd riscv-yocto
repo init -u git://github.com/riscv/meta-riscv  -b master -m tools/manifests/riscv-yocto.xml
repo sync
repo start work --all
. ./meta-riscv/setup.sh

**Describe the results you received:**
ready to build condition

**Describe the results you expected:**
bblayer not added and cannot build.

**Additional information you deem important (e.g. issue happens only occasionally):**
issue is up to bugzilla 
 : https://bugzilla.yoctoproject.org/show_bug.cgi?id=13183

**Additional details (revisions used, host distro, etc.):**
Host dev pc : ubuntu 16.04 64bit

pino-kim, 2019-02-25:
when type ... apt-get upgrade both 16.04 18.04
issue reproduce.

> pino@pino-dev:~$ sudo apt-get upgrade 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  gconf-service gconf-service-backend gconf2-common libgconf-2-4
  x11proto-dri2-dev x11proto-gl-dev
Use 'sudo apt autoremove' to remove them.
The following packages will be upgraded:
  bind9-host dnsutils ghostscript ghostscript-x libbind9-160 libdns-export1100
  libdns1100 libgs9 libgs9-common libirs160 libisc-export169 libisc169
  libisccc160 libisccfg160 liblwres160
15 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/9,912 kB of archives.
After this operation, 11.3 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
(Reading database ... 195919 files and directories currently installed.)
Preparing to unpack .../00-libisc-export169_1%3a9.11.3+dfsg-1ubuntu1.5_amd64.deb ...
Unpacking libisc-export169:amd64 (1:9.11.3+dfsg-1ubuntu1.5) over (1:9.11.3+dfsg-1ubuntu1.3) ...
Preparing to unpack .../01-libdns-export1100_1%3a9.11.3+dfsg-1ubuntu1.5_amd64.deb ...
Unpacking libdns-export1100 (1:9.11.3+dfsg-1ubuntu1.5) over (1:9.11.3+dfsg-1ubuntu1.3) ...
Preparing to unpack .../02-libirs160_1%3a9.11.3+dfsg-1ubuntu1.5_amd64.deb ...
Unpacking libirs160:amd64 (1:9.11.3+dfsg-1ubuntu1.5) over (1:9.11.3+dfsg-1ubuntu1.3) ...
Preparing to unpack .../03-bind9-host_1%3a9.11.3+dfsg-1ubuntu1.5_amd64.deb ...
Unpacking bind9-host (1:9.11.3+dfsg-1ubuntu1.5) over (1:9.11.3+dfsg-1ubuntu1.3) ...
Preparing to unpack .../04-dnsutils_1%3a9.11.3+dfsg-1ubuntu1.5_amd64.deb ...
Unpacking dnsutils (1:9.11.3+dfsg-1ubuntu1.5) over (1:9.11.3+dfsg-1ubuntu1.3) ...
Preparing to unpack .../05-libbind9-160_1%3a9.11.3+dfsg-1ubuntu1.5_amd64.deb ...
Unpacking libbind9-160:amd64 (1:9.11.3+dfsg-1ubuntu1.5) over (1:9.11.3+dfsg-1ubuntu1.3) ...
Preparing to unpack .../06-libisccfg160_1%3a9.11.3+dfsg-1ubuntu1.5_amd64.deb ...
Unpacking libisccfg160:amd64 (1:9.11.3+dfsg-1ubuntu1.5) over (1:9.11.3+dfsg-1ubuntu1.3) ...
Preparing to unpack .../07-libisccc160_1%3a9.11.3+dfsg-1ubuntu1.5_amd64.deb ...
Unpacking libisccc160:amd64 (1:9.11.3+dfsg-1ubuntu1.5) over (1:9.11.3+dfsg-1ubuntu1.3) ...
Preparing to unpack .../08-libdns1100_1%3a9.11.3+dfsg-1ubuntu1.5_amd64.deb ...
Unpacking libdns1100:amd64 (1:9.11.3+dfsg-1ubuntu1.5) over (1:9.11.3+dfsg-1ubuntu1.3) ...
Preparing to unpack .../09-libisc169_1%3a9.11.3+dfsg-1ubuntu1.5_amd64.deb ...
Unpacking libisc169:amd64 (1:9.11.3+dfsg-1ubuntu1.5) over (1:9.11.3+dfsg-1ubuntu1.3) ...
Preparing to unpack .../10-liblwres160_1%3a9.11.3+dfsg-1ubuntu1.5_amd64.deb ...
Unpacking liblwres160:amd64 (1:9.11.3+dfsg-1ubuntu1.5) over (1:9.11.3+dfsg-1ubuntu1.3) ...
Preparing to unpack .../11-ghostscript-x_9.26~dfsg+0-0ubuntu0.18.04.6_amd64.deb ...
Unpacking ghostscript-x (9.26~dfsg+0-0ubuntu0.18.04.6) over (9.26~dfsg+0-0ubuntu0.18.04.4) ...
Preparing to unpack .../12-ghostscript_9.26~dfsg+0-0ubuntu0.18.04.6_amd64.deb ...
Unpacking ghostscript (9.26~dfsg+0-0ubuntu0.18.04.6) over (9.26~dfsg+0-0ubuntu0.18.04.4) ...
Preparing to unpack .../13-libgs9_9.26~dfsg+0-0ubuntu0.18.04.6_amd64.deb ...
Unpacking libgs9:amd64 (9.26~dfsg+0-0ubuntu0.18.04.6) over (9.26~dfsg+0-0ubuntu0.18.04.4) ...
Preparing to unpack .../14-libgs9-common_9.26~dfsg+0-0ubuntu0.18.04.6_all.deb ...
Unpacking libgs9-common (9.26~dfsg+0-0ubuntu0.18.04.6) over (9.26~dfsg+0-0ubuntu0.18.04.4) ...
Setting up libgs9-common (9.26~dfsg+0-0ubuntu0.18.04.6) ...
Setting up libisc169:amd64 (1:9.11.3+dfsg-1ubuntu1.5) ...
Setting up libisccc160:amd64 (1:9.11.3+dfsg-1ubuntu1.5) ...
Setting up libisc-export169:amd64 (1:9.11.3+dfsg-1ubuntu1.5) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Setting up libdns-export1100 (1:9.11.3+dfsg-1ubuntu1.5) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Setting up libdns1100:amd64 (1:9.11.3+dfsg-1ubuntu1.5) ...
Setting up libgs9:amd64 (9.26~dfsg+0-0ubuntu0.18.04.6) ...
Setting up liblwres160:amd64 (1:9.11.3+dfsg-1ubuntu1.5) ...
Setting up ghostscript (9.26~dfsg+0-0ubuntu0.18.04.6) ...
Setting up libisccfg160:amd64 (1:9.11.3+dfsg-1ubuntu1.5) ...
Setting up ghostscript-x (9.26~dfsg+0-0ubuntu0.18.04.6) ...
Setting up libirs160:amd64 (1:9.11.3+dfsg-1ubuntu1.5) ...
Setting up libbind9-160:amd64 (1:9.11.3+dfsg-1ubuntu1.5) ...
Setting up bind9-host (1:9.11.3+dfsg-1ubuntu1.5) ...
Setting up dnsutils (1:9.11.3+dfsg-1ubuntu1.5) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...

kraj, 2019-02-25:
Can you use yoe distro ?
https://github.com/YoeDistro/yoe-distro
and see if it works

pino-kim, 2019-02-28:
@kraj 
Oh, my bad late reply.
I will check it and make comment soon.

pino-kim, 2019-03-01:
@kraj
I update my ubuntu package and head to new commit all OE git.
And Issue not seem to happen. and bblayer.conf  filled path correctly
So close it.

[yoe-distro]
apt-get update
apt-get upgrade
git clone --recurse-submodules -j8 -b master git://github.com/YoeDistro/yoe-distro.git yoe
cd yoe
. ./qemuriscv64-envsetup.sh 
yoe_setup
bitbake-layers

[meta-riscv]
mkdir riscv-yocto-test && cd riscv-yocto-test
repo init -u git://github.com/riscv/meta-riscv  -b master -m tools/manifests/riscv-yocto.xml
repo sync
repo start work --all
. ./meta-riscv/setup.sh 
bitbake-layers

closed 2019-03-01 by pino-kim