riscv/meta-riscv #108: Qemu build fails on Debian-RiscV host

tmagik, 2019-03-07:
**Steps to reproduce the issue:**
1 run bitbake core-image-sato on a RiscV debian host

**Describe the results you received:**
```
ERROR: qemu-native-3.1.0-r0 do_configure: Function failed: do_configure (log file is located at /home/troyb/yocto/build/tmp-glibc/work/riscv64-linux/qemu-native/3.1.0-r0/temp/log.do_configure.7604)
ERROR: Logfile of failure stored in: /home/troyb/yocto/build/tmp-glibc/work/riscv64-linux/qemu-native/3.1.0-r0/temp/log.do_configure.7604
Log data follows:
| DEBUG: Executing shell function do_configure
|
| ERROR: Unsupported CPU = riscv64, try --enable-tcg-interpreter
|
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/troyb/yocto/build/tmp-glibc/work/riscv64-linux/qemu-native/3.1.0-r0/temp/log.do_configure.7604)
ERROR: Task (/home/troyb/yocto/openembedded-core/meta/recipes-devtools/qemu/qemu-native_3.1.0.bb:do_configure) failed with exit code '1'
```

alistair23, 2019-03-07:
Are you trying to build QEMU for a RISC-V host?

alistair23, 2019-03-07:
I just re read your issue. That's not supported in QEMU 3.1, you will need to use master QEMU or QEMU 4.0 when it's released.

kraj, 2019-03-07:
@alistair23 we can add a master recipe to meta-riscv if someone contributes

alistair23, 2019-03-07:
We can, we would just have to point to the QEMU git repo and pull in the latest from master. Probably just worth using a bbappend to write over the current QEMU recipe. I probably won't get around to this though.

For reference the release will be next month: https://wiki.qemu.org/Planning/4.0

kraj, 2019-03-07:
Yeah maybe @tmagik can do that override in a bbappend in private layer

alistair23, 2019-05-04:
QEMU 4.0 has been added to OE. You can now run QEMU on a RISC-V host.

kraj, 2019-07-08:
> QEMU 4.0 has been added to OE. You can now run QEMU on a RISC-V host.

closing

closed 2019-07-08 by kraj