riscv/meta-riscv #112: bitbake -ctestimage fails for qemuriscv64

kraj, 2019-03-08:
When running qemuriscv64 tests ptest fails to detect that system has booted, it open another backdoor serial to get that information but fails to communicate
add following in `local.conf`
```
IMAGE_CLASSES += "testimage testsdk"

TESTIMAGE_AUTO_qemuall = "1"
TEST_TARGET_IP_qemuall = "192.168.7.2"
TEST_QEMUBOOT_TIMEOUT = "50"
```
then run `bitbake core-image-base`

```
DEBUG: Not using kvm for runqemu
DEBUG: Created listening socket for qemu serial console on: 127.0.0.1:35303
DEBUG: Created listening socket for qemu serial console on: 127.0.0.1:36829
DEBUG: launchcmd=runqemu snapshot   nographic qemuriscv64 /mnt/a/yoe/build/tmp/deploy/images/qemuriscv64/core-image-base-qemuriscv64.ext4 tcpserial=36829 bootparams="console=tty1 console=ttyS0,115200n8 printk.time=1" qemuparams="-serial tcp:127.0.0.1:35303 -pidfile pidfile_14851"
DEBUG: runqemu started, pid is 14856
DEBUG: waiting at most 120 seconds for qemu pid (03/07/19 20:59:18)
DEBUG: qemu started in 0.5011918544769287 seconds - qemu procces pid is 14883 (03/07/19 20:59:18)
DEBUG: qemu cmdline used:
/mnt/a/yoe/build/tmp/work/x86_64-linux/qemu-helper-native/1.0-r1/recipe-sysroot-native/usr/bin/qemu-system-riscv64 -device virtio-net-device,netdev=net0,mac=52:54:00:12:34:02 -netdev tap,id=net0,ifname=tap0,script=no,downscript=no -drive id=disk0,file=/mnt/a/yoe/build/tmp/deploy/images/qemuriscv64/core-image-base-qemuriscv64.ext4,if=none,format=raw -device virtio-blk-device,drive=disk0 -show-cursor -monitor null -device loader,file=/mnt/a/yoe/build/tmp/deploy/images/qemuriscv64/Image,addr=0x80200000 -nographic -machine virt -m 512 -device virtio-serial-device -chardev socket,id=virtcon,port=36829,host=127.0.0.1 -device virtconsole,chardev=virtcon -serial tcp:127.0.0.1:35303 -pidfile pidfile_14851 -snapshot -kernel /mnt/a/yoe/build/tmp/deploy/images/qemuriscv64/fw_jump.elf -append root=/dev/vda rw highres=off  console=ttyS0 mem=512M ip=192.168.7.2::192.168.7.1:255.255.255.0 earlycon=sbi console=tty1 console=ttyS0,115200n8 printk.time=1
DEBUG: Target IP: 192.168.7.2
DEBUG: Server IP: 192.168.7.1
DEBUG: Starting logging thread
DEBUG: Starting thread event loop
DEBUG: Connection request received
DEBUG: Setting connection established event
DEBUG: Output from runqemu:
runqemu - INFO - Continuing with the following parameters:

runqemu - INFO - Setting up tap interface under sudo
runqemu - INFO - Network configuration: 192.168.7.2::192.168.7.1:255.255.255.0
runqemu - INFO - Running /mnt/a/yoe/build/tmp/work/x86_64-linux/qemu-helper-native/1.0-r1/recipe-sysroot-native/usr/bin/qemu-system-riscv64 -device virtio-net-device,netdev=net0,mac=52:54:00:12:34:02 -netdev tap,id=net0,ifname=tap0,script=no,downscript=no -drive id=disk0,file=/mnt/a/yoe/build/tmp/deploy/images/qemuriscv64/core-image-base-qemuriscv64.ext4,if=none,format=raw -device virtio-blk-device,drive=disk0 -show-cursor -monitor null -device loader,file=/mnt/a/yoe/build/tmp/deploy/images/qemuriscv64/Image,addr=0x80200000  -nographic -machine virt  -m 512 -device virtio-serial-device -chardev socket,id=virtcon,port=36829,host=127.0.0.1 -device virtconsole,chardev=virtcon  -serial tcp:127.0.0.1:35303 -pidfile pidfile_14851 -snapshot -kernel /mnt/a/yoe/build/tmp/deploy/images/qemuriscv64/fw_jump.elf -append 'root=/dev/vda rw highres=off  console=ttyS0 mem=512M ip=192.168.7.2::192.168.7.1:255.255.255.0 earlycon=sbi console=tty1 console=ttyS0,115200n8 printk.time=1'

DEBUG: Waiting at most 50 seconds for login banner (03/07/19 20:59:18)
DEBUG: Connection from 127.0.0.1:34768
DEBUG: Target didn't reach login banner in 50 seconds (03/07/19 21:00:08)
DEBUG: Last 25 lines of text:

[    6.531979] random: dbus-uuidgen: uninitialized urandom read (12 bytes read)
[    6.532810] random: dbus-uuidgen: uninitialized urandom read (8 bytes read)
Could not rename /var/lib/dbus/machine-id.MaM0qYMB to /var/lib/dbus/machine-id: Function not implemented

Starting system message bus: dbus.

Starting rpcbind daemon...done.

hwclock: Cannot access the Hardware Clock via any known method.

hwclock: Use the --verbose option to see the details of our search for an access method.

Starting syslogd/klogd: done

 * Starting Avahi mDNS/DNS-SD Daemon: avahi-daemon

[A[74G[ ok ]

umount: /mnt/.psplash: not mounted.

Yoe Linux 2.6 qemuriscv64 ttyS0

qemuriscv64 login:
DEBUG: Check full boot log: /mnt/a/yoe/build/tmp/work/qemuriscv64-yoe-linux-musl/core-image-base/1.0-r0/testimage/qemu_boot_log.20190308045351
WARNING: Qemu ended unexpectedly, dump data from host is in /mnt/a/yoe/build/tmp/log/runtime-hostdump/201903072100_qemu
DEBUG: Stopping logging thread
DEBUG: Stop event received
DEBUG: Tearing down logging thread
DEBUG: Sending SIGTERM to runqemu
ERROR: testimage received SIGTERM, shutting down...
ERROR: core-image-base - FAILED - tests were interrupted during execution
ERROR:
DEBUG: Python function do_testimage finished
ERROR: Function failed: do_testimage

```

kraj, 2019-03-08:
@alistair23 this the autotest issue that we need to solve soon

alistair23, 2019-03-08:
Can you attach the full boot log?

pino-kim, 2019-03-08:
Dear @kraj  @alistair23 

To fllowing your issue.
I under stand top comment attach below line at local.conf (yepp I mean auto.conf)
And just run `bitbake core-image-base` ?

>IMAGE_CLASSES += "testimage testsdk"
TESTIMAGE_AUTO_qemuall = "1"
TEST_TARGET_IP_qemuall = "192.168.7.2"
TEST_QEMUBOOT_TIMEOUT = "50"

alistair23, 2019-03-13:
This works for me, at least it does detect the serial prompt

alistair23, 2019-03-13:
This is the error I see (seems fair as I don't have GTK in the guest):

WARNING: core-image-base-1.0-r0 do_testimage: Not using kvm for runqemu
WARNING: core-image-base-1.0-r0 do_testimage: Created listening socket for qemu serial console on: 127.0.0.1:35975
WARNING: core-image-base-1.0-r0 do_testimage: Created listening socket for qemu serial console on: 127.0.0.1:45141
WARNING: core-image-base-1.0-r0 do_testimage: launchcmd=runqemu snapshot   qemuriscv64 /scratch/alistair/yocto/oe-master/build/tmp-glibc/deploy/images/qemuriscv64/core-image-base-qemuriscv64.ext4 tcpserial=45141 bootparams="console=tty1 console=ttyS0,115200n8 printk.time=1" qemuparams="-serial tcp:127.0.0.1:35975 -pidfile pidfile_8921"
WARNING: core-image-base-1.0-r0 do_testimage: runqemu started, pid is 8924
WARNING: core-image-base-1.0-r0 do_testimage: waiting at most 120 seconds for qemu pid (03/13/19 14:30:25)
WARNING: core-image-base-1.0-r0 do_testimage: qemu started in 0.5011353492736816 seconds - qemu procces pid is 8927 (03/13/19 14:30:25)
WARNING: core-image-base-1.0-r0 do_testimage: qemu cmdline used:
/scratch/alistair/yocto/oe-master/build/tmp-glibc/work/x86_64-linux/qemu-helper-native/1.0-r1/recipe-sysroot-native/usr/bin/qemu-system-riscv64 -device virtio-net-device,netdev=net0,mac=52:54:00:12:34:02 -netdev tap,id=net0,ifname=tap0,script=no,downscript=no -drive id=disk0,file=/scratch/alistair/yocto/oe-master/build/tmp-glibc/deploy/images/qemuriscv64/core-image-base-qemuriscv64.ext4,if=none,format=raw -device virtio-blk-device,drive=disk0 -show-cursor -monitor null -device loader,file=/scratch/alistair/yocto/oe-master/build/tmp-glibc/deploy/images/qemuriscv64/Image,addr=0x80200000 -machine virt -m 512 -device virtio-serial-device -chardev socket,id=virtcon,port=45141,host=127.0.0.1 -device virtconsole,chardev=virtcon -serial tcp:127.0.0.1:35975 -pidfile pidfile_8921 -snapshot -kernel /scratch/alistair/yocto/oe-master/build/tmp-glibc/deploy/images/qemuriscv64/fw_jump.elf -append root=/dev/vda rw highres=off  mem=512M ip=192.168.7.2::192.168.7.1:255.255.255.0 earlycon=sbi console=tty1 console=ttyS0,115200n8 printk.time=1 
WARNING: core-image-base-1.0-r0 do_testimage: Target IP: 192.168.7.2
WARNING: core-image-base-1.0-r0 do_testimage: Server IP: 192.168.7.1
WARNING: core-image-base-1.0-r0 do_testimage: Starting logging thread
WARNING: core-image-base-1.0-r0 do_testimage: Starting thread event loop
WARNING: core-image-base-1.0-r0 do_testimage: Connection request received
WARNING: core-image-base-1.0-r0 do_testimage: Setting connection established event
WARNING: core-image-base-1.0-r0 do_testimage: Output from runqemu:
runqemu - INFO - Continuing with the following parameters:

runqemu - INFO - Using preconfigured tap device tap0
runqemu - INFO - If this is not intended, touch /tmp/qemu-tap-locks/tap0.skip to make runqemu skip tap0.
runqemu - INFO - Network configuration: 192.168.7.2::192.168.7.1:255.255.255.0
runqemu - INFO - Running /scratch/alistair/yocto/oe-master/build/tmp-glibc/work/x86_64-linux/qemu-helper-native/1.0-r1/recipe-sysroot-native/usr/bin/qemu-system-riscv64 -device virtio-net-device,netdev=net0,mac=52:54:00:12:34:02 -netdev tap,id=net0,ifname=tap0,script=no,downscript=no -drive id=disk0,file=/scratch/alistair/yocto/oe-master/build/tmp-glibc/deploy/images/qemuriscv64/core-image-base-qemuriscv64.ext4,if=none,format=raw -device virtio-blk-device,drive=disk0 -show-cursor -monitor null -device loader,file=/scratch/alistair/yocto/oe-master/build/tmp-glibc/deploy/images/qemuriscv64/Image,addr=0x80200000  -machine virt  -m 512 -device virtio-serial-device -chardev socket,id=virtcon,port=45141,host=127.0.0.1 -device virtconsole,chardev=virtcon  -serial tcp:127.0.0.1:35975 -pidfile pidfile_8921 -snapshot -kernel /scratch/alistair/yocto/oe-master/build/tmp-glibc/deploy/images/qemuriscv64/fw_jump.elf -append 'root=/dev/vda rw highres=off  mem=512M ip=192.168.7.2::192.168.7.1:255.255.255.0 earlycon=sbi console=tty1 console=ttyS0,115200n8 printk.time=1'

WARNING: core-image-base-1.0-r0 do_testimage: Waiting at most 50 seconds for login banner (03/13/19 14:30:25)
WARNING: core-image-base-1.0-r0 do_testimage: Connection from 127.0.0.1:54668
WARNING: core-image-base-1.0-r0 do_testimage: Reached login banner in 29.547985553741455 seconds (03/13/19 14:30:55)
WARNING: core-image-base-1.0-r0 do_testimage: Logged as root in serial console
Not starting HTTPService for directory tmp-glibc/deploy/deb/all which doesn't exist
Test requires apt to be installed
Stopped HTTPService on 192.168.7.1:0
Traceback (most recent call last):
  File "/scratch/alistair/yocto/oe-master/meta/lib/oeqa/core/decorator/__init__.py", line 32, in wrapped_f
    return func(*args, **kwargs)
  File "/scratch/alistair/yocto/oe-master/meta/lib/oeqa/core/decorator/__init__.py", line 32, in wrapped_f
    return func(*args, **kwargs)
  File "/scratch/alistair/yocto/oe-master/meta/lib/oeqa/core/decorator/__init__.py", line 32, in wrapped_f
    return func(*args, **kwargs)
  [Previous line repeated 2 more times]
  File "/scratch/alistair/yocto/oe-master/meta/lib/oeqa/runtime/cases/buildgalculator.py", line 28, in test_galculator
    self.project.run_configure()
  File "/scratch/alistair/yocto/oe-master/meta/lib/oeqa/utils/buildproject.py", line 49, in run_configure
    return self._run('cd %s; gnu-configize; %s ./configure %s' % (self.targetdir, extra_cmds, configure_args))
[... cut ...]
ERROR: Logfile of failure stored in: /scratch/alistair/yocto/oe-master/build/tmp-glibc/work/qemuriscv64-oe-linux/core-image-base/1.0-r0/temp/log.do_testimage.8921
ERROR: Task (/scratch/alistair/yocto/oe-master/meta/recipes-core/images/core-image-base.bb:do_testimage) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4019 tasks of which 3822 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /scratch/alistair/yocto/oe-master/meta/recipes-core/images/core-image-base.bb:do_testimage
Summary: There were 25 WARNING messages shown.
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.

alistair23, 2019-03-15:
I have merged some improvements. Let me know if you still see this issue. It now works for me with and without the meta-riscv layer (with some extra patches)

kraj, 2019-03-15:
With the latest patchset, this has started to work, there are failures but thats a different issue

closed 2019-03-15 by kraj