riscv/meta-riscv #113: mariadb has been upgraded

tuanha-viosoft, 2019-03-08:
**Description**
mariadb in meta-openembedded has been upgrade to version 10.3.13. It makes mariadb_5%.bbappend in rmeta-riscv failed.

**Steps to reproduce the issue:**
1. $ bitbake mariadb

**Describe the results you received:**
ERROR: No recipes available for:
  /home/ubuntu/riscv-yocto/meta-riscv/recipes-dbs/mariadb/mariadb_5%.bbappend

**Describe the results you expected:**
No error

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

pino-kim, 2019-03-08:
Dear @tuanha-viosoft
Join your issue.
Are you wish to use mariadb v10 or mariadb v5?

pino-kim, 2019-03-08:
Dear @tuanha-viosoft
Join your issue. 

I run bitbake "mariadb right now"
Can you share your build env?

Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64 littleendian"
meta                 = "work:eb12c672c8f9a0d85a38e84cf0f6aa000bb1cf9f"
meta-oe
meta-multimedia
meta-python
meta-networking      = "work:70ccd088eac1ebf45d53b2d0fc9ac211a3905ade"
meta-riscv           = "work:eac35fc45a6a475c9ba5dc85f01b4ab077c5722f"

kraj, 2019-03-08:
delete `meta-riscv/recipes-dbs/mariadb/mariadb_5%.bbappend`

pino-kim, 2019-03-08:
@tuanha-viosoft, @kraj
In my case. `bitbake MariaDB` fetch MariaDB-5.5.62 and build success.
Did you manded meta-openembedded's MariaDB  5.?.? to 10.3.13?
If not can you give me more information to get your error log?

tuanha-viosoft, 2019-03-08:
You should checkout the newest commit in master-next branch then try again.

pino-kim, 2019-03-08:
@tuanha-viosoft 
Thanks to reply I make PR for this issue.
#115

kraj, 2019-03-08:
merged

closed 2019-03-08 by kraj