riscv/meta-riscv #114: Support golang

tuanha-viosoft, 2019-03-08:
Can you consider to support golang for riscv?

pino-kim, 2019-03-08:
@tuanha-viosoft 
I'm interesting your issue.
By the way. current upstreaming go only support riscv64 versoion 1.11?

tuanha-viosoft, 2019-03-09:
@pino-kim The upstream golang doesn’t support riscv. I found a porting golang for riscv here
https://github.com/riscv/riscv-go

pino-kim, 2019-03-09:
@tuanha-viosoft 
Thanks to reply. yepp you right.
I ask  some Question to golang. 
https://github.com/golang/go/issues/30697

kraj, 2019-03-11:
I am certainly open to have bbappend to point to risvc fork.

alistair23, 2019-09-23:
Just an update, it looks like RISC-V support is going into go-lang: https://github.com/golang/go/issues/27532

4a6f656c, 2019-10-04:
A version of Go that supports RISC-V (riscv64) can be found at https://github.com/4a6f656c/riscv-go - this repo contains branches with Go 1.13 and Go tip (along with older Go releases). I'm in the process of getting this merged upstream, which will hopefully mean it becomes available in the Go 1.14 release.

PR #203 (2020-01-26): go: Add recipes to support go on riscv64
Alter the SRC_URIs to fetch 1.13 from the right fork.

Fixes #114
commit 301946f3 (2020-01-26): go: Add recipes to support go on riscv64

Alter the SRC_URIs to fetch 1.13 from the right fork.
Rename refreshed 0002-cmd-go-make-content-based-hash-generation-less-pedan.patch
so it does not conflict with oe-core go recipes

Fixes #114
closed 2020-01-26 by kraj