riscv/meta-riscv #117: baremetal gcc-cross-canadian fails to build

alistair23, 2019-03-14:
**Description**

Unable to build a baremetal GCC toolchain, this error is seen:
The directory that should contain system headers does not exist:
  /scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/recipe-sysroot/usr/include/

**Steps to reproduce the issue:**
MACHINE=baremetal-riscv64 bitbake meta-toolchain

**Describe the results you received:**

ERROR: gcc-cross-canadian-riscv64-8.3.0-r0 do_compile: oe_runmake failed
ERROR: gcc-cross-canadian-riscv64-8.3.0-r0 do_compile: Function failed: do_compile (log file is located at /scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/temp/log.do_compile.26727)
ERROR: Logfile of failure stored in: /scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/temp/log.do_compile.26727
Log data follows:
| DEBUG: SITE files ['endian-little', 'common-linux', 'common-glibc', 'bit-64', 'x86_64-linux', 'common']
| DEBUG: Executing shell function do_compile
| NOTE: make -j 32 all-host configure-target-libgcc
| make[1]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/libiberty'
| make[1]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/intl'
| make[1]: Nothing to be done for 'all'.
| make[1]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/intl'
| make[1]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/build-x86_64-linux/libiberty'
| make[2]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/libiberty/testsuite'
| make[2]: Nothing to be done for 'all'.
| make[2]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/libiberty/testsuite'
| make[1]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/libbacktrace'
| make[1]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/libiberty'
| make  all-am
| make[2]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/libbacktrace'
| make[2]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/build-x86_64-linux/libiberty/testsuite'
| make[2]: Nothing to be done for 'all'.
| make[2]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/build-x86_64-linux/libiberty/testsuite'
| true  DO=all multi-do # make
| make[1]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/libdecnumber'
| make[1]: Nothing to be done for 'all'.
| make[1]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/libdecnumber'
| make[1]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/build-x86_64-linux/libiberty'
| make[2]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/libbacktrace'
| make[1]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/libbacktrace'
| make[1]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/fixincludes'
| make[1]: Nothing to be done for 'all'.
| make[1]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/fixincludes'
| make[1]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/lto-plugin'
| make  all-am
| make[2]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/lto-plugin'
| make[1]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/build-x86_64-linux/fixincludes'
| make[1]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/build-x86_64-linux/libcpp'
| test -f config.h || (rm -f stamp-h1 && make stamp-h1)
| make[1]: Nothing to be done for 'all'.
| make[1]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/build-x86_64-linux/fixincludes'
| make[2]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/lto-plugin'
| make[1]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/lto-plugin'
| make[1]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/build-x86_64-linux/libcpp'
| make[1]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/libcpp'
| test -f config.h || (rm -f stamp-h1 && make stamp-h1)
| make[1]: Leaving directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/libcpp'
| make[1]: Entering directory '/scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/gcc-8.3.0/build.x86_64-oesdk-linux.riscv64-oe-elf/gcc'
| rm -rf include-fixed; mkdir include-fixed
| chmod a+rx include-fixed
| if [ -d ../prev-gcc ]; then \
|   cd ../prev-gcc && \
|   make real-install-headers-tar DESTDIR=`pwd`/../gcc/ \
|     libsubdir=. ; \
| else \
|   set -e; for ml in `cat fixinc_list`; do \
|     sysroot_headers_suffix=`echo ${ml} | sed -e 's/;.*$//'`; \
|     multi_dir=`echo ${ml} | sed -e 's/^[^;]*;//'`; \
|     fix_dir=include-fixed${multi_dir}; \
[... cut ...]
| ERROR: Function failed: do_compile (log file is located at /scratch/alistair/yocto/oe-master/build/tmp-baremetal/work/x86_64-nativesdk-oesdk-linux/gcc-cross-canadian-riscv64/8.3.0-r0/temp/log.do_compile.26727)
ERROR: Task (/scratch/alistair/yocto/oe-master/meta/recipes-devtools/gcc/gcc-cross-canadian_8.3.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1429 tasks of which 1426 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /scratch/alistair/yocto/oe-master/meta/recipes-devtools/gcc/gcc-cross-canadian_8.3.bb:do_compile
Summary: There were 2 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

kraj, 2019-03-15:
Can you try with gcc 9 ?

kraj, 2019-03-15:
I think I see the problem, its actually OE-Core issue.

kraj, 2019-03-15:
Please test this [OE-Core patch](https://patchwork.openembedded.org/patch/159592/)

alistair23, 2019-03-15:
Yep, that fixes the compile failure for me

alistair23, 2019-04-11:
Reopening the issue as the patch has been reverted

kraj, 2019-04-11:
Please test https://patchwork.openembedded.org/patch/160382/

alistair23, 2019-04-15:
Yep, it is fixed now.

closed 2019-04-15 by alistair23