riscv/meta-riscv #121: freedom-u540 unleashed board root password and ping

santvasu, 2019-03-20:
**Description**
**freedom-u540 unleashed board root password** 

I have followed the steps mentioned in the meta-riscv github page ..
After 
**`. ./meta-riscv/setup.sh`**
i edited `conf/local.conf` and updated `MACHINE=freedom-u540`
and built the image
**`bitbake core-image-full-cmdline`**

After the above steps
I have generated the SD Card image as mentioned in "Sparse Image Creation" and 
wrote the image to SD Card

After this the freedom-u540 board was booted with the SD Card and i got the login prompt
(via putty)

**`OpenEmbedded nodistro.0 freedom-u540 ttySIF0 `
`freedom-u540 login: root`
`Login incorrect`**

Can you pl tell what could be mistake i am doing.

Further when i mounted the SD Card (/dev/sdc2) on the host system. i saw that
**`etc/shadow`  does not have password `root::17974:0:99999:7:::`**

I am also not able to login via ssh (no IP addr has been assigned)
Does the generated image had `dhcp` enabled?

(earlier i was able to boot the board with networking ( as described in the  SiFive manual ) 

any help will be appreciated and thanks for the meta-riscv project. :-)

alistair23, 2019-03-20:
I have seen issues logging in before, I'm not sure what causes it though. You should be able to fix it by setting a root password with this in your local.conf:
INHERIT += "extrausers"
EXTRA_USERS_PARAMS_freedom-u540 = "usermod -P root root;"

alistair23, 2019-03-20:
As you probably have debug-tweaks enabled there is no root password, so what you see on the host is not surprising. It should allow you to login without a password, but there is a bug somewhere and sometimes it does not work.

alistair23, 2019-03-20:
You probably need ipv4 set as a distro feature for udhcpc to work. I'm not sure if that's the default

santvasu, 2019-03-20:
> As you probably have debug-tweaks enabled there is no root password, so what you see on the host is not surprising. It should allow you to login without a password, but there is a bug somewhere and sometimes it does not work.

Thank you for the comments. :-)

Like you mentioned  in `local.conf`, i have 
`EXTRA_IMAGE_FEATURES ?= "debug-tweaks"`

Do i have to disable "debug-tweaks" and add the following or both can co-exist.
`INHERIT += "extrausers"`
`EXTRA_USERS_PARAMS_freedom-u540 = "usermod -P root root;"`

santvasu, 2019-03-20:
> You probably need ipv4 set as a distro feature for udhcpc to work. I'm not sure if that's the default

Thanks again.
Will check if udhcpc is enabled

alistair23, 2019-03-20:
You can have both

santvasu, 2019-03-20:
> You can have both

Thank you very much
Will check and get back..

santvasu, 2019-03-20:
i tied the following changes in local.conf
INHERIT += "extrausers"
EXTRA_USERS_PARAMS_freedom-u540 = "usermod -P root root;"

Some how, the root login (thru the serial line) did not still go thru.

Elsewhere in another mailing list, it was mentioned that about  ROOTFS_POSTPROCESS_COMMAND
So i added the following to 
`.riscv-yocto/openembedded-core/meta/recipes-extended/images/core-image-lsb-sdk.bb`

`# Additional root filesystem processing`
`my_modify_root_passwd() {`
  `sed "s%^root:[^:]*:%root:`openssl passwd root`:%" \`
      `< ${IMAGE_ROOTFS}/etc/shadow \`
     ` > ${IMAGE_ROOTFS}/etc/shadow.new`
 ` mv ${IMAGE_ROOTFS}/etc/shadow.new ${IMAGE_ROOTFS}/etc/shadow`
`}`

`ROOTFS_POSTPROCESS_COMMAND += "my_modify_root_passwd;"`

This also did not did not allow for serial login.

However the image was built using 
`bitbake core-image-lsb-sdk`
which had  networking support with sshd, which allowed for ssh access to the board.

Thanks for all the inputs.:)

kraj, 2019-03-20:
@alistair I wonder if the SERIAL_CONSOLES variable is set correctly or not for the machine which matches the latest kernel can you check ?

alistair23, 2019-03-21:
I see the serial out and I can type. I can login as non-root users without any problems, I only see issues with root.

santvasu, 2019-03-21:
> I see the serial out and I can type. I can login as non-root users without any problems, I only see issues with root.

Very true. 
I added a test user (after ssh-ing as root) to Unleashed U540 board.
And then i logged in thru the serial console, and lo, i was able to login.

So it turns out that the tty allocated to the serial console is **`ttySIF0`**
And its  not listed in **`/etc/securetty`**

This was added and now i am able to login as root over serial line.

Guess this can be added to the recipe file (yet to try this)

Thanks for all the inputs

kraj, 2019-03-21:
right, `/etc/securetty`  populated with consoles from SERIAL_CONSOLES,  can you try this pull

https://github.com/riscv/meta-riscv/pull/123

santvasu, 2019-03-21:
> right, `/etc/securetty` populated with consoles from SERIAL_CONSOLES, can you try this pull
> 
> #123

Thanks for the update :-)
trying the change..
will report back

santvasu, 2019-03-21:
> > right, `/etc/securetty` populated with consoles from SERIAL_CONSOLES, can you try this pull
> > #123
> 
> Thanks for the update :-)
> trying the change..
> will report back

Cool.  root login from the `ttySIF0` works 👍 
One query, the pull request suggested, 
-SERIAL_CONSOLE = "115200 console"
+SERIAL_CONSOLES = "115200;ttySIF0"

These is a `;` between 115200 and ttySiF0. Since the previous line did not contain the `;`
the line that was added to the freedom-u540.conf did not contain the `;`
Not sure if it would make a diff, though.

Thanks again for the fix :-)

kraj, 2019-03-21:
SERIAL_CONSOLE has been deprecated read
https://www.yoctoproject.org/docs/current/mega-manual/mega-manual.html#var-SERIAL_CONSOLE

santvasu, 2019-03-21:
> SERIAL_CONSOLE has been deprecated read
> https://www.yoctoproject.org/docs/current/mega-manual/mega-manual.html#var-SERIAL_CONSOLE

Ok, oops , sorry, my bad, did not notice SERIAL_CONSOLE**S**
Thanks for pointing it out. :-)

kraj, 2019-03-21:
please close it if everything in this report is working now

closed 2019-03-21 by santvasu