riscv/meta-riscv #126: opensbi fails on qemuriscv32

futaris, 2019-03-26:
Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv32-oe-linux"
MACHINE              = "qemuriscv32"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv32 littleendian"
meta                 = "work:3da578a79195a63fb6d4bbd68d8f1548defd205f"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      = "work:24dd9328f7ec964bbad25b7061366f9f2ff6194d"
meta-riscv           = "work:ddbc816c35cd5968d95695ac824715eb70962d2f"

undefined reference to `__stack_chk_guard'
undefined reference to `__stack_chk_fail'

errors

kraj, 2019-03-26:
Can you try adding following to `conf/layer.conf` 

```
SECURITY_CFLAGS_riscv32 = ""
SECURITY_CFLAGS_pn-libgcc_riscv32 = ""
GCCPIE_riscv32 = ""
GLIBCPIE_riscv32 = ""
```

kraj, 2019-03-26:
@alistair23 have you tried PIE on riscv32 ?

alistair23, 2019-03-26:
I haven't tried PIE, should I add the above to my local.conf? I can't reproduce this issue at the moment

kraj, 2019-03-26:
yeah some distros eg. Yoe or poky uses pie by default now a days, while OE-core does not. So we should just inoculate ourselves from these differences.

kraj, 2019-03-26:
https://github.com/riscv/meta-riscv/pull/127 should fix it

futaris, 2019-03-27:
#127 fixes it.

$ bitbake core-image-full-cmdline

```
Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv32-oe-linux"
MACHINE              = "qemuriscv32"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv32 littleendian"
meta                 = "work:aeaca9bb1b1c8bf44818945dc4b2cbd6d4b5cef2"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      = "work:24dd9328f7ec964bbad25b7061366f9f2ff6194d"
meta-riscv           = "HEAD:682f56e1b8634f861f63320fcabf5d725bcaccfd"

Initialising tasks: 100% |#######################################| Time: 0:00:01
Sstate summary: Wanted 974 Found 0 Missed 974 Current 114 (0% match, 10% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Tasks Summary: Attempted 3212 tasks of which 1056 didn't need to be rerun and all succeeded.
NOTE: Writing buildhistory
NOTE: Build completion summary:
NOTE:   do_populate_sysroot: 0.0% sstate reuse(0 setscene, 152 scratch)
NOTE:   do_package_qa: 0.0% sstate reuse(0 setscene, 144 scratch)
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 137 scratch)
NOTE:   do_packagedata: 0.0% sstate reuse(0 setscene, 138 scratch)
NOTE:   do_package_write_ipk: 0.0% sstate reuse(0 setscene, 144 scratch)
NOTE:   do_populate_lic: 0.0% sstate reuse(0 setscene, 255 scratch)

```

$ runqemu nographic

```
OpenEmbedded nodistro.0 qemuriscv32 ttyS0

qemuriscv32 login: root
root@qemuriscv32:~# cat /proc/cpuinfo
processor	: 0
hart		: 0
isa		: rv32imafdcu

root@qemuriscv32:~# 
```

closed 2019-03-27 by futaris