riscv/meta-riscv #136: commit 9e781551e58f5cf53f0fc489b2b8bd12cc0eb722 broke qemu build

tkrypton, 2019-04-28:
Hi,

the commit 9e781551e58f5cf53f0fc489b2b8bd12cc0eb722 (that's the latest commit, freedom-u540: Setup u-boot autoboot for TFTP) broke the qemu build.

It took me a while to figure out that a setting for the variable UBOOT_ENV was missing, which got introduced by the above commit. I think adding a default setting for it would solve the issue for other builds as well - I only tried the qemu-riscv64 build. But I don't really know where to add that default setting, I've added it like this:
```
diff --git a/recipes-bsp/u-boot/u-boot_2019.04.bbappend b/recipes-bsp/u-boot/u-boot_2019.04.bbappend
index 9a5a4b0..9ce821b 100644
--- a/recipes-bsp/u-boot/u-boot_2019.04.bbappend
+++ b/recipes-bsp/u-boot/u-boot_2019.04.bbappend
@@ -25,6 +25,7 @@ SRC_URI += "file://0001-riscv-add-infrastructure-for-calling-functions-on-ot.pat
 DEPENDS += "u-boot-tools-native"
 
 UBOOT_ENTRYPOINT = "0x80200000"
+UBOOT_ENV ?= "tftp-boot"
 
 # Overwrite this for your server
 TFTP_SERVER_IP ?= "127.0.0.1"
```
Which fixed the build.

**Steps to reproduce the issue:**
1. bitbake core-image-full-cmdline

**Describe the results you received:**

The error message was: 

```
...
Sstate summary: Wanted 248 Found 0 Missed 248 Current 867 (0% match, 77% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: u-boot-1_2019.04-r0 do_configure: Function failed: do_configure (log file is located at /home/ut/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/u-boot/1_2019.04-r0/temp/log.do_configure.2607)
ERROR: Logfile of failure stored in: /home/ut/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/u-boot/1_2019.04-r0/temp/log.do_configure.2607
Log data follows:
| DEBUG: Executing shell function do_configure
| mkimage: Can't stat /home/ut/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/u-boot/1_2019.04-r0/.txt: No such file or directory
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/ut/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/u-boot/1_2019.04-r0/temp/log.do_configure.2607)
ERROR: Task (/home/ut/riscv-yocto/openembedded-core/meta/recipes-bsp/u-boot/u-boot_2019.04.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2408 tasks of which 2406 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Build completion summary:
NOTE:   do_package_write_ipk: 0.0% sstate reuse(0 setscene, 1 scratch)
```

**Describe the results you expected:**

Build completion

alistair23, 2019-04-28:
Hmm. I haven't noticed any failures but I have extra things in my local.conf which maybe have fixed it for me. I can look at a fix this week, unless you would like to send a PR. We should put a check around the operations in do_configure_prepend(). Unless there is some way to only apply do_configure_prepend() for specific machines

tkrypton, 2019-04-29:
Well, I got it to work, so there's no need to hurry from my point of view. I certainly don't have a local.conf, so your guess may be right.
I think all that's missing is a default for UBOOT_ENV, but I am at a loss where the proper place would be to add that.

alistair23, 2019-05-04:
Does https://github.com/riscv/meta-riscv/pull/137 fix it for you?

pino-kim, 2019-05-04:
HI, @alistair23 , @tkrypton
continue from #138 
I replace ```MACHINE = "qemuriscv64"``` to ```MACHINE = "freedom-u540"``` at auto.conf.   and build it, do ```bitbake core-image-full-cmdline``` and build fine(attach log below). 
But I can't check the image work. because don't have ```freedom-u540``` 
So need help anyone have a board.
thanks.

## meta-riscv diff (from #137)
```diff
diff --git a/recipes-bsp/u-boot/u-boot_2019.04.bbappend b/recipes-bsp/u-boot/u-boot_2019.04.bbappend
index 9a5a4b0..57c85b3 100644
--- a/recipes-bsp/u-boot/u-boot_2019.04.bbappend
+++ b/recipes-bsp/u-boot/u-boot_2019.04.bbappend
@@ -32,8 +32,10 @@ TFTP_SERVER_IP ?= "127.0.0.1"
 do_configure_prepend() {
     sed -i -e 's,@SERVERIP@,${TFTP_SERVER_IP},g' ${S}/include/configs/sifive-fu540.h
 
-    mkimage -A arm -O linux -T script -C none -n "U-Boot boot script" \
-        -d ${WORKDIR}/${UBOOT_ENV}.txt ${WORKDIR}/boot.scr.uimg
+     if [ -f "${WORKDIR}/${UBOOT_ENV}.txt" ]; then
+        mkimage -A arm -O linux -T script -C none -n "U-Boot boot script" \
+            -d ${WORKDIR}/${UBOOT_ENV}.txt ${WORKDIR}/boot.scr.uimg
+    fi
 }
 
 do_deploy_append () {
```

## build log
```text
NOTE: Fetching uninative binary shim from http://downloads.yoctoproject.org/releases/uninative/2.4/x86_64-nativesdk-libc.tar.bz2;sha256sum=06f91685b782f2ccfedf3070b3ba0fe4a5ba2f0766dad5c9d1642dccf95accd0
Initialising tasks: 100% |#################################################################################################################################################################################################################################################| Time: 0:00:02
Sstate summary: Wanted 1115 Found 0 Missed 1115 Current 0 (0% match, 0% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: flex-native-2.6.0-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/flex/flex-2.6.0.tar.bz2, attempting MIRRORS if available
WARNING: xz-native-5.2.4-r0 do_fetch: Failed to fetch URL http://tukaani.org/xz/xz-5.2.4.tar.gz, attempting MIRRORS if available
WARNING: zlib-native-1.2.11-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/libpng/zlib/1.2.11/zlib-1.2.11.tar.xz, attempting MIRRORS if available
WARNING: libtirpc-native-1.1.4-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/libtirpc/libtirpc-1.1.4.tar.bz2, attempting MIRRORS if available
WARNING: expat-native-2.2.6-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/expat/expat-2.2.6.tar.bz2, attempting MIRRORS if available
WARNING: cracklib-native-2.9.5-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/cracklib/cracklib-2.9.5.tar.gz, attempting MIRRORS if available
WARNING: re2c-native-1.0.1-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/re2c/re2c-1.0.1.tar.gz, attempting MIRRORS if available
WARNING: unzip-native-1_6.0-r5 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/infozip/UnZip%206.x%20%28latest%29/UnZip%206.0/unzip60.tar.gz, attempting MIRRORS if available
WARNING: quota-4.05-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/project/linuxquota/quota-tools/4.05/quota-4.05.tar.gz, attempting MIRRORS if available
WARNING: procps-3.3.15-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/project/procps-ng/Production/procps-ng-3.3.15.tar.xz, attempting MIRRORS if available
WARNING: cwautomacros-native-20110201-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/cwautomacros.berlios/cwautomacros-20110201.tar.bz2, attempting MIRRORS if available
WARNING: rpcbind-1.2.5-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/rpcbind/rpcbind-1.2.5.tar.bz2, attempting MIRRORS if available
WARNING: zip-3.0-r2 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/infozip/Zip%203.x%20%28latest%29/3.0/zip30.tar.gz, attempting MIRRORS if available
WARNING: swig-native-3.0.12-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/swig/swig-3.0.12.tar.gz, attempting MIRRORS if available
WARNING: gptfdisk-native-1.0.4-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/gptfdisk/1.0.4/gptfdisk-1.0.4.tar.gz, attempting MIRRORS if available
NOTE: Tasks Summary: Attempted 3307 tasks of which 5 didn't need to be rerun and all succeeded.
NOTE: Writing buildhistory
NOTE: Build completion summary:
NOTE:   do_populate_sysroot: 0.0% sstate reuse(0 setscene, 258 scratch)
NOTE:   do_package_qa: 0.0% sstate reuse(0 setscene, 147 scratch)
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 147 scratch)
NOTE:   do_packagedata: 0.0% sstate reuse(0 setscene, 147 scratch)
NOTE:   do_package_write_ipk: 0.0% sstate reuse(0 setscene, 147 scratch)
NOTE:   do_populate_lic: 0.0% sstate reuse(0 setscene, 262 scratch)
```
## Host build env
```text
Build Configuration:
BB_VERSION           = "1.43.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "freedom-u540"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64 littleendian"
meta                 = "work:e79f0db0745625b35dd83483747e486d08eebfde"
meta-oe
meta-multimedia
meta-python
meta-networking      = "work:50108c18e5ca21cc5c014a45b81c66c5b110fc8b"
meta-riscv           = "work:9e781551e58f5cf53f0fc489b2b8bd12cc0eb722"
```

tkrypton, 2019-05-05:
Hi, sorry for the late answer, but I'm away from my machine at home and can't check your fix for some time. Sorry for the inconvenience, CU Uli

alistair23, 2019-05-06:
#137 has been merged and should fix the issue for you.

tkrypton, 2019-05-26:
Hi, yes, everything does run just fine now. Thanks for your patience,

CU,
Uli

closed 2019-05-26 by tkrypton