riscv/meta-riscv #138: [qemuriscv64] u-boot configure fail issue

pino-kim, 2019-05-04:
**Description**
bitbake u-boot fail at do_configure fail
I think @alistair23  send PR to fix it (#137)

**Steps to reproduce the issue:**
1. regular setup as we know.
2.  . ./meta-riscv/setup.sh
3. bitbake core-image-sato

**Describe the results you received:**
```text
ERROR: u-boot-1_2019.04-r0 do_configure: Function failed: do_configure (log file is located at /home/pino/workspace/Opensource/RISCV/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/u-boot/1_2019.04-r0/temp/log.do_configure.30883)
ERROR: Logfile of failure stored in: /home/pino/workspace/Opensource/RISCV/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/u-boot/1_2019.04-r0/temp/log.do_configure.30883
Log data follows:
| DEBUG: Executing shell function do_configure
| mkimage: Can't stat /home/pino/workspace/Opensource/RISCV/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/u-boot/1_2019.04-r0/.txt: No such file or directory
| WARNING: /home/pino/workspace/Opensource/RISCV/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/u-boot/1_2019.04-r0/temp/run.do_configure.30883:1 exit 1 from 'mkimage -A arm -O linux -T script -C none -n "U-Boot boot script" -d /home/pino/workspace/Opensource/RISCV/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/u-boot/1_2019.04-r0/.txt /home/pino/workspace/Opensource/RISCV/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/u-boot/1_2019.04-r0/boot.scr.uimg'
| ERROR: Function failed: do_configure (log file is located at /home/pino/workspace/Opensource/RISCV/riscv-yocto/build/tmp-glibc/work/qemuriscv64-oe-linux/u-boot/1_2019.04-r0/temp/log.do_configure.30883)
ERROR: Task (/home/pino/workspace/Opensource/RISCV/riscv-yocto/openembedded-core/meta/recipes-bsp/u-boot/u-boot_2019.04.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4073 tasks of which 4072 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory

Summary: 1 task failed:
  /home/pino/workspace/Opensource/RISCV/riscv-yocto/openembedded-core/meta/recipes-bsp/u-boot/u-boot_2019.04.bb:do_configure
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

**Describe the results you expected:**
Not expect build fail at u-boot. Maybe? 

**Additional information you deem important (e.g. issue happens only occasionally):**
```text
Build Configuration:
BB_VERSION           = "1.43.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64 littleendian"
meta                 = "work:e79f0db0745625b35dd83483747e486d08eebfde"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      = "work:50108c18e5ca21cc5c014a45b81c66c5b110fc8b"
meta-riscv           = "work:9e781551e58f5cf53f0fc489b2b8bd12cc0eb722"
```

**Additional details (revisions used, host distro, etc.):**
host : ubuntu 18.04 amd 64bit

pino-kim, 2019-05-04:
@alistair23 
#137  work for me.

pino-kim, 2019-05-05:
@alistair23 
Sorry to bother you.
Can you check my diff comment at #137?

alistair23, 2019-05-05:
Yep, I have updated my PR. Let me know if there are still any problems

pino-kim, 2019-05-06:
#137 close this issue.

closed 2019-05-06 by pino-kim