riscv/meta-riscv #143: Update to Linux 5.2

alistair23, 2019-06-07:
This is mostly just a place holder to discuss what is happening with Linux 5.2.

It looks like Linux 5.2 will have a lot of the HiFive Unleased drivers merged, so hopefully we will no longer require patches on top of mainline. Unfortunately as part of the merge process it appears the DTS compatible stings have changed. So it looks like we need to update the device tree.

At the moment the device tree comes from the firmware. In the future the plan is for the device tree source to be in the kernel, but that isn't merged yet. Hopefully that will be in for 5.3.

So we have two options that we can do
1. Build FSBL for the Unleashed using the new device tree (from the kernel) and inform users of how to update FSBL. At the moment though this will break U-Boot as it expects the current device tree strings.
2. Just build a device tree from the kernel and include it as part of the loading process (ideally just load it with U-Boot). This will be straight forward when U-Boot has Unleashed SD card support, until then we might be stuck with including it in OpenSBI.

As both options need patches on top of the 5.2 kernel I am proposing that we just skip the 5.2 kernel and stay with 5.1. We can then re-evaluate updating to 5.3 when hopefully the device tree changes are in. When that happens we can build the device tree with the kernel and include it in U-Boot or OpenSBI.  Maybe we can look at updating FSBL, but I would rather avoid us getting stuck with building it.

If anyone has better ideas or wants to start submitting patches please feel free to. In the meantime the 5.1 kernel seems the best bet for the Unleashed.

palmer-dabbelt, 2019-06-08:
Thanks for dealing with this, I know this long series of DTS flag days is a huge headache.

pino-kim, 2019-06-28:
I understand what you mean, kernel 5.2 still not stable yet and DTS string many changed. So let kernel build DT and load DT latest u-boot or latest OpenSBI(I think you planed version 0.4). 

Make it short, stabled kernel version 5.1 suit for Unleashed until kernel version 5.2 stable to Unleashed.

I ask some Questions about what makes a huge headache in DTS.
Anyone explain it briefly?

alistair23, 2019-06-28:
So this issue has actually ended up out of date. The new plan is to wait for u-boot 2019.07, OpenSBI 0.4 and Linux 5.2. After that the upgrade process should be fairly straightforward.

In the meantime we will stick with Linux 5.1. We need all three projects to be released and then updates in OE-Core of OpenSBI and u-boot. I suspect it will be a month or so until we can sort this all out.

The SiFive guys are probably the best to explain the DTS problems. More of less the bindings that are in the firmware DTS don't match what was accepted upstream by the DTS/kernel maintainers. The upstream kernel bindings don't match the upstream u-boot bindings (being fixed now) so u-boot and the kernel can't use the same DTS. It's a pain, but after this it should be simpler as things are slowly starting to go upstream.

pino-kim, 2019-06-28:
@alistair23 
Thank your comment. It makes me more clear.
So still waiting for kernel and OpenSBI, U-boot is ready to suit Unleashed kernel version 5.2.
That's what I understand your full context.

How about Qemurisv64, 32? I think It's fine to make kernel v5.2 to run it I think ...
The mentioned issue affects to Qemuriscv too? (I think you well know about it.)

And one more, You mean DTS trouble is like this your PR review #147? 
https://github.com/riscv/meta-riscv/pull/147#discussion_r298388358

alistair23, 2019-06-28:
> So still waiting for kernel and OpenSBI, U-boot is ready to suit Unleashed kernel version 5.2.
> That's what I understand your full context.

Yep, that is what my plan is.

> How about Qemurisv64, 32? I think It's fine to make kernel v5.2 to run it I think ...
> The mentioned issue affects to Qemuriscv too? (I think you well know about it.)

I think the QEMU machines should be fine to update to 5.2. Although they also might need to use OpenSBI v0.4. I have not tested them. I was going to hold off and update everything at once as I think that is easier.

> And one more, You mean DTS trouble is like this your PR review #147?
> [#147 (comment)](https://github.com/riscv/meta-riscv/pull/147#discussion_r298388358)

Yes

alistair23, 2019-07-03:
Closing this as the 5.2 update isn't actually that bad.

closed 2019-07-03 by alistair23