riscv/meta-riscv #146: how to setup bitbake environment without rebuilding from scratch

sbz-12d, 2019-06-20:
I have cloned the repository and followed the steps mentioned in README and  i was able to build the risc-v cmdline image. But it seems that in order to setup environment to run bitbake again (like after closing the current terminal and/or after rebooting the system) i should run meta-riscv/setup.sh, which cleans the build dir and builds the image from scratch.

1. Is there anyway to set the bitbake environment for already built image without rebuilding from scratch ?.The readme file only talks about setting up build env to build an image.

2. I have tried "source openembedded-core/oe-init-build-env " but when run 
"bitbake core-image-full-cmdline" , the system logs-out (all the process the current user owns is getting killed). 

3. Also i have notice that running the command "bitbake core-image-lsb-sdk" causes the same error as mentioned in point 2

I am working on 16.04.1-Ubuntu (64bit). I have cloned the repo to the SSD connected to my system. SSD has enough space (~ 200GB), so the error mentioned in point  2 and 4 is not because of free space availability

pino-kim, 2019-06-24:
Hi @sbz-12d 

can I join your issue? 
At first,I have ask some questoion about your issue.
1. I don't never heard about scratch. So can you give me more information scratch???
Is a yocto's tool? Or meaniing something? (I understand 'scratch' meanning base. am i right?)

Answer to you question as I known
1. If remove line 11, in setup.sh, build dir will not removed by script.
https://github.com/riscv/meta-riscv/blob/4157f3e0123e12871fd039acfdaa8a3ee465d8b9/setup.sh#L11

2, 3. I think you should reference auto.conf. when setup.sh run, It well be generated. you can find it /build/conf. Also add meta directory in your bblayer.conf. 

your host build system is look fine to build riscv yocto. I mean enough space.
If you have error log, I recommend you to attach it.

kraj, 2019-06-24:
Can you tell me where is `sstate-cache` directory inside build tree.

sbz-12d, 2019-06-25:
Hi @pino-kim 

Thanks for your reply. 

> 1. I don't never heard about scratch. So can you give me more information scratch???
>    Is a yocto's tool? Or meaniing something? (I understand 'scratch' meanning base. am i right?)
> 
1. "From scratch" in this context means from beginning or from starting. I meant to say that running setup.sh  again starting building the from beginning.

> 1. If remove line 11, in setup.sh, build dir will not removed by script.
>    https://github.com/riscv/meta-riscv/blob/4157f3e0123e12871fd039acfdaa8a3ee465d8b9/setup.sh#L11

2. I will remove " rm -rf $DIR " line from setup.sh and will check it. But currently what i'm doing is, once i have built the image i made a copy of local.conf file  from build/conf/local.conf and whenever i want to setup environment for bitbake, i'm doing "source openembedded-core/oe-init-build-env " and copying the backup local.conf file to build/conf folder (replacing the file inside it.). I was able to run the bitbake and build the image without any problem.

> If you have error log, I recommend you to attach it.
3. There was no error log generated. since system was abruptly logging out by killing all the processes current user owns. i have checked the dmesg as well.
   3.1 please note that the error is getting generated when i  tried to setup bitbake environment to already built image by doing "source openembedded-core/oe-init-build-env " only and running "bitbake core-image-full-cmdline" command.
  3.2 I was also getting the same error when i did . ./setup.sh and did "bitbake core-image-lsb-sdk".

sbz-12d, 2019-06-25:
Hi @kraj 

 Thanks for your  reply

> Can you tell me where is `sstate-cache` directory inside build tree.

It is in build directory only. build/sstate-cache

pino-kim, 2019-06-25:
@sbz-12d 
Thanks to let me understand your full context.

2. I think ...you'd better make some bash script to change local.conf in build/conf when set environment.  

>3. There was no error log generated. since system was abruptly logging out by killing all the processes current user owns. i have checked the dmesg as well.
3.1 please note that the error is getting generated when i tried to setup bitbake environment to already built image by doing "source openembedded-core/oe-init-build-env " only and running "bitbake core-image-full-cmdline" command.
3.2 I was also getting the same error when i did . ./setup.sh and did "bitbake core-image-lsb-sdk".

-> When you encounter your issue.When set up build environment? or while Running bitbake?
Dmesg log point you something wrong? I guess you attach your system by terminal program(protocal ssh). If true, do you use **tmux** or **screen** when set up yocto and run bitbake?

sbz-12d, 2019-07-03:
@pino-kim 

> 1. I think ...you'd better make some bash script to change local.conf in build/conf when set environment.

  As you suggested, by commenting the line "rm -f $DIR" works. I was able to setup environment without rebuilding.

> -> When you encounter your issue.When set up build environment? or while Running bitbake?
> Dmesg log point you something wrong? I guess you attach your system by terminal program(protocal ssh). If true, do you use **tmux** or **screen** when set up yocto and run bitbake?

Now everything works fine. Thanks a lot

kraj, 2019-07-08:
@sbz-12d are there any changes we need to the script to make it better please propose them with a pull request.

kraj, 2019-09-07:
this should be fixed now

closed 2019-09-07 by kraj