riscv/meta-riscv #150: libgfortran is not working

shreyaparmara11, 2019-07-06:
I am trying to bitbake libgfortran and getting below error.

ERROR: Error for /source/apps_proc/poky/meta/recipes-devtools/gcc/libgfortran_5.2.bb, dependency ${COMPILERDEP} does not contain ':' character

I am using recipe from https://github.com/openembedded/openembedded-core/blob/master/meta/recipes-devtools/gcc/libgfortran.inc

Steps to reproduce the issue:
bitbake libgfortran

Describe the results you received
ERROR: Error for /source/apps_proc/poky/meta/recipes-devtools/gcc/libgfortran_5.2.bb, dependency ${COMPILERDEP} does not contain ':' character
. Task 'depends' should be specified in the form 'packagename:task'
ERROR: Command execution failed: Exited with 1

Describe the results you expected:
bitbake Libgfortran successful

I have followed this link https://stackoverflow.com/questions/49344910/error-build-fortran-under-yocto .But no luck for this.

kraj, 2019-07-08:
which branch are you using since gcc 5.2 is pretty old and does not support risc-v out of box, you are better of using latest OE release which uses gcc 9.x and has risc-v support out of box.

shreyaparmara11, 2019-07-18:
thank you.
I will try this.

closed 2019-07-18 by shreyaparmara11