riscv/meta-riscv #151: [Question] About Opensbi0.4 work and some questions

pino-kim, 2019-07-08:
@alistair23  @kraj 
Kernel 5.2 is out and Opensbi0.4 is out too. 
I think I can handle opensbi update(maybe ?) ... so start it.
If work done first to test it qemuriscv32, qemuriscv64 at kernel v5.1 and next v5.2. 
Below is my temp working branch.

https://github.com/pino-kim/openembedded-core/commits/pino_opensbi_0_4

Not work done yet, success lunch it **qemuriscv64** kernel 5.1(linux-yocto)
I have some question to ask. Below is my Qustion

1. Before upstream to OE. Is it better to add bbappend to update opensbi0.4 in meta-riscv?
2.  I never met build this waring message before build v0.3 All three machine need below package to ship it?

`WARNING: opensbi-0.4-r0 do_package: QA Issue: opensbi: Files/directories were installed but not shipped in any package:
  /platform/qemu/virt/firmware/fw_dynamic.elf
  /platform/qemu/virt/firmware/fw_dynamic.bin
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
opensbi: 2 installed and not shipped files. [installed-vs-shipped]`

alistair23, 2019-07-08:
Hello,

Thanks for the interest in updating OpeSBI 0.4, unfortunately I have already sent a patch to OE-Core, which you can see here: http://lists.openembedded.org/pipermail/openembedded-core/2019-July/284488.html

1. No, I think it makes the most sense to just update OpenSBI. bbappends are used for layer specific overrides, in this case OpenSBI should be updated for everyone.
2. Yep, that will need to be fixed. I have installed the new binary in my patch

pino-kim, 2019-07-08:
Thanks to give a information about patch already sent.
I can reference your patch. If I have a another chance next time. 

>No, I think it makes the most sense to just update OpenSBI. bbappends are used for layer specific overrides, in this case OpenSBI should be updated for everyone.

In your context, I understand bbapend just use in layer specific package.
Thanks your  comment and close this issue.

closed 2019-07-08 by pino-kim

alistair23, 2019-07-08:
Thanks for contributing anyway! It's great to have another pair of eyes looking at commits and testing things.