riscv/meta-riscv #156: Sanity check fails for qemuriscv32 machine

palmer-dabbelt, 2019-07-11:
**Description**

I'm trying to build a 32-bit setup and I'm getting an unexpected error message.  I've had this work before but I can't figure out what I'm doing differently this time around.

**Steps to reproduce the issue:**
1. Create a new repo according to the README.
2. Modify auto.conf to target qemuriscv32 instead of qemuriscv64
3. `bitbake core-image-minimal`

**Describe the results you received:**

````
$ bitbake core-image-minimal
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    MACHINE=qemuriscv32 is invalid. Please set a valid MACHINE in your local.conf, environment or other configuration file.

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
````

**Describe the results you expected:**

The build to run.

**Additional information you deem important (e.g. issue happens only occasionally):**

````
$ cat conf/auto.conf 
MACHINE ?= "qemuriscv32"
#IMAGE_FEATURES += "tools-debug"
#IMAGE_FEATURES += "tools-tweaks"
#IMAGE_FEATURES += "dbg-pkgs"
# rootfs for debugging
#IMAGE_GEN_DEBUGFS = "1"
#IMAGE_FSTYPES_DEBUGFS = "tar.gz"
EXTRA_IMAGE_FEATURES_append = " ssh-server-dropbear"
EXTRA_IMAGE_FEATURES_append = " package-management"
PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
USER_CLASSES ?= "buildstats buildhistory buildstats-summary image-mklibs image-prelink"

require conf/distro/include/no-static-libs.inc
require conf/distro/include/yocto-uninative.inc
require conf/distro/include/security_flags.inc

INHERIT += "uninative"

DISTRO_FEATURES_append = " largefile opengl ptest multiarch wayland pam  systemd "
DISTRO_FEATURES_BACKFILL_CONSIDERED += "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
HOSTTOOLS_NONFATAL_append = " ssh"
````

if I change that qemuriscv32 back to qemuriscv64 it starts building, but I need a 32-bit setup to test QEMU.

**Additional details (revisions used, host distro, etc.):**

Gentoo... :)

alistair23, 2019-07-11:
I've never used the script, so I'm not sure what's going on here. What does your bblayers.conf file look like? It looks like you are missing the meta-riscv layer as qemuriscv64 is included in OE-Core but qemuriscv32 isn't.

kraj, 2019-07-11:
this scfript is good for automation in general when you are testing just `meta-riscv` for full distro use something like Yoe Distro (https://github.com/YoeDistro/yoe-distro)

alistair23, 2019-07-11:
I can't reproduce this with a clean repo setup following the README

alistair23, 2019-07-11:
^ With the latest meta-riscv that has Palmers PR merged

pino-kim, 2019-07-11:
me too, in Ubuntu 18.04, meta-riscv qemuriscv32 build run fine.

palmer-dabbelt, 2019-07-11:
It WFM after I pull in my fix, so I'm guessing those exceptions were just garbageing up my setup.

closed 2019-07-11 by palmer-dabbelt

alistair23, 2019-07-11:
That is still a strange error, good job figuring out the fix! I'm still surprised that no else has seen the problem before