riscv/meta-riscv #173: [qemuriscv32/qemuriscv64] opensbi 4.0 compile fail

pino-kim, 2019-08-20:
**Description**

**Steps to reproduce the issue:**
follow README

> mkdir riscv-yocto && cd riscv-yocto
repo init -u git://github.com/riscv/meta-riscv  -b master -m tools/manifests/riscv-yocto.xml
repo sync
repo start work --all
. ./meta-riscv/setup.sh
MACHINE=qemuriscv32 bitbake core-image-full-cmdline
and
MACHINE=qemuriscv64 bitbake core-image-full-cmdline
 
**Describe the results you received:**
opensbi4.0 build fail

**Describe the results you expected:**
success core-image-full-cmdline

**Additional information you deem important (e.g. issue happens only occasionally):**
attach part of build log(be)
I'll attach part of log. 
At night time i will attach full log file at night time. Korea is day time now :-) 

```
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt_ro.c:699: undefined reference to `__stack_chk_fail'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt_ro.c:704: undefined reference to `__stack_chk_guard'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/build/platform/qemu/virt/lib/libplatsbi.a(fdt_ro.o): in function `fdt_getprop':
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt_ro.c:508: undefined reference to `__stack_chk_fail'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/build/platform/qemu/virt/lib/libplatsbi.a(fdt_ro.o): in function `fdt_stringlist_count':
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt_ro.c:732: undefined reference to `__stack_chk_guard'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/build/platform/qemu/virt/lib/libplatsbi.a(fdt_ro.o): in function `fdt_stringlist_search':
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt_ro.c:759: undefined reference to `__stack_chk_fail'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt_ro.c:762: undefined reference to `__stack_chk_guard'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/build/platform/qemu/virt/lib/libplatsbi.a(fdt_ro.o): in function `fdt_stringlist_get':
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt_ro.c:781: undefined reference to `__stack_chk_fail'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt_ro.c:801: undefined reference to `__stack_chk_guard'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/build/platform/qemu/virt/lib/libplatsbi.a(fdt_ro.o): in function `fdt_node_offset_by_compatible':
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt_ro.c:838: undefined reference to `__stack_chk_fail'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/build/platform/qemu/virt/lib/libplatsbi.a(fdt.o): in function `fdt_check_prop_offset_':
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt.c:237: undefined reference to `__stack_chk_guard'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/build/platform/qemu/virt/lib/libplatsbi.a(fdt.o): in function `fdt_next_node':
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt.c:271: undefined reference to `__stack_chk_fail'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt.c:248: undefined reference to `__stack_chk_guard'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/build/platform/qemu/virt/lib/libplatsbi.a(fdt.o): in function `fdt_first_subnode':
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt.c:286: undefined reference to `__stack_chk_fail'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt.c:291: undefined reference to `__stack_chk_guard'
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/../../libexec/riscv64-oe-linux/gcc/riscv64-oe-linux/9.1.0/ld: /home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/build/platform/qemu/virt/lib/libplatsbi.a(fdt.o): in function `fdt_next_subnode':
/home/sw.kim/workspace/opensource_workspace/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/opensbi/0.4-r0/git/lib/utils/libfdt/fdt.c:302: undefined reference to `__stack_chk_fail'
collect2: error: ld returned 1 exit status
[... cut ...]
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64 littleendian"
meta                 = "work:6b36db836547a23f43c5f97bf3706d7b210c209c"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      = "work:509d40153d6c54367abcd023fd45c884a343acab"
meta-riscv           = "work:04e6b2c12e5bcf4b8222a756df394ca31b8afc3f"

alistair23, 2019-08-20:
Someone else pointed the issue out to me today. If you add these lines
```
SECURITY_CFLAGS_pn-opensbi = ""
SECURITY_LDFLAGS_pn-opensbi = ""
```
to `meta/conf/distro/include/security_flags.in` that should fix it.

pino-kim, 2019-08-20:
@alistair23 
Thank's to fast reply. 
I will try it.

kraj, 2019-08-20:
Please cherry-pick https://patchwork.openembedded.org/patch/164123/ until its pushed to core

pino-kim, 2019-08-20:
@kraj 
Thanks, I will try it.

pino-kim, 2019-08-20:
@alistair23, @kraj
the patch is work to me.
Both Machine(qemuriscv32, qemuriscv64) run well.
Thanks.

closed 2019-08-20 by pino-kim

alistair23, 2019-08-20:
Great! I realised that I didn't see this when testing as I don't use a distro, which is where I think the flags come from.