riscv/meta-riscv #177: Unable to build gobject-introspection for HiFive Unleashed

alistair23, 2019-09-04:
Recently I have been seeing this error message when building gobject-introspection for the freedom-u540 machine.

> | /lib/ld-linux-riscv64-lp64d.so.1: No such file or directory
> | If the above error message is about missing .so libraries, then setting up GIR_EXTRA_LIBS_PATH in the recipe should help.
> | (typically like this: GIR_EXTRA_LIBS_PATH="${B}/something/.libs" )
> | Command '['/scratch/alistair/yocto/oe-master/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.60.2-r0/build/g-ir-scanner-qemuwrapper', '/scratch/alistair/yocto/oe-master/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.60.2-r0/build/tmp-introspect9q164ksx/GLib-2.0', '--introspect-dump=/scratch/alistair/yocto/oe-master/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.60.2-r0/build/tmp-introspect9q164ksx/functions.txt,/scratch/alistair/yocto/oe-master/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.60.2-r0/build/tmp-introspect9q164ksx/dump.xml']' returned non-zero exit status 1.
> | ninja: build stopped: subcommand failed.
> | WARNING: /scratch/alistair/yocto/oe-master/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.60.2-r0/temp/run.do_compile.10574:1 exit 1 from 'ninja -v -j 32'

I have tried setting `GIR_EXTRA_LIBS_PATH` in the recipe but that doesn't seem to help. Any ideas?

kraj, 2019-09-04:
Is `/lib/ld-linux-riscv64-lp64d.so.1` in right place ? secondly, check if qemu is crashing that's second reason for such errors

alistair23, 2019-09-04:
I'm using usermerge, so it's under `/usr/lib/ld-linux-riscv64-lp64d.so.1`

I don't think QEMU is crashing, this isn't from the recent QEMU upgrade.

kraj, 2019-09-04:
I see, so whats the ABI-mandated location for ldso on risc-v

kraj, 2019-09-04:
I have opened an issue to discuss this for ELF ABI for RISC-V here https://github.com/riscv/riscv-elf-psabi-doc/issues/114 we can modify OE in one way or other, but we want to do the it per ABI

alistair23, 2019-09-04:
Thanks Khem

kraj, 2019-09-05:
it is most likely a regression I introduced in gcc which I sent a fix for here

https://patchwork.openembedded.org/patch/164650/

Can you cherry-pick this into your oe-core and retry ?

alistair23, 2019-09-05:
Yeah, that fixes it for me :)

kraj, 2019-09-07:
patch in oe-core/gcc is merged now

closed 2019-09-07 by kraj