riscv/meta-riscv #193: ERROR: Nothing PROVIDES 'core-image-lsb-sdk'.

geonheec, 2020-01-13:
Hi, I want to run QEMU with riscv32 architecture.
Additionally, I want that virtual machine to have gdb tool.

I followed the guide at README in this repo.

1.
> mkdir riscv-yocto && cd riscv-yocto
> repo init -u git://github.com/riscv/meta-riscv  -b master -m tools/manifests/riscv-yocto.xml
> repo sync
> repo start work --all

2.
> . ./meta-riscv/setup.sh

3.
> MACHINE=qemuriscv32 bitbake core-image-lsb-sdk

and I got the error message as following:

>   ERROR: Nothing PROVIDES 'core-image-lsb-sdk'. Close matches:
>   core-image-sato-sdk
>   core-image-rt-sdk
>   core-image-base
> 
>   Summary: There was 1 ERROR message shown, returning a non-zero exit code.

The first question is,
**[1] How can I get core-image-lsb-sdk ??**

So, I tried "MACHINE=qemuriscv32 bitbake core-image-base" with EXTRA_IMAGE_FEATURES tools-sdk and tools-debug in local.conf
Then I got the follwoing error message

> ERROR: Task (/home/ghcho/riscv-yocto/openembedded-core/meta/recipes devtools/cmake/cmake-native_3.16.1.bb:do_compile) failed with exit code '1'

The second question is,
**[2] Even if I solved the first question, this error is likely to be occur.
What should I do...?**

I'm using Ubuntu16.04 with 4.15.0-74-generic kernel.

Thank you.

kraj, 2020-01-13:
@geonheec core-image-lsb has been removed from oe-core so this image is not available anymore, however if you want gdb in your image then add

```
EXTRA_IMAGE_FEATURES_append = " tools-debug dbg-pkgs"
```
in your `local.conf` and then perhaps build `core-image-full-cmdline`

cmake-native not building is a different issue perhaps unrelated to riscv32, please post exact error, where it fails. The full logs will be in a file called `log.do_compile` in build area for cmake-native

geonheec, 2020-01-13:
Thanks for answer.

The full log is too long, so I'm not sure how to show the log.
The following message is the ERROR part from the log. If the full log is needed, please tell me.

[image]

kraj, 2020-01-13:
if you can find `/home/ghcho/riscv-yocto/build/tmp-glibc/work/x86_64-linux/cmake-native/3.16.1-r0/temp/log.do_compile` then paste that to some pastebin site.

geonheec, 2020-01-14:
I tried on other machine and it has another problem.
It seems like riscv32 does not support for strace, the following error message is appeared.
`ERROR:` Task (/home/yoda/riscv-yocto/openembedded-core/meta/recipesdevtools/strace/strace_5.4.bb:do_configure) failed with exit code '1'
But I don't need strace, I only need gdb.
How can I configure tools-debug with excepting strace?

kraj, 2020-01-14:
in `openembedded-core/meta/recipes-core/packagegroups/packagegroup-core-tools-debug.bb` delete strace something like below
```diff
 --- a/meta/recipes-core/packagegroups/packagegroup-core-tools-debug.bb
+++ b/meta/recipes-core/packagegroups/packagegroup-core-tools-debug.bb
@@ -16,6 +16,5 @@ MTRACE_libc-glibc = "libc-mtrace"
RDEPENDS_${PN} = "\
gdb \
gdbserver \
-    strace \
${MTRACE} \
"
```

geonheec, 2020-01-14:
Oh, It works well! Thank you so much.

Can I ask one more question?
Is it possible to include objdump in vm??
Which extra image configuration should I add?

kraj, 2020-01-14:
in `local.conf` add
```
IMAGE_INSTALL_append = " binutils"
```

geonheec, 2020-01-14:
Thank you very much. It works fine.
Finally I would like to ask you about the ssh connection....
Sorry for asking so many questions.

I add "hostfwd=tcp::10000-:22" to **_build/tmp-glibc/deploy/images/qemuriscv32/core-image-full-cmdline-qemuriscv32-20200114024648.qemuboot.conf_** file.
But runqemu says hostfwd is invalid option. (redir option as well)

How can I set ssh connection with the virtual machine??

kraj, 2020-01-14:
Just add sshd to your image via `local.conf` add
```
EXTRA_IMAGE_FEATURES_append = " ssh-server-dropbear"
```
then use `runqemu nographic` to boot the image into console

 then `ssh -l root 192.168.7.2` from machine you are running qemu on should work fine

geonheec, 2020-01-14:
Thank you.
But... I cannot execute binay file in virtual machine.
I send the riscv32-compiled binary program to virtual machine via scp.
The program's name is "bomb", and when I try the command "./bomb", the message "No such file or directory" is appeared.
Should I do something more...?

kraj, 2020-01-14:
that means the ABIs are not matching. Can you run `readelf -e ./bomb| grep "Requesting program interpreter"` and see where its pointing for ldso. perhaps its not same as what you see in /lib

Its best to use SDK generate with yocto to build your programs, you can do so by
```
bitbake -cpopulate_sdk <your_image>

```

which will generate a .sh installer file which then you can insgtall on your build host as
cross compiler and use it to compile your app.

read https://himvis.com/clang-based-cross-sdk-with-openembedded-framework/

geonheec, 2020-01-14:
Thank you so much. All works well.

Have a nice day! :)

closed 2020-01-14 by geonheec