riscv/meta-riscv #195: How can host route the ssh connection to riscv32 virtual machine ?

geonheec, 2020-01-14:
I can't find the way to route ssh to riscv32 virtual machine.

I can connect to virtual machine from only local host now.

I want to connect to virtual machine from remote machine.

For example, **machine A** hosts the riscv32 virtual machine and I want to connect to the virtual machine from remote machine **"machine B"** via ssh.

To do that, port routing on machine A seems to be necessary.
So I tried to add _"hostfwd=tcp::10000-:22"_ at 
**_riscv-yocto/build/tmp-glibc/deploy/images/qemuriscv32/core-image-full-cmdline-qemuriscv32-20200114134928.qemuboot.conf_** file.
But runqemu says hostfwd is invalid option. (redir option as well)
(Even if I add sshd to my image via adding _EXTRA_IMAGE_FEATURES_append = " ssh-server-dropbear"_ in _local.conf_ )

In short, the following is what I want.
when I try to connect to port **_x_** on the **host machine A** via ssh , I want the connection to be routed to virtual machine.

I would appreciate it if you could let me know what to do.

kraj, 2020-01-15:
Please use `slirp` option to `runqemu` luckily for qemuriscv machines we have `QB_SLIRP_OPT` already set which will forward port 2222 on host to qemu

```
runqemu nographic slirp
```

The from client machine
```
ssh -p 2222 -l root <host-ip>
```
will get you right into qemu

geonheec, 2020-01-15:
I tried, but it does not work.
Even the ssh connection from host to qemu does not work.

I tried the following command on host machine.
> ssh -l root 192.168.7.2
And It does not work with any output.

So I check the sshd in qemu by the command `systemctl restart sshd` and the following message is appeared. 
> Failed to restart ssh.service: Unit ssh.service not found. 
I already add EXTRA_IMAGE_FEATURES_append = " ssh-server-dropbear" in local.conf. Should I do some more configuration??

About the connection from remote machine, I'll try after fix this problem.

kraj, 2020-01-15:
you should check if dropbear is running in qemu, I guess thats the problem. btw. what image target do you build ? you can also try to replace dropbear with 
```
EXTRA_IMAGE_FEATURES_append = " ssh-server-openssh"
```

to use openSSH instead

geonheec, 2020-01-15:
`MACHINE=qemuriscv32 bitbake core-image-full-cmdline` is the image target.

I checked if dropbear is running in qemu by `ps -ef` but there's no process related with dropbear.
Then I build with ssh-server-openssh, but there's no process related with ssh as well.

geonheec, 2020-01-15:
I tried another solution.
I remove `slirp` option because it makes the ssh connection impossible. (I not sure, but when I remove this option, ssh connection from host machine to virtual machine works well.)

Then I route the _`host machine's port **_x_**`_ to _`192.168.7.2:22`_  
It allow the remote machine to connect to virtual machine by using `ssh -p **_x_** -l <root host-ip>`

+) I have another question, can I change the port from 22 to another without using slirp?

geonheec, 2020-01-15:
I tried to add "hostfwd=tcp::10000-:22" at `~/riscv-yocto/openembedded-core/meta/conf/machine/include/riscv/qemuriscv.inc` file.
But the following message is appeared.
`runqemu - ERROR - Failed to run qemu: qemu-system-riscv32: Invalid parameter 'hostfwd'`

+) Can I make many virtual machine ? The above question is for this reason. I want to make more virtual machine and assign different port to each virtual machine.
+) Is it possible to increase virtual machine's memory over 1GB? I tried to boot 2GB virtual machine, but it has only 1002mb memory.

kraj, 2020-01-15:
interesting, I have tried `runqemu nographic slirp` here with qemu/mips though and it worked all fine and also `ssh -p 2222 -l root <host-ip>` worked. Secondly, overriding should have achieved something similar, so I am not sure whats going on your end.

for increasing memory it is qemu's limitation. On some arches it is 512M on some it is 1G
and some has higher limits. I think for riscv32 its 1G

Since runqemu is primarily used for debugging/development and not scaled deployment
it does not have support to launch multiple instances since it uses same image to run

one probably needs to copy the kernel + image and have a custom scripts to launch and
maybe even make it easy to launch new instance easily.

geonheec, 2020-01-15:
I'll correct what I said above. I have confirmed that slirp works fine.

About launching many vm, I followed your answer and it works well!

Thanks to you, it was a big help. 
I hope you have a nice day.

+ I'll close this issue.

closed 2020-01-15 by geonheec