riscv/meta-riscv #196: ParseError report

pino-kim, 2020-01-22:
**Description**

parse error rep start bitbake

**Steps to reproduce the issue:**
1.  follow doc to setup build meta-riscv
2.  . ./meta-riscv/setup.sh
3.  MACHINE=qemuriscv64 bitbake core-image-full-cmdline

**Describe the results you received:**

$ MACHINE=qemuriscv64 bitbake core-image-full-cmdline
ERROR: ParseError at /home/sw.kim/workspace/opensource_workspace/riscv-yocto/meta-openembedded/meta-python/recipes-devtools/python/python-yappi.inc:10: Could not inherit file classes/setuptools.bbclass

**Describe the results you expected:**
build start? maybe?

**Additional details (revisions used, host distro, etc.):**
I check build error ubuntu 18.04 Host system.

kraj, 2020-01-22:
this is a known problem. You can add meta-python2 to the layer mix and thatt will fix it.

pino-kim, 2020-01-22:
@kraj, 
Thanks to let me know. I will try your comment. ^^.

geonheec, 2020-01-22:
I have same problem.
How can I add meta-python2 ??

kraj, 2020-01-22:
source the script
```
cd ..
git clone git://git.openembedded.org/meta-python2
cd -
bitbake add-layer ../meta-python2

```

geonheec, 2020-01-22:
I tried your suggestion but the same error message is appeared when I tried `bitbake add-layer ../meta-python2`
`ERROR: ParseError at /home/ca2020/riscv-yocto/meta-openembedded/meta-python/recipes-extended/python-cson/python-cson_git.bb:18: Could not inherit file classes/setuptools.bbclass`

kraj, 2020-01-22:
ah yes, just add it manually into conf/bblayers.conf

geonheec, 2020-01-22:
I already tried to manually add meta-python2 into conf/bblayers.conf.
But the similar problem is occurred when I tried `. . /meta-riscv/setup.sh`

```
ca2020@bomblab:~/riscv-yocto$ . ./meta-riscv/setup.sh
Init OE

### Shell environment set up for builds. ###

You can now run 'bitbake <target>'

Common targets are:
    core-image-minimal
    core-image-sato
    meta-toolchain
    meta-ide-support

You can also run generated qemu images with a command like 'runqemu qemux86'.

Other commonly useful commands are:
 - 'devtool' and 'recipetool' handle common recipe tasks
 - 'bitbake-layers' handles common layer tasks
 - 'oe-pkgdata-util' handles common target package tasks
Adding layers
NOTE: Starting bitbake server...
Traceback (most recent call last):
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/bin/bitbake-layers",                 line 95, in <module>
    ret = main()
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/bin/bitbake-layers",                 line 63, in main
    tinfoil.prepare(True)
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/tinfoil.py", l                ine 408, in prepare
    self.run_command('parseConfiguration')
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/tinfoil.py", l                ine 466, in run_command
    raise TinfoilCommandFailed(result[1])
bb.tinfoil.TinfoilCommandFailed: Traceback (most recent call last):
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/command.py", l                ine 74, in runCommand
    result = command_method(self, commandline)
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/command.py", l                ine 275, in parseConfiguration
    command.cooker.parseConfiguration()
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/cooker.py", li                ne 433, in parseConfiguration
    self.handleCollections(self.data.getVar("BBFILE_COLLECTIONS"))
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/cooker.py", li                ne 1225, in handleCollections
    raise CollectionError("Errors during parsing layer configuration")
bb.cooker.CollectionError: Errors during parsing layer configuration

NOTE: Starting bitbake server...
Traceback (most recent call last):
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/bin/bitbake-layers",                 line 95, in <module>
    ret = main()
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/bin/bitbake-layers",                 line 63, in main
    tinfoil.prepare(True)
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/tinfoil.py", l                ine 408, in prepare
    self.run_command('parseConfiguration')
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/tinfoil.py", l                ine 466, in run_command
    raise TinfoilCommandFailed(result[1])
bb.tinfoil.TinfoilCommandFailed: Traceback (most recent call last):
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/command.py", l                ine 74, in runCommand
    result = command_method(self, commandline)
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/command.py", l                ine 275, in parseConfiguration
    command.cooker.parseConfiguration()
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/cooker.py", li                ne 433, in parseConfiguration
    self.handleCollections(self.data.getVar("BBFILE_COLLECTIONS"))
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/cooker.py", li                ne 1225, in handleCollections
    raise CollectionError("Errors during parsing layer configuration")
bb.cooker.CollectionError: Errors during parsing layer configuration

NOTE: Starting bitbake server...
Traceback (most recent call last):
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/bin/bitbake-layers",                 line 95, in <module>
    ret = main()
[... 53 lines cut ...]
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/cooker.py", line 433, in parse
    self.handleCollections(self.data.getVar("BBFILE_COLLECTIONS"))
  File "/home/ca2020/riscv-yocto/openembedded-core/bitbake/lib/bb/cooker.py", line 1225, in hand
    raise CollectionError("Errors during parsing layer configuration")
bb.cooker.CollectionError: Errors during parsing layer configuration

Creating auto.conf
To build an image run
---------------------------------------------------
MACHINE=qemuriscv64 bitbake core-image-full-cmdline
---------------------------------------------------

Buildable machine info
---------------------------------------------------
* qemuriscv64: The 64-bit RISC-V machine
* qemuriscv32: The 32-bit RISC-V machine
* freedom-u540: The SiFive HiFive Unleashed board
---------------------------------------------------

```

When I ignore the above error and run `MACHINE=qemuriscv32 bitbake core-image-full-cmdline`, the following error is occurred.
`ERROR: Layer 'multimedia-layer' depends on layer 'meta-python', but this layer is not enabled in your configuration
ERROR: Layer 'networking-layer' depends on layer 'meta-python', but this layer is not enabled in your configuration
`

kraj, 2020-01-22:
This is fixed on meta-openembedded and changes will land tomorrow, but I suggest you wait or do the following suggestion or perhaps start using Yoe Distro

https://github.com/YoeDistro/yoe-distro

hmmm it seems you replaced meta-python with meta-python2 actually both layers should be there. Perhaps  modify setup.sh  like below.

diff --git a/setup.sh b/setup.sh
index a38ce6b..af836a8 100644
--- a/setup.sh
+++ b/setup.sh
@@ -36,6 +36,7 @@ export BASH_SOURCE="openembedded-core/oe-init-build-env"

 # add the missing layers
 echo "Adding layers"
+bitbake-layers add-layer ../meta-python2
 bitbake-layers add-layer ../meta-openembedded/meta-oe
 bitbake-layers add-layer ../meta-openembedded/meta-python
 bitbake-layers add-layer ../meta-openembedded/meta-multimedia

geonheec, 2020-01-22:
I followed your suggestion and the `MACHINE=qemuriscv32 bitbake core-image-full-cmdline` command can be started.
But it fails with the following error message...

```
ERROR: glibc-2.30.9000-r0 do_patch: Command Error: 'quilt --quiltrc /home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/glibc/2.30.9000-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
Applying patch 0024-eglibc-Forward-port-cross-locale-generation-support.patch
patching file locale/Makefile
patching file locale/catnames.c
patching file locale/localeinfo.h
Hunk #1 succeeded at 224 (offset -6 lines).
patching file locale/programs/charmap-dir.c
patching file locale/programs/ld-collate.c
patching file locale/programs/ld-ctype.c
patching file locale/programs/ld-time.c
patching file locale/programs/linereader.c
patching file locale/programs/localedef.c
patching file locale/programs/locfile.c
patching file locale/programs/locfile.h
patching file locale/setlocale.c
Hunk #1 FAILED at 63.
1 out of 1 hunk FAILED -- rejects in file locale/setlocale.c
Patch 0024-eglibc-Forward-port-cross-locale-generation-support.patch does not apply (enforce with -f)
ERROR: Logfile of failure stored in: /home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/glibc/2.30.9000-r0/temp/log.do_patch.4002
ERROR: Task (/home/ca2020/riscv-yocto/openembedded-core/meta/recipes-core/glibc/glibc_2.31.bb:do_patch) failed with exit code '1'
```

I think it's better to try after the changes tomorrow.

pino-kim, 2020-01-22:
@kraj, @geonheec  
Thank's to let me know. I will wait for OE's changes merged.

geonheec, 2020-01-23:
Has the patch been reflected?
I try again and get no Parse error, but I keep getting the new error mentioned above.

> I followed your suggestion and the `MACHINE=qemuriscv32 bitbake core-image-full-cmdline` command can be started.
> But it fails with the following error message...
>
> ```
> ERROR: glibc-2.30.9000-r0 [...]

alistair23, 2020-01-23:
Argh, it looks like the glibc patches have changed. I can update the patches.

kraj, 2020-01-23:
yes, we are in process of moving to 2.31 so for now we have moved core to a pre-release version of 2.31 ( master for now )

kraj, 2020-01-23:
> @kraj, @geonheec
> Thank's to let me know. I will wait for OE's changes merged.

meta-openembedded changes for dealing with python2 removal are in now. So, please retry

geonheec, 2020-01-23:
> Argh, it looks like the glibc patches have changed. I can update the patches.

@kraj Thank you for your help :)

@alistair23 Could you please let me know when the update is complete? There is still an error.

pino-kim, 2020-01-23:
@kraj, @alistair23, @geonheec 

$ repo sync
$ . ./meta-riscv
$ MACHINE=qemuriscv64 bitbake core-image-full-cmdline

qemuriscv64 build is fine to me.
and no parse error

-----

$ MACHINE=qemuriscv32 bitbake core-image-full-cmdline

but still error in qemuriscv32.
same error occur to me like  below link.
https://github.com/riscv/meta-riscv/issues/196#issuecomment-577479775

kraj, 2020-01-23:
@pino-kim rv32 is also fixed now kudos to @alistair23

kraj, 2020-01-24:
I am marking it fixed please reopen if needed.

closed 2020-01-24 by kraj