riscv/meta-riscv #199: qemuriscv32 kernel panic when tools-sdk installed

geonheec, 2020-01-24:
Hi, I tried to make riscv32 virtual machine and it was successful.

But when I tried to install tools-sdk additionally for gcc, the kernel panic is occurred and the virtual machine stop booting.

I add the following line in local.conf
`EXTRA_IMAGE_FEATURES_append = " tools-debug tools-sdk dbg-pkgs ssh-server-openssh"`

`MACHINE=qemuriscv32 bitbake core-image-full-cmdline` works well and seems to be no problem.
But `MACHINE=qemuriscv32 runqemu nographic slirp` does not work. It hang with kernel panic during boot.

Is there another way to add gcc in qemuriscv32 vm?

geonheec, 2020-01-24:
ah, let me correct what I said above. 
This error seems to occur whether or not tools-sdk is added.

kraj, 2020-01-24:
instead of adding all tools-sdk Can you just try to add
```
IMAGE_INSTALL_append = " cpp"
```
and see if this fails too ?

kraj, 2020-01-24:
ah I see the real issue seems to be 
```
FATAL: kernel too old
```

Its a config setting that should match glibc's expectations

kraj, 2020-01-24:
@alistair23  in your fork of glibc for rv32 you bumped the minimum kernel needed for rv32 to 5.2  [1] but linux-yocto is still at 5.2, thats why we see the boot failure above. should we use linux-mainline_5.4.bb in interim ?

[1] https://github.com/alistair23/glibc/commit/d0c0df6f2d5b7ab1b62bbec7dfe20f4732f938d1

alistair23, 2020-01-25:
Sorry I have other changes. You need the 5.4 kernel. You can use linux-mainline instead of linux-yocto which should worm

PR #200 (2020-01-25): qemuriscv32: Use linux-yocto-dev for now
Latest glibc wants kernel 5.4+ ABI and linux-yocto is still at 5.2, so
until that switches to 5.4 lets pin to dev kernel which is 5.5+

Fixes #199
commit c95f5d9d (2020-01-25): qemuriscv32: Use linux-yocto-dev for now

Latest glibc wants kernel 5.4+ ABI and linux-yocto is still at 5.2, so
until that switches to 5.4 lets pin to dev kernel which is 5.5+

Fixes #199
closed 2020-01-25 by kraj