riscv/meta-riscv #201: vim cannot be installed

geonheec, 2020-01-25:
Hi, I want to install vim in my qemuriscv32 machine.
I already asked about it and got the answer.
I added `IMAGE_INSTALL_append = " vim"`, but the following error is appeared.
```
ERROR: libxshmfence-1.3-r0 do_compile: oe_runmake failed
ERROR: libxshmfence-1.3-r0 do_compile: Execution of '/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/temp/run.do_compile.1897' failed with exit code 1:
make  all-recursive
make[1]: Entering directory '/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/build'
Making all in src
make[2]: Entering directory '/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/build/src'
../riscv32-oe-linux-libtool  --tag=CC   --mode=compile riscv32-oe-linux-gcc    -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../libxshmfence-1.3/src -I..    -Wall -Wpointer-arith -Wmissing-declarations -Wformat=2 -Wstrict-prototypes -Wmissing-prototypes -Wnested-externs -Wbad-function-cast -Wold-style-definition -Wdeclaration-after-statement -Wunused -Wuninitialized -Wshadow -Wmissing-noreturn -Wmissing-format-attribute -Wredundant-decls -Wlogical-op -Werror=implicit -Werror=nonnull -Werror=init-self -Werror=main -Werror=missing-braces -Werror=sequence-point -Werror=return-type -Werror=trigraphs -Werror=array-bounds -Werror=write-strings -Werror=address -Werror=int-to-pointer-cast -Werror=pointer-to-int-cast -fno-strict-aliasing -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0=/usr/src/debug/libxshmfence/1.3-r0                      -fdebug-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0=/usr/src/debug/libxshmfence/1.3-r0                      -fdebug-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot-native=   -fvisibility=hidden -c -o xshmfence_alloc.lo ../../libxshmfence-1.3/src/xshmfence_alloc.c
../riscv32-oe-linux-libtool  --tag=CC   --mode=compile riscv32-oe-linux-gcc    -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../libxshmfence-1.3/src -I..    -Wall -Wpointer-arith -Wmissing-declarations -Wformat=2 -Wstrict-prototypes -Wmissing-prototypes -Wnested-externs -Wbad-function-cast -Wold-style-definition -Wdeclaration-after-statement -Wunused -Wuninitialized -Wshadow -Wmissing-noreturn -Wmissing-format-attribute -Wredundant-decls -Wlogical-op -Werror=implicit -Werror=nonnull -Werror=init-self -Werror=main -Werror=missing-braces -Werror=sequence-point -Werror=return-type -Werror=trigraphs -Werror=array-bounds -Werror=write-strings -Werror=address -Werror=int-to-pointer-cast -Werror=pointer-to-int-cast -fno-strict-aliasing -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0=/usr/src/debug/libxshmfence/1.3-r0                      -fdebug-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0=/usr/src/debug/libxshmfence/1.3-r0                      -fdebug-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot-native=   -fvisibility=hidden -c -o xshmfence_futex.lo ../../libxshmfence-1.3/src/xshmfence_futex.c
riscv32-oe-linux-libtool: compile:  riscv32-oe-linux-gcc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../libxshmfence-1.3/src -I.. -Wall -Wpointer-arith -Wmissing-declarations -Wformat=2 -Wstrict-prototypes -Wmissing-prototypes -Wnested-externs -Wbad-function-cast -Wold-style-definition -Wdeclaration-after-statement -Wunused -Wuninitialized -Wshadow -Wmissing-noreturn -Wmissing-format-attribute -Wredundant-decls -Wlogical-op -Werror=implicit -Werror=nonnull -Werror=init-self -Werror=main -Werror=missing-braces -Werror=sequence-point -Werror=return-type -Werror=trigraphs -Werror=array-bounds -Werror=write-strings -Werror=address -Werror=int-to-pointer-cast -Werror=pointer-to-int-cast -fno-strict-aliasing -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0=/usr/src/debug/libxshmfence/1.3-r0 -fdebug-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0=/usr/src/debug/libxshmfence/1.3-r0 -fdebug-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot= -fdebug-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot-native= -fvisibility=hidden -c ../../libxshmfence-1.3/src/xshmfence_alloc.c  -fPIC -DPIC -o .libs/xshmfence_alloc.o
riscv32-oe-linux-libtool: compile:  riscv32-oe-linux-gcc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../libxshmfence-1.3/src -I.. -Wall -Wpointer-arith -Wmissing-declarations -Wformat=2 -Wstrict-prototypes -Wmissing-prototypes -Wnested-externs -Wbad-function-cast -Wold-style-definition -Wdeclaration-after-statement -Wunused -Wuninitialized -Wshadow -Wmissing-noreturn -Wmissing-format-attribute -Wredundant-decls -Wlogical-op -Werror=implicit -Werror=nonnull -Werror=init-self -Werror=main -Werror=missing-braces -Werror=sequence-point -Werror=return-type -Werror=trigraphs -Werror=array-bounds -Werror=write-strings -Werror=address -Werror=int-to-pointer-cast -Werror=pointer-to-int-cast -fno-strict-aliasing -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0=/usr/src/debug/libxshmfence/1.3-r0 -fdebug-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0=/usr/src/debug/libxshmfence/1.3-r0 -fdebug-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot= -fdebug-prefix-map=/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot-native= -fvisibility=hidden -c ../../libxshmfence-1.3/src/xshmfence_futex.c  -fPIC -DPIC -o .libs/xshmfence_futex.o
In file included from ../../libxshmfence-1.3/src/xshmfenceint.h:32,
                 from ../../libxshmfence-1.3/src/xshmfence_alloc.c:27:
../../libxshmfence-1.3/src/xshmfence_futex.h: In function 'sys_futex':
../../libxshmfence-1.3/src/xshmfence_futex.h:58:17: error: 'SYS_futex' undeclared (first use in this function); did you mean 'sys_futex'?
   58 |  return syscall(SYS_futex, addr1, op, val1, timeout, addr2, val3);
      |                 ^~~~~~~~~
      |                 sys_futex
../../libxshmfence-1.3/src/xshmfence_futex.h:58:17: note: each undeclared identifier is reported only once for each function it appears in
../../libxshmfence-1.3/src/xshmfence_alloc.c: At top level:
../../libxshmfence-1.3/src/xshmfence_alloc.c:51: warning: "MFD_CLOEXEC" redefined
   51 | #define MFD_CLOEXEC  0x0001U
      |
In file included from /home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot/usr/include/bits/mman-linux.h:113,
                 from /home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot/usr/include/bits/mman.h:32,
                 from /home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/recipe-sysroot/usr/include/sys/mman.h:41,
                 from ../../libxshmfence-1.3/src/xshmfenceint.h:28,
[... cut ...]
make[1]: Leaving directory '/home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/build'
Makefile:405: recipe for target 'all' failed
make: *** [all] Error 2
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/ca2020/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/libxshmfence/1.3-r0/temp/log.do_compile.1897
```
`ERROR: Task (/home/ca2020/riscv-yocto/openembedded-core/meta/recipes-graphics/xorg-lib/libxshmfence_1.3.bb:do_compile) failed with exit code '1'`

How can I solve this problem?

kraj, 2020-01-25:
I think this is due to discrepency between glibc and kernel syscall understanding. @alistair23 I have replied to relevant patch that is on glibc mailing list to pin this issue.

geonheec, 2020-01-29:
May I ask when the bug will be fixed?

kraj, 2020-01-30:
> May I ask when the bug will be fixed?

I think a workaround might be to define SYS_futex locally in term of 64bit futex call in libxshmfence package, but there might be more packages in same boat.

alistair23, 2020-01-30:
I have fixed up the "core" packages, so I can build a full RV32 rootfs and upstreamed the changes. Any program that is doing it's own syscalls might need to be updated (based on the syscalls it's doing). I haven't looked into packages that I don't directly need yet. This could be a fairly complex change depending on the syscalls.

You can see the discussion here for more information about handling the syscall safely: https://patchwork.sourceware.org/patch/37316/

In this case you could probably get away with just adding a simple:
```c
# ifndef __NR_futex
#  define __NR_futex __NR_futex_time64
# endif
```
for RISC-V

PR #205 (2020-02-01): mesa,libxshmfence: Define __NR_futex for rv32
Helps these nosy apps build
Fixes #201
commit 8b73b069 (2020-02-01): mesa,libxshmfence: Define __NR_futex for rv32

Helps these nosy apps build
Fixes #201
closed 2020-02-01 by kraj