riscv/meta-riscv #202: List of packages failing World build on RV32

kraj, 2020-01-26:
This is a placeholder for tracking world build failures on rv32, the current list and respective error logs are here [1]

[1] http://errors.yoctoproject.org/Errors/Build/97499/

pino-kim, 2020-01-26:
@kraj ,
hmm.....
I want to get samething in my bild system.
just build below command?
                 MACHINE=qemuriscv32 bitbkae core-image-full-cmdline

kraj, 2020-01-26:
@pino-kim you will run `bitbake -k world`

alistair23, 2020-01-30:
That's a great list!

I had a quick look and it looks like a lot of these just don't have RV32 support yet, so this will be a fair amount of work.

kraj, 2020-04-25:
@alistair23  I think some of the packages fail due to syscalls e.g.
```
error: 'SYS_mmap' was not declared in this scope
error: 'SYS_futex' was not declared in this scope
```
These perhaps are fixable easily

some are atomics related can be fixed by adding -pthread to linker cmdline

a big category is then of the not ported ones.

alistair23, 2020-04-25:
Cool! Thanks for looking into this.

ajf58, 2022-01-03:
This has a label of "help wanted", is that still the case? If so, what additional help is required?

kraj, 2022-01-07:
@ajf58 yes perhaps fixing the 64bit time_t issues for rv32 is a  big one and also some key packages like
libseccomp, libunwind are key elements needed to enable various yocto features.

sanjay-embedded, 2026-01-15:
@kraj ,

Do we still have issue with world build ?
Or it's quite stable.

Any specific points where we can contribute for risc-v ?

kraj, 2026-10-12:
world builds have been fixed for some time now, some packages keep regressing but that should be tracked separately when it happens.

closed 2026-10-12 by kraj