riscv/meta-riscv #210: kprobes support to riscv

kirana1996, 2020-02-24:
hello,
we are building the yocto image for riscv with kernel version is 5.4,in that we are tried for enable the all kprobes dependencies but is not working. can anyone conclude me is it there kprobes support for riscv..

thanks,

kraj, 2020-02-24:
have you tried enabling 
```

CONFIG_FUNCTION_TRACER=y
CONFIG_KPROBES=y
CONFIG_MODULES=y
```
in kernel config

kirana1996, 2020-02-24:
yes i tried those things but those are not working, getting ´´linux_mainline-5.4.bb:do_compile error´´..

alistair23, 2020-02-24:
Can you please share the error?

kirana1996, 2020-02-25:
hi @alistair23, this is the error  
ERROR: linux-mainline-5.4.x+gitAUTOINC+f7688b48ac-r0 do_compile: oe_runmake failed
ERROR: linux-mainline-5.4.x+gitAUTOINC+f7688b48ac-r0 do_compile: Execution of '/home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work/exaleapsemi-oe-linux/linux-mainline/5.4.x+gitAUTOINC+f7688b48ac-r0/temp/run.do_compile.19532' failed with exit code 1:
  GEN     Makefile
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/expr.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --syncconfig Kconfig
  GEN     Makefile
  CALL    /home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/scripts/atomic/check-atomics.sh
sha1sum not available, skipping atomic header checks.
  CALL    /home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/scripts/checksyscalls.sh
  AS      usr/initramfs_data.o
  CHK     include/generated/compile.h
  AR      usr/built-in.a
  CC      kernel/sysctl.o
In file included from /home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/kernel/sysctl.c:59:
/home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/linux/kprobes.h:70:2: error: unknown type name 'kprobe_opcode_t'
   70 |  kprobe_opcode_t *addr;
      |  ^~~~~~~~~~~~~~~
/home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/linux/kprobes.h:91:2: error: unknown type name 'kprobe_opcode_t'
   91 |  kprobe_opcode_t opcode;
      |  ^~~~~~~~~~~~~~~
/home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/linux/kprobes.h:94:28: error: field 'ainsn' has incomplete type
   94 |  struct arch_specific_insn ainsn;
      |                            ^~~~~
/home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/linux/kprobes.h:161:2: error: unknown type name 'kprobe_opcode_t'
  161 |  kprobe_opcode_t *ret_addr;
      |  ^~~~~~~~~~~~~~~
  CC      lib/bsearch.o
/home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/linux/kprobes.h:235:34: error: unknown type name 'kprobe_opcode_t'
  235 | extern bool kprobe_on_func_entry(kprobe_opcode_t *addr, const char *sym, unsigned long offset);
      |                                  ^~~~~~~~~~~~~~~
In file included from /home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/asm-generic/percpu.h:7,
                 from ./arch/riscv/include/generated/asm/percpu.h:1,
                 from /home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/asm-generic/mmiowb.h:26,
                 from /home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/arch/riscv/include/asm/mmiowb.h:12,
                 from /home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/linux/spinlock.h:60,
                 from /home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/linux/seqlock.h:36,
                 from /home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/linux/time.h:6,
                 from /home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/linux/stat.h:19,
                 from /home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/linux/module.h:10,
                 from /home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/kernel/sysctl.c:22:
/home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/linux/kprobes.h: In function 'get_kprobe_ctlblk':
/home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/include/linux/percpu-defs.h:220:52: error: invalid use of undefined type 'struct kprobe_ctlblk'
  220 |  const void __percpu *__vpp_verify = (typeof((ptr) + 0))NULL; \
      |                                                    ^
[... cut ...]
/home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work-shared/exaleapsemi/kernel-source/Makefile:179: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/exaleap-vrt2/khan/yocto/2020_02_05-10.09.20/riscv-yocto/openembedded-exa-core/riscv-build/tmp-glibc/work/exaleapsemi-oe-linux/linux-mainline/5.4.x+gitAUTOINC+f7688b48ac-r0/temp/log.do_compile.19532

alistair23, 2020-02-25:
There is no kprobe support for RISC-V merged in mainline Linux. There is a patch on the mailing list but it looks like it was never merged: https://lwn.net/Articles/771756/

closed 2020-02-25 by alistair23