riscv/meta-riscv #222: qemuriscv64

snathicktechno, 2020-05-03:
Build machine : ubuntu 18.04

Build qemuriscv64 image using meta-riscv layer

no graphic mode is working fine

in graphic mode i cannot pass

ctrl+c, ctrl+z and ctrl+d options

verified libs on host : libsdl2.0 gtk2.0

Help me to fix this issue

Thanks
Snathick

kraj, 2020-05-14:
@alistair23 any idea on this ?

alistair23, 2020-05-14:
I doubt it's a RISC-V specific issue. I never use graphic mode so I'm not sure.

pino-kim, 2020-05-15:
longtime no see. 
I have a same Question, Can we use graphic image like core-image-sato(?)

alistair23, 2020-05-18:
It should work. Please file an issue if it doesn't

kraj, 2021-06-02:
we have now this fixed in master oe-core https://git.openembedded.org/openembedded-core/commit/?id=ec085b75a1edb14c6e4dd1dc2f5cdf62f44d0e39

please try core-image-sato oe core-image-weston and it should boot to graphics.

closed 2021-06-02 by kraj