riscv/meta-riscv #225: Failing Extensible SDK build for RISCV

ghalib210, 2020-06-02:
Hello,
          I have build an Riscv Extensible SDK yocto project, with help of following step's.

cd $(PWD)/ && \
git clone git://git.openembedded.org/openembedded-core 
cd $(PWD)/openembedded-core && \ 
git clone git://git.openembedded.org/bitbake && \ 
git clone git://github.com/riscv/meta-riscv 
source oe-init-build-env riscv-build/ 
bitbake-layers add-layer ../meta-riscv 
vim conf/local.conf
Added MACHINE = "qemuriscv64"
bitbake core-image-minimal
bitbake core-image-minimal -c populate_sdk
cd tmp-glibc/deploy/sdk
./oecore-x86_64-riscv64-toolchain-ext-nodistro.0.sh

Here I got an sstate cache Error, as follows in the attachment.
[e-sdk-build.log](https://github.com/riscv/meta-riscv/files/4718267/e-sdk-build.log)

ERROR: Task (/home/khan/khan/sdk-with-meta-riscv/openembedded-core/riscv-build/tmp-glibc/deploy/sdk/e-sdk/layers/openembedded-core/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_fetch) failed with exit code 'setscene whitelist'

Can anyone help me to build an Extensible SDK with openembedded-core & meta-riscv.

Thanks & Regards,
Ghalib khan.

kraj, 2020-06-02:
@ghalib210 I think you should use uninative for SDK to work reliably on different hosts.

add 

https://github.com/YoeDistro/meta-yoe/blob/master/conf/distro/yoe.inc#L94
https://github.com/YoeDistro/meta-yoe/blob/master/conf/distro/yoe.inc#L97

to your local.conf.

Even better if you use yoe distro then its already taken care.

kraj, 2021-06-02:
@ghalib210 is this issue still pertinent ?

kraj, 2026-10-12:
closing for lack of activity

closed 2026-10-12 by kraj