riscv/meta-riscv #228: when I build gdb8.3 with musl occured  ERROR: oe_runmake failed

TONG369, 2020-07-10:
Hello~Thank you very much for your contribution to this project. At present, I have integrated the musl C library on yocto.  I added the layer to bblayers.conf and also added TCLIBC = "musl" to local.conf I initialized the environment and then run bitbake core-image-full-cmdline-dev, I have got the error outpu.

At first, I thought there were broken packages. Firstly, try cleaning it and rebuilding: bitbake -c cleanall -f core-image-full-cmdline-ncnn-dev. But when I compile again, I still get the following output, why is this? When I input MACHINE=freedom-u540 TCLIBC=musl bitbake core-image-full-cmdline-ncnn can be executed normally, while executing MACHINE=freedom-u540 TCLIBC=musl bitbake core-image-full-cmdline-ncnn-dev is wrong.Is there a conflict between gdb8.3 and musl?The execution environment of the them is the same. I don't know the reason for the error. I am looking forward to your reply!Thank you !How should it be solved?

NOTE: recipe gettext-0.19.8.1-r0: task do_populate_sysroot: Succeeded
NOTE: recipe gcc-9.2.0-r0: task do_populate_sysroot: Succeeded
NOTE: recipe strace-5.2-r0: task do_populate_sysroot: Succeeded
NOTE: recipe sysprof-3.32.0-r0: task do_populate_sysroot: Succeeded
ERROR: gdb-8.3-r0 do_compile: oe_runmake failed
ERROR: gdb-8.3-r0 do_compile: Execution of '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/temp/run.do_compile.310737' failed with exit code 1:
make[1]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl'
make[1]: Nothing to be done for 'all-target'.
make[2]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/intl'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/intl'
make[2]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/etc'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/etc'
make[2]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/libdecnumber'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/libdecnumber'
make[2]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/libiberty'
make[3]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/libiberty/testsuite'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/libiberty/testsuite'
make[2]: Leaving directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/libiberty'
make[2]: Entering directory '/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/build-riscv64-oe-linux-musl/zlib'
true "AR_FLAGS=rc" "CC_FOR_BUILD=gcc " "CFLAGS=-O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0 -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0 -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot= -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot-native= " "CXXFLAGS=-O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0 -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0 -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot= -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot-native= -fvisibility-inlines-hidden" "CFLAGS_FOR_BUILD=-isystem/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot-native/usr/include -O2 -pipe" "CFLAGS_FOR_TARGET=-O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0 -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0 -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot= -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot-native= " "INSTALL=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/hosttools/install -c" "INSTALL_DATA=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/hosttools/install -c -m 644" "INSTALL_PROGRAM=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/hosttools/install -c" "INSTALL_SCRIPT=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/hosttools/install -c" "LDFLAGS=-static-libstdc++ -static-libgcc -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now" "LIBCFLAGS=-O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0 -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0 -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot= -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot-native= " "LIBCFLAGS_FOR_TARGET=-O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0 -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0 -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot= -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot-native= " "MAKE=make" "MAKEINFO=makeinfo --split-size=5000000 --split-size=5000000 " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/usr" "infodir=/usr/share/info" "libdir=/usr/lib" "prefix=/usr" "tooldir=/usr/riscv64-oe-linux-musl" "AR=riscv64-oe-linux-musl-gcc-ar" "AS=riscv64-oe-linux-musl-as " "CC=riscv64-oe-linux-musl-gcc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0 -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0=/usr/src/debug/gdb/8.3-r0 -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8.3-r0/recipe-sysroot= -fdebug-prefix-map=/builds/gitlab/riscv/meta-riscv/riscv-yocto/build/tmp-musl/work/riscv64-oe-linux-musl/gdb/8 [... cut ...]
NOTE: do_populate_sysroot: 0.0% sstate reuse(0 setscene, 4 scratch)
NOTE: do_package: 0.0% sstate reuse(0 setscene, 4 scratch)
Summary: 1 task failed:
/builds/gitlab/riscv/meta-riscv/riscv-yocto/openembedded-core/meta/recipes-devtools/gdb/gdb_8.3.bb:do_compile
Summary: There were 2 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
this is my logfile:
[log.do_compile.310737.txt](https://github.com/riscv/meta-riscv/files/4900331/log.do_compile.310737.txt)
@asb @arunthomas @gurjeet @alistair23  @kraj

alistair23, 2020-07-10:
GDB 8.3 is pretty old, can you update your layers and see if that helps?

TONG369, 2020-07-11:
Hello! Thank you very much for your reply. Before running glibc, these errors did not occur. Appeared when running musl.

@alistair23 @asb @arunthomas @gurjeet @kraj

kraj, 2020-07-11:
All we are asking you is to use latest master branch of yocto layers and you are not reporting back if you did so 
We do not ship gdb 8.3 anymore so you are getting it from either an old layer or different place 

So it’s better if you provide relevant information that is being asked in order for us to support you better

TONG369, 2020-07-14:
Sorry, the reply is late, what information do I need to provide?please indicate.Thank you very much.
@alistair23 @kraj @asb @arunthomas

TONG369, 2020-07-16:
What other information do I need to provide?
@alistair23 @kraj @asb @arunthomas

kraj, 2020-09-01:
@TONG369 I would suggest to use master or dunfell release, you seem to be using 3.0/zeus and meta-riscv I could not match. musl/gdb works ok on master, I will close this issue, please re-open if you need some other resolution

closed 2020-09-01 by kraj