riscv/meta-riscv #232: [Question] MACHINE=freedom-u540 bitbake core-image-wayland can't build

pino-kim, 2020-07-23:
**Description**
I make this issue(**question**) about MACHINE=freedom-u540 bitbake **core-image-wayland**
I can't build MACHINE=freedom-u540 bitbake **core-image-wayland** 
Should I add more meta-layer for build **core-image-wayland**  or read more doc to follow? 

**Steps to reproduce the issue:**
1. follow the README.md  
2. run $MACHINE=freedom-u540 bitbake core-image-wayland**

**Describe the results you received:**
build starting.

**Describe the results you expected:**
can not found **core-image-wayland**.

**Additional information you deem important (e.g. issue happens only occasionally):** 

**Additional details (revisions used, host distro, etc.):**
Build Host : ubuntu 18.04

kraj, 2020-07-23:
@pino-kim there is no pre-existing `core-image-wayland` but there is a similar reference image called `core-image-weston` please use that as target

pino-kim, 2020-07-23:
@kraj 
Thanks for your reply. 
I can't find core-image-wayland too.

So, follow your guild.
$MACHINE=freedom-u540 bitbake core-image-wayland
to
$MACHINE=freedom-u540 bitbake core-image-weston

Thanks.

pino-kim, 2020-07-24:
Thanks @kraj 
I can build **core-image-weston** image.
Would you check up PR #233 ?

Close this issue.

closed 2020-07-24 by pino-kim