riscv/meta-riscv #235: [Question] MACHINE=qemuriscv32 bitbake core-image-lsb-sdk is not work

pino-kim, 2020-08-09:
**Description**
MACHINE=qemuriscv32 bitbake core-image-lsb-sdk is not work.
And I can't find that **core-image-lsb-sdk.bb** recipe after this commit message
>pino@lovrecina:~/workspace/opensource/riscv-yocto/openembedded- 
  core/meta/recipes-extended$ git show f
  b064356af615d67d85b65942103bf943d84d290
  commit fb064356af615d67d85b65942103bf943d84d290
  Author: Adrian Bunk <bunk@stusta.de>
  Date:   Sun Aug 25 20:21:15 2019 +0300

    Remove LSB support

    LSB as a standard isn't current and isn't well suited to embedded
    anyway. Its putting artifical constraints on the system and with modern
    layer technology, would now be better off as its own layer. As such
    its time to split it out.

    The only part with some (marginal) usage is lsb_release,
    which is split from the lsb package into an own lsb-release
    package.

    Signed-off-by: Adrian Bunk <bunk@stusta.de>
    Signed-off-by: Richard Purdie <richard.purdie@linuxfoundation.org>
 

**Steps to reproduce the issue:**
1. Read the README.md 
2. Setup riscv-yocto
3. Run  'MACHINE=qemuriscv32 bitbake core-image-lsb-sdk'

**Describe the results you received:**

> Loading cache: 100% |                                                                    | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |###################################################################| Time: 0:01:29
Parsing of 2243 .bb files complete (0 cached, 2243 parsed). 3432 targets, 169 skipped, 0 masked, 0 errors.
ERROR: Nothing PROVIDES 'core-image-lsb-sdk'. Close matches:
  core-image-base
  core-image-rt-sdk
  core-image-sato-sdk

**Describe the results you expected:**
hmm... parsing complete?

**Additional details (revisions used, host distro, etc.):**
build host : ubuntu 18.04 x86

pino-kim, 2020-08-10:
@kraj, @alistair23 
If am I right, I think LSB is deprecated in recent yocto.
https://www.yoctoproject.org/docs/latest/ref-manual/ref-manual.html#migration-3.0-lsb-support-removed

So I think we need some change in README.md to create SDK
Anygood idea?

kraj, 2020-08-10:
yes lsb images are deprecated, use `core-image-full-cmdline`

pino-kim, 2020-08-10:
@kraj, Thanks to your double-check.
I create some PR to generate SDK in README.md

pino-kim, 2020-08-25:
@kraj 
I push late PR.
Can you check #238 please?
Close.

closed 2020-08-25 by pino-kim