riscv/meta-riscv #236: build

pino-kim, 2020-08-10:
**Description**
Build fail **MACHINE=qemuriscv64 bitbake core-image-full-cmdline** current master.
Build failed in do_compile gobject-introspection

**Steps to reproduce the issue:**
1. Follow README.md
2. setup riscv-yocto
3. start build **MACHINE=qemuriscv64 bitbake core-image-full-cmdline**

**Describe the results you received:**
* below log is a clue to solve the issue.
* and add full log file too.
```
| In file included from /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot/usr/include/bits/libc-header-start.h:33,
|                  from /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot/usr/include/limits.h:26,
|                  from /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot-native/usr/lib/riscv64-oe-linux/gcc/riscv64-oe-linux/10.1.0/include-fixed/limits.h:195,
|                  from /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot-native/usr/lib/riscv64-oe-linux/gcc/riscv64-oe-linux/10.1.0/include-fixed/syslimits.h:7,
|                  from /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot-native/usr/lib/riscv64-oe-linux/gcc/riscv64-oe-linux/10.1.0/include-fixed/limits.h:34,
|                  from /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot/usr/lib/glib-2.0/include/glibconfig.h:11,
|                  from /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot/usr/include/glib-2.0/glib/gtypes.h:32,
|                  from /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot/usr/include/glib-2.0/glib/galloca.h:32,
|                  from /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot/usr/include/glib-2.0/glib.h:30,
|                  from /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot/usr/include/glib-2.0/gobject/glib-types.h:24,
|                  from /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/build/g-ir-cpp-an2r1syq.c:4:
| /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot/usr/include/features.h:397:4: warning: #warning _FORTIFY_SOURCE requires compiling with optimization (-O) [-Wcpp]
|   397 | #  warning _FORTIFY_SOURCE requires compiling with optimization (-O)
|       |    ^~~~~~~
| /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/build/g-ir-scanner-qemuwrapper: line 6: 25864 Segmentation fault      (core dumped) PSEUDO_UNLOAD=1 qemu-riscv64 -r 4.15 -L /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot -E LD_LIBRARY_PATH=$GIR_EXTRA_LIBS_PATH:.libs:/home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot//usr/lib:/home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/recipe-sysroot//lib "$@"
| If the above error message is about missing .so libraries, then setting up GIR_EXTRA_LIBS_PATH in the recipe should help.
| (typically like this: GIR_EXTRA_LIBS_PATH="${B}/something/.libs" )
| Command '['/home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/build/g-ir-scanner-qemuwrapper', '/home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/build/tmp-introspect20_iujub/GLib-2.0', '--introspect-dump=/home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/build/tmp-introspect20_iujub/functions.txt,/home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/build/tmp-introspect20_iujub/dump.xml']' returned non-zero exit status 1.
| ninja: build stopped: subcommand failed.
| WARNING: /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/gobject-introspection/1.64.1-r0/temp/run.do_compile.25611:1 exit 1 from 'ninja -v -j 4'
|
[build_fail_gobject-introspection.log](https://github.com/riscv/meta-riscv/files/5048844/build_fail_gobject-introspection.log)
```

**Describe the results you expected:**
build complete.

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
build host : x86 ubuntu 18.04 
* meta-openembedded  ff59a16b563341ba44c119a0c6faefe3ddb2b70d
* meta-riscv f6438269ebe2631f96908e6b115e0dccf79600e5
* openembedded-core fe3d5136ef02721f9bbe01d98c1743b8dbcb69c5

kraj, 2020-08-10:
it looks like qemu is crashing in usermode

alistair23, 2020-08-10:
It looks like it started from a binutils update. This is the first breaking commit for me:

```diff
commit 7a7667127625244aae2b1ff179e431fc9ad25737
Author: Khem Raj <raj.khem@gmail.com>
Date:   Thu Jul 30 00:53:43 2020 -0700

    binutils: Upgrade to 2.35 release
    
    assembler supports dwarf-5 format line numbers
    linker has -M -MP gcc-like option
    
    for detailed release notes see [1]
    
    [1] https://lists.gnu.org/archive/html/info-gnu/2020-07/msg00008.html
    
    Signed-off-by: Khem Raj <raj.khem@gmail.com>
    Signed-off-by: Richard Purdie <richard.purdie@linuxfoundation.org>
```

alistair23, 2020-08-10:
This also looks to be the same problem as: https://errors.yoctoproject.org/Errors/Details/427088/

alistair23, 2020-08-10:
Confirmed that reverting the binutils upgrade fixes the failure.

kraj, 2020-08-10:
that means we need to bisect binutils now.

alistair23, 2020-08-10:
Looking in QEMU the seg fault happens between the jump from: `_dl_relocate_object` to `check_match`.

In the instruction and strace from QEMU I don't see any difference between the working and non working version. I'm not sure what is different.

pino-kim, 2020-08-10:
@kraj, @alistair23 
revert commit 7a7667127625244aae2b1ff179e431fc9ad25737, binutils: Upgrade to 2.35 release.
Build completed.
I wonder about this issue for a day. But nothing clear to me.
Is there something big change in **Binutils** 2.35 and 2.34?

alistair23, 2020-08-11:
This diff seems to fix it for me:

```diff
diff --git a/meta/recipes-devtools/binutils/binutils-2.35.inc b/meta/recipes-devtools/binutils/binutils-2.35.inc
index 9aa5e7efe3..5ae9ec4226 100644
--- a/meta/recipes-devtools/binutils/binutils-2.35.inc
+++ b/meta/recipes-devtools/binutils/binutils-2.35.inc
@@ -41,6 +41,5 @@ SRC_URI = "\
      file://0014-Fix-rpath-in-libtool-when-sysroot-is-enabled.patch \
      file://0015-sync-with-OE-libtool-changes.patch \
      file://0016-Check-for-clang-before-checking-gcc-version.patch \
-     file://0017-Do-not-emit-R_RISCV_NONE-reloc-in-the-extra-unused-r.patch \
 "
 S  = "${WORKDIR}/git"
```

The description in that commit points to here: https://sourceware.org/bugzilla/show_bug.cgi?id=24673 but I don't see the patch there. That discussion is also a year old, but the patch was only just added to OE.

@kraj any idea where/why you applied this?

kraj, 2020-08-11:
I think that patch was an attempt to fix pie issues I was seeing back then we can remove this patch please send this patch to remove it from Oe-core

pino-kim, 2020-08-11:
@kraj, @alistair23 
If you send the path,
Would you link the path's link to this issue? 
I want to learn about how to send the path to OE and it's progress.
cheers

alistair23, 2020-08-11:
Patch sent, I CCed you @pino-kim.

You can also see the patch here: https://lists.openembedded.org/g/openembedded-core/topic/patch_binutils_remove/76127630?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,76127630

pino-kim, 2020-08-11:
@alistair23 
Thanks, It really helps me a lot.

closed 2020-08-12 by kraj