riscv/meta-riscv #239: [Issue] riscv32 : glibc2.32 fetch issue

pino-kim, 2020-08-25:
**Description**
riscv32 has to fetch issue 

```text
ERROR: glibc-2.32-r0 do_fetch: Fetcher failure: Unable to find revision b3ef7a831e6e80d44ea36a9712395b3eda076b82 in branch alistair/rv32.2.3
2 even from upstream                                                                                                                        
ERROR: glibc-2.32-r0 do_fetch: Fetcher failure for URL: 'git://github.com/alistair23/glibc.git;branch=alistair/rv32.2.32;name=glibc'. Unable
 to fetch URL from any source.                                                                                                              
ERROR: Logfile of failure stored in: /home/pino/workspace/opensource_work/riscv-yocto/build/tmp-glibc/work/riscv32-oe-linux/glibc/2.32-r0/te
mp/log.do_fetch.3813                                                                                                                        
ERROR: Task (/home/pino/workspace/opensource_work/riscv-yocto/openembedded-core/meta/recipes-core/glibc/glibc_2.32.bb:do_fetch) failed with 
exit code '1'   
```

**Steps to reproduce the issue:**
1. follow README.md to set build env
2. MACHINE=qemuriscv32 bitbake core-image-full-cmdline

**Describe the results you received:**
glibc fetch fail.

**Describe the results you expected:**
run risv32 on qemu

**Additional information you deem important (e.g. issue happens only occasionally):**
add a full logfile.
[rv32_glibc_fetch_issue.log](https://github.com/riscv/meta-riscv/files/5126676/rv32_glibc_fetch_issue.log)

**Additional details (revisions used, host distro, etc.):**
build host x86 ubuntu 18.04

alistair23, 2020-08-25:
Ha, I deleted that branch. Apparently it was being used though. I'll send a patch.

pino-kim, 2020-08-25:
@alistair23 
Wow, really quick reply.
Oh... I can't find a branch on your git.

alistair23, 2020-08-26:
I opened a PR: https://github.com/riscv/meta-riscv/pull/240

closed 2020-08-26 by kraj