riscv/meta-riscv #244: [Question] fuzzing waring mesa, building riscv32

pino-kim, 2020-08-31:
**Description**
I watched a mesa patch fuzzing warning when building qemuriscv32.
And ask about is fine and I mended it PR #243

```text
Build Configuration:
BB_VERSION           = "1.47.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv32-oe-linux"
MACHINE              = "qemuriscv32"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv32"
meta                 = "work:09f4db415fb6a1398e9e9b359630043c833f6118"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      = "work:c32d2eb448ce343463dc75cc6120f395e32f0177"
meta-riscv           = "HEAD:6ca7222f59e9a030923530cbe30d94434d27a5f2"

Initialising tasks: 100% |###################################################################################################| Time: 0:00:02
Sstate summary: Wanted 1092 Found 33 Missed 1059 Current 313 (3% match, 24% complete)
NOTE: Executing Tasks
WARNING: mesa-2_20.1.5-r0 do_patch: Fuzz detected:

Applying patch rv32-sys-futex.patch
patching file src/util/futex.h
Hunk #1 succeeded at 27 with fuzz 2 (offset 1 line).

The context lines in the patches can be updated with devtool:

    devtool modify mesa
    devtool finish --force-patch-refresh mesa <layer_path>

Don't forget to review changes done by devtool!

WARNING: mesa-2_20.1.5-r0 do_patch: QA Issue: Patch log indicates that patches do not apply cleanly. [patch-fuzz]
NOTE: Tasks Summary: Attempted 4193 tasks of which 1570 didn't need to be rerun and all succeeded.
```

**Steps to reproduce the issue:**
1. build setup with README.md
2. 

**Describe the results you received:**

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

kraj, 2020-08-31:
it perhaps needs refreshing, so go to mesa build dir, and perform a quilt refresh after popping this patch to top of quilt series.

```
quilt pop  rv32-sys-futex.patch
quilt refresh
```

and this should refresh the patch and send that pull request here

pino-kim, 2020-08-31:
@kraj. 
Thanks for your reply and your guide.

closed 2020-08-31 by pino-kim