riscv/meta-riscv #245: [Issue] When type poweroff in shell, return opensbi dump log and shell hang both qemuriscv64, 32.

pino-kim, 2020-08-31:
**Description**

**To. @kraj, @alistair23** 
Find something strange things both qemuriscv64 and qemuriscv32 when cmd **poweroff** in qemu shell.
I guess it's not only meta-riscv 's issue. so I open the issue for sharing log to get more info.
And someone tells me where to report this issue. (opensbi, kernel, qemu ?)

----
**qemuriscv64**
```text
root@qemuriscv64:~# uname -a
Linux qemuriscv64 5.8.4-yocto-standard #1 SMP PREEMPT Mon Aug 31 14:53:11 UTC 2020 riscv64 riscv64 riscv64 GNU/Linux
root@qemuriscv64:~# poweroff
...
[  639.207157] systemd-shutdown[1]: Syncing filesystems and block devices.
[  639.221252] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[  639.276840] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[  639.317552] systemd-shutdown[1]: Unmounting file systems.
[  639.331773] [208]: Remounting '/' read-only in with options '(null)'.
[  639.341736] EXT4-fs (vda): re-mounted. Opts: (null)
[  639.346889] systemd-shutdown[1]: All filesystems unmounted.
[  639.347202] systemd-shutdown[1]: Deactivating swaps.
[  639.348186] systemd-shutdown[1]: All swaps deactivated.
[  639.348477] systemd-shutdown[1]: Detaching loop devices.
[  639.353158] systemd-shutdown[1]: All loop devices detached.
[  639.353464] systemd-shutdown[1]: Detaching DM devices.
[  639.356555] systemd-shutdown[1]: All DM devices detached.
[  639.356856] systemd-shutdown[1]: All filesystems, swaps, loop devices and DM devices detached.
[  639.366499] systemd-shutdown[1]: Syncing filesystems and block devices.
[  639.372414] systemd-shutdown[1]: Powering off.
[  640.373271] reboot: Power down
sbi_trap_error: hart0: trap handler failed (error -2)
sbi_trap_error: hart0: mcause=0x0000000000000007 mtval=0x0000000000100000
sbi_trap_error: hart0: mepc=0x000000008000d9fa mstatus=0x0000000000001822
sbi_trap_error: hart0: ra=0x0000000080009cd0 sp=0x0000000080016c68
sbi_trap_error: hart0: gp=0xffffffe00111bd70 tp=0xffffffe00b338000
sbi_trap_error: hart0: s0=0x0000000080016c78 s1=0x0000000000000040
sbi_trap_error: hart0: a0=0x0000000000000000 a1=0x0000000080003fb0
sbi_trap_error: hart0: a2=0x0000000080003fb0 a3=0x0000000080003fb0
sbi_trap_error: hart0: a4=0x0000000000100000 a5=0x0000000000005555
sbi_trap_error: hart0: a6=0x0000000000003fb0 a7=0x0000000080012158
sbi_trap_error: hart0: s2=0x0000000000000000 s3=0x0000000080017000
sbi_trap_error: hart0: s4=0x0000000000000000 s5=0x0000000000000000
sbi_trap_error: hart0: s6=0x0000000000000001 s7=0x0000000000000000
sbi_trap_error: hart0: s8=0x0000000000000000 s9=0x0000000000000000
sbi_trap_error: hart0: s10=0x0000000000000000 s11=0x0000000000000008
sbi_trap_error: hart0: t0=0x0000000000000000 t1=0x0000000000000000
sbi_trap_error: hart0: t2=0x0000000000000000 t3=0x0000000000000000
sbi_trap_error: hart0: t4=0x0000000000000000 t5=0x0000000000000000
sbi_trap_error: hart0: t6=0x0000000000000000
[hang shell...]
```

---
**qemuriscv32**
```text
qemuriscv32 login: root
root@qemuriscv32:~# uname -a
Linux qemuriscv32 5.8.4-yocto-standard #1 SMP PREEMPT Mon Aug 31 13:39:10 UTC 2020 riscv32 riscv32 riscv32 GNU/Linux
root@qemuriscv32:~# poweroff
...
[   48.564253] systemd-shutdown[1]: Syncing filesystems and block devices.
[   48.576562] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[   48.630689] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[   48.670501] systemd-shutdown[1]: Unmounting file systems.
[   48.683380] [197]: Remounting '/' read-only in with options '(null)'.
[   48.692327] EXT4-fs (vda): re-mounted. Opts: (null)
[   48.696857] systemd-shutdown[1]: All filesystems unmounted.
[   48.697103] systemd-shutdown[1]: Deactivating swaps.
[   48.698135] systemd-shutdown[1]: All swaps deactivated.
[   48.698368] systemd-shutdown[1]: Detaching loop devices.
[   48.702714] systemd-shutdown[1]: All loop devices detached.
[   48.702957] systemd-shutdown[1]: Detaching DM devices.
[   48.705575] systemd-shutdown[1]: All DM devices detached.
[   48.705840] systemd-shutdown[1]: All filesystems, swaps, loop devices and DM devices detached.
[   48.714813] systemd-shutdown[1]: Syncing filesystems and block devices.
[   48.720298] systemd-shutdown[1]: Powering off.
[   49.721033] reboot: Power down
sbi_trap_error: hart0: trap handler failed (error -2)
sbi_trap_error: hart0: mcause=0x00000007 mtval=0x00100000
sbi_trap_error: hart0: mepc=0x8000b4e4 mstatus=0x00001822
sbi_trap_error: hart0: ra=0x80007d4a sp=0x80012e34
sbi_trap_error: hart0: gp=0xc14cc9d0 tp=0xcf4a0000
sbi_trap_error: hart0: s0=0x80012e44 s1=0x00000020
sbi_trap_error: hart0: a0=0x00000000 a1=0x00000003
sbi_trap_error: hart0: a2=0x00000080 a3=0x00002e80
sbi_trap_error: hart0: a4=0x00100000 a5=0x00005555
sbi_trap_error: hart0: a6=0x00000000 a7=0x00000000
sbi_trap_error: hart0: s2=0x00000000 s3=0x80013000
sbi_trap_error: hart0: s4=0x00000000 s5=0x00000001
sbi_trap_error: hart0: s6=0x00000000 s7=0x00000000
sbi_trap_error: hart0: s8=0x6888afb8 s9=0x9c4e0e84
sbi_trap_error: hart0: s10=0x6888a8c8 s11=0x00000000
sbi_trap_error: hart0: t0=0x00000000 t1=0x00000000
sbi_trap_error: hart0: t2=0x00000000 t3=0x8000f014
sbi_trap_error: hart0: t4=0x00000000 t5=0x00000008
sbi_trap_error: hart0: t6=0x00000000
hang shell...
```

**Describe the results you received:**
poweroff qemuriscv32, qemuriscv64

**Describe the results you expected:**
**opensbi print dump log.**

pino-kim, 2020-08-31:
Reboot cmd make shell hang too.
Not reboot cmd work both qemuriscv64, qemuriscv32.

alistair23, 2020-08-31:
It's a QEMU bug, it has already been reported to the mailing list. Hopefully a patch will be available soon.

pino-kim, 2020-08-31:
@alistair23 
Thanks for letting me know where the problem is.
I was really curious why this issue pop out.
Shall we open this issue qemu bug fix or not?

alistair23, 2020-08-31:
We don't need to open an issue, it's already being discussed.

pino-kim, 2020-08-31:
ok got it

alistair23, 2020-08-31:
You can leave this issue opened though. That way we can track back porting the patch.

alistair23, 2021-06-24:
This has since been fixed in QEMU

closed 2021-06-24 by alistair23