riscv/meta-riscv #257: How to install fakeroot package for riscv64 yocto image?

Ankita-020696, 2020-11-27:
**Description**

**Steps to reproduce the issue:**
1.
2.
3.

**Describe the results you received:**

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

1

I want to build a yocto image for FreedomU540(riscv64) with fakeroot package. If I use following way of package installation **CORE_IMAGE_EXTRA_INSTALL += "git ruby bsdtar fakeroot"**

get the following error

**ERROR: Nothing RPROVIDES 'fakeroot' (but /home/openembedded-core/meta/recipes-core/images/core-image-minimal.bb RDEPENDS on or otherwise requires it)**

I tried installing using fakeroot deb package, it did not work.

Is there any other way to install fakeroot for riscv64?

Thanks

kraj, 2020-11-27:
@Ankita-020696 you can start with http://cgit.openembedded.org/meta-openembedded/tree/meta-oe/recipes-core/fakeroot/fakeroot_1.18.4.bb?h=morty but this is a bit outdated since upstream master branch for meta-openembedded has dropped this package, this is because OE has an equivalent package called pseudo which is uses. But give it a shot and see how far you can get. You will have to remove `PNBLACKLIST` to let the build start building it and then fix the errors

Ankita-020696, 2020-11-30:
> @Ankita-020696 you can start with http://cgit.openembedded.org/meta-openembedded/tree/meta-oe/recipes-core/fakeroot/fakeroot_1.18.4.bb?h=morty but this is a bit outdated since upstream master branch [...]

I tried with this recipe, get following errors:

**ERROR: fakeroot-1.18.4-r0 do_package_qa: QA Issue: /usr/bin/fakeroot contained in package fakeroot requires /bin/bash, but no providers found in RDEPENDS_fakeroot? [file-rdeps]
ERROR: fakeroot-1.18.4-r0 do_package_qa: QA Issue: -dev package contains non-symlink .so: fakeroot-dev path '/work/riscv64-oe-linux/fakeroot/1.18.4-r0/packages-split/fakeroot-dev/usr/lib/libfakeroot-0.so' [dev-elf]
ERROR: fakeroot-1.18.4-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: Logfile of failure stored in: /workspace/yocto/openembedded-core/build/tmp-glibc/work/riscv64-oe-linux/fakeroot/1.18.4-r0/temp/log.do_package_qa.14788
ERROR: Task (/workspace/yocto/openembedded-core/meta-riscv/recipes-core/fakeroot/fakeroot_1.18.4.bb:do_package_qa) failed with exit code '1'**

I am a newbie to yocto image building, could you please guide me to resolve these errors?

Thanks

kraj, 2020-11-30:
> I tried with this recipe, get following errors:
>
> **ERROR: fakeroot-1.18.4-r0 do_package_qa: QA Issue: /usr/bin/fakeroot contained in package fakeroot requires /bin/bash, but no providers found in [...]

right, no problem. To fix this you would add

`RDEPENDS_${PN} += "bash"`

to the fakeroot recipe file.  This will ensure that fakeroot gets the right indirect dependency on bash at runtime
since it seems to have shell scripts which are explicitly calling out for /bin/bash interpreter.
> Thanks

Ankita-020696, 2020-12-01:
> right, no problem. To fix this you would add
>
> `RDEPENDS_${PN} += "bash"`
>
> to the fakeroot recipe file. This will ensure that fakeroot gets the right indirect dependency on bash at runtime
> since it [...]

Hi,
I am still getting following errors:

**ERROR: fakeroot-1.18.4-r0 do_package_qa: QA Issue: -dev package contains non-symlink .so: fakeroot-dev path '/work/riscv64-oe-linux/fakeroot/1.18.4-r0/packages-split/fakeroot-dev/usr/lib/libfakeroot-0.so' [dev-elf]

ERROR: fakeroot-1.18.4-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.

ERROR: Logfile of failure stored in: /workspace/yocto/openembedded-core/build/tmp-glibc/work/riscv64-oe-linux/fakeroot/1.18.4-r0/temp/log.do_package_qa.9978

ERROR: Task (/workspace/yocto/openembedded-core/meta-riscv/recipes-core/fakeroot/fakeroot_1.18.4.bb:do_package_qa) failed with exit code '1'**

Is there anything that I should check?

Thanks

kraj, 2020-12-01:
> Hi,
> I am still getting following errors:
>
> **ERROR: fakeroot-1.18.4-r0 do_package_qa: QA Issue: -dev package contains non-symlink .so: fakeroot-dev path [...]

try adding
```
SOLIBS = ".so"
FILES_SOLIBSDEV = ""
```

Ankita-020696, 2020-12-02:
@kraj It worked. Thank you for your help.

kraj, 2020-12-02:
@Ankita-020696 glad it worked for you. I will close this issue, feel free to provide further feedback if any.

closed 2020-12-02 by kraj