riscv/meta-riscv #260: What's the password of default images

drcut, 2021-01-05:
**Description**

Hi, I just following the steps in README for using riscv64qemu, and now I can succeed build the Image, however, I do not find the password while executing 
```bash
 MACHINE=qemuriscv64 runqemu nographic
```
Would you please give me suggestions? Thanks

kraj, 2021-01-05:
make sure you have `debug-tweaks` in `IMAGE_FEATURES` and then it will have no password for `root` user just add it via conf/local.conf 

`IMAGE_FEATURES_append = " debug-tweaks"`

drcut, 2021-01-06:
> make sure you have `debug-tweaks` in `IMAGE_FEATURES` and then it will have no password for `root` user just add it via conf/local.conf
> 
> `IMAGE_FEATURES_append = " debug-tweaks"`

Thanks for your suggestion! Now I just though that password is not for QEMU image, but for host.
```bash
robinhan@rover:~/riscv-yocto$ MACHINE=qemuriscv64 runqemu nographic
runqemu - INFO - Running MACHINE=qemuriscv64 bitbake -e ...
runqemu - INFO - Continuing with the following parameters:
KERNEL: [/tmp/Yocto-glibc/deploy/images/qemuriscv64/Image--5.10.2+git0+6b12385f7d_0c61b00f3d-r0-qemuriscv64-20210106042421.bin]
BIOS: [/tmp/Yocto-glibc/deploy/images/qemuriscv64/fw_jump.elf]
MACHINE: [qemuriscv64]
FSTYPE: [ext4]
ROOTFS: [/tmp/Yocto-glibc/deploy/images/qemuriscv64/core-image-full-cmdline-qemuriscv64-20210106042421.rootfs.ext4]
CONFFILE: [/tmp/Yocto-glibc/deploy/images/qemuriscv64/core-image-full-cmdline-qemuriscv64-20210106042421.qemuboot.conf]

runqemu - INFO - Setting up tap interface under sudo
sudo: unable to execute /home/robinhan/riscv-yocto/openembedded-core/scripts/runqemu-ifup: Permission denied
runqemu - ERROR - Setting up tap device failed:
Command '('sudo', '/home/robinhan/riscv-yocto/openembedded-core/scripts/runqemu-ifup', '1388350517', '7001', '/tmp/Yocto-glibc/work/x86_64-linux/qemu-helper-native/1.0-r1/recipe-sysroot-native/usr/bin/')' returned non-zero exit status 1.
Run runqemu-gen-tapdevs to manually create one.
runqemu - INFO - Cleaning up
```

Seems weird why even using sudo to execute runqemu-ifup also raise Permission denied :(

alistair23, 2021-01-06:
When starting QEMU you can specify `slirip` to avoid settign up the TAP interface.

`$ runqemu slirp`

drcut, 2021-01-07:
> When starting QEMU you can specify `slirip` to avoid settign up the TAP interface.
> 
> `$ runqemu slirp`

Great! It works. But why it failed? I think it does use root to execute the runqemu-ifup script?

alistair23, 2021-01-07:
You need root to setup a TAP interface: https://wiki.qemu.org/Documentation/Networking#Tap

A TAP interface is more powerful and can be used to expose QEMU guests to public IPs. If you just need internet though slirp networking is more then enough, just remember that PING won't work.

closed 2021-01-08 by drcut