riscv/meta-riscv #261: Erros when building images

drcut, 2021-01-12:
**Description**

Excuse me. When I run ```MACHINE=qemuriscv32 bitbake core-image-full-cmdline```, it raises the error below:
```
Loading cache: 100% |                                                                                                                                                                       | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |######################################################################################################################################################################| Time: 0:01:07
Parsing of 2323 .bb files complete (0 cached, 2323 parsed). 3562 targets, 185 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.49.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "riscv32-oe-linux"
MACHINE              = "qemuriscv32"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv32"
meta                 = "work:cf10ad80a59bbeab1db0b1761bdf168d3e281474"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      = "work:759058bfb832161eb0d21f5c51a49c8361e727bc"
meta-riscv           = "work:9d10fd4f37d9f209847e79953fcc9bc018b1fe5e"

NOTE: Fetching uninative binary shim http://downloads.yoctoproject.org/releases/uninative/2.9/x86_64-nativesdk-libc.tar.xz;sha256sum=d07916b95c419c81541a19c8ef0ed8cbd78ae18437ff28a4c8a60ef40518e423 (will check PREMIRRORS first)
Initialising tasks: 100% |###################################################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 1192 Found 0 Missed 1192 Current 0 (0% match, 0% complete)
NOTE: Executing Tasks
WARNING: cracklib-native-2.9.5-r0 do_fetch: Failed to fetch URL https://downloads.sourceforge.net/cracklib/cracklib-2.9.5.tar.gz, attempting MIRRORS if available
WARNING: libmnl-1.0.4-r0 do_fetch: Failed to fetch URL https://netfilter.org/projects/libmnl/files/libmnl-1.0.4.tar.bz2;name=tar, attempting MIRRORS if available
WARNING: openssh-8.4p1-r0 do_fetch: Failed to fetch URL http://ftp.openbsd.org/pub/OpenBSD/OpenSSH/portable/openssh-8.4p1.tar.gz, attempting MIRRORS if available
WARNING: quota-4.06-r0 do_fetch: Failed to fetch URL https://downloads.sourceforge.net/project/linuxquota/quota-tools/4.06/quota-4.06.tar.gz, attempting MIRRORS if available
WARNING: rpcbind-1.2.5-r0 do_fetch: Failed to fetch URL https://downloads.sourceforge.net/rpcbind/rpcbind-1.2.5.tar.bz2, attempting MIRRORS if available
WARNING: libjpeg-turbo-native-1_2.0.6-r0 do_fetch: Failed to fetch URL https://downloads.sourceforge.net/libjpeg-turbo/libjpeg-turbo-2.0.6.tar.gz, attempting MIRRORS if available
WARNING: zip-3.0-r2 do_fetch: Failed to fetch URL https://downloads.sourceforge.net/infozip/Zip%203.x%20%28latest%29/3.0/zip30.tar.gz, attempting MIRRORS if available
WARNING: systemd-1_247.2-r0 do_fetch: Failed to fetch URL git://github.com/systemd/systemd-stable.git;protocol=git;branch=v247-stable, attempting MIRRORS if available
ERROR: core-image-full-cmdline-1.0-r0 do_rootfs: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_rootfs(d)
     0003:
File: '/home/SENSETIME/hanruobing/riscv-yocto/openembedded-core/meta/classes/image.bbclass', lineno: 245, function: do_rootfs
     0241:    progress_reporter.next_stage()
     0242:
     0243:    # generate rootfs
     0244:    d.setVarFlag('REPRODUCIBLE_TIMESTAMP_ROOTFS', 'export', '1')
 *** 0245:    create_rootfs(d, progress_reporter=progress_reporter, logcatcher=logcatcher)
     0246:
     0247:    progress_reporter.finish()
     0248:}
     0249:do_rootfs[dirs] = "${TOPDIR}"
File: '/home/SENSETIME/hanruobing/riscv-yocto/openembedded-core/meta/lib/oe/rootfs.py', lineno: 372, function: create_rootfs
     0368:
     0369:    img_type = d.getVar('IMAGE_PKGTYPE')
     0370:
     0371:    cls = get_class_for_type(img_type)
 *** 0372:    cls(d, manifest_dir, progress_reporter, logcatcher).create()
     0373:    os.environ.clear()
     0374:    os.environ.update(env_bkp)
     0375:
     0376:
File: '/home/SENSETIME/hanruobing/riscv-yocto/openembedded-core/meta/lib/oe/rootfs.py', lineno: 214, function: create
     0210:        execute_pre_post_process(self.d, rootfs_post_install_cmds)
     0211:
     0212:        self.pm.run_intercepts()
     0213:
 *** 0214:        execute_pre_post_process(self.d, post_process_cmds)
     0215:
     0216:        if self.progress_reporter:
     0217:            self.progress_reporter.next_stage()
     0218:
File: '/home/SENSETIME/hanruobing/riscv-yocto/openembedded-core/meta/lib/oe/utils.py', lineno: 263, function: execute_pre_post_process
     0259:    for cmd in cmds.strip().split(';'):
     0260:        cmd = cmd.strip()
     0261:        if cmd != '':
     0262:            bb.note("Executing %s ..." % cmd)
 *** 0263:            bb.build.exec_func(cmd, d)
     0264:
     0265:# For each item in items, call the function 'target' with item as the first
     0266:# argument, extraargs as the other arguments and handle any exceptions in the
     0267:# parent thread
File: '/home/SENSETIME/hanruobing/riscv-yocto/openembedded-core/bitbake/lib/bb/build.py', lineno: 256, function: exec_func
     0252:    with bb.utils.fileslocked(lockfiles):
     0253:        if ispython:
     0254:            exec_func_python(func, d, runfile, cwd=adir)
     0255:        else:
 *** 0256:            exec_func_shell(func, d, runfile, cwd=adir)
     0257:
     0258:    try:
     0259:        curcwd = os.getcwd()
     0260:    except:
File: '/home/SENSETIME/hanruobing/riscv-yocto/openembedded-core/bitbake/lib/bb/build.py', lineno: 503, function: exec_func_shell
     0499:    with open(fifopath, 'r+b', buffering=0) as fifo:
     0500:        try:
     0501:            bb.debug(2, "Executing shell function %s" % func)
     0502:            with open(os.devnull, 'r+') as stdin, logfile:
 *** 0503:                bb.process.run(cmd, shell=False, stdin=stdin, log=logfile, extrafiles=[(fifo,readfifo)])
     0504:        except bb.process.ExecutionError as exe:
     0505:            # Find the backtrace that the shell trap generated
     0506:            backtrace_marker_regex = re.compile(r"WARNING: Backtrace \(BB generated script\)")
     0507:            stdout_lines = (exe.stdout or "").split("\n")
File: '/home/SENSETIME/hanruobing/riscv-yocto/openembedded-core/bitbake/lib/bb/process.py', lineno: 184, function: run
     0180:        if not stderr is None:
     0181:            stderr = stderr.decode("utf-8")
     0182:
     0183:    if pipe.returncode != 0:
 *** 0184:        raise ExecutionError(cmd, pipe.returncode, stdout, stderr)
     0185:    return stdout, stderr
Exception: bb.process.ExecutionError: Execution of '/home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/temp/run.ssh_allow_empty_password.30210' failed with exit code 2:
sed: can't read 0: No such file or directory
WARNING: exit code 2 from a shell command.

ERROR: Logfile of failure stored in: /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/temp/log.do_rootfs.30210
ERROR: Task (/home/SENSETIME/hanruobing/riscv-yocto/openembedded-core/meta/recipes-extended/images/core-image-full-cmdline.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3701 tasks of which 1 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 110 seconds
NOTE: Build completion summary:
NOTE:   do_populate_sysroot: 0.0% sstate reuse(0 setscene, 295 scratch)
NOTE:   do_package_qa: 0.0% sstate reuse(0 setscene, 148 scratch)
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 148 scratch)
NOTE:   do_packagedata: 0.0% sstate reuse(0 setscene, 148 scratch)
NOTE:   do_package_write_ipk: 0.0% sstate reuse(0 setscene, 148 scratch)
NOTE:   do_populate_lic: 0.0% sstate reuse(0 setscene, 298 scratch)

Summary: 1 task failed:
  /home/SENSETIME/hanruobing/riscv-yocto/openembedded-core/meta/recipes-extended/images/core-image-full-cmdline.bb:do_rootfs
Summary: There were 8 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

```
Any suggestions? Thanks!

alistair23, 2021-01-12:
Maybe delete your `tmp` dir and then try again?

drcut, 2021-01-12:
Thanks for your suggestions. But after removing tmp, it raises another error
```
Loading cache: 100% |                                                                                                                                                                       | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |######################################################################################################################################################################| Time: 0:00:50
Parsing of 2323 .bb files complete (0 cached, 2323 parsed). 3562 targets, 245 skipped, 0 masked, 0 errors.
Removing 1 recipes from the allarch sysroot: 100% |##########################################################################################################################################| Time: 0:00:00
Removing 5 recipes from the qemuriscv32 sysroot: 100% |######################################################################################################################################| Time: 0:00:00
Removing 3 recipes from the riscv32 sysroot: 100% |##########################################################################################################################################| Time: 0:00:00
Removing 2 recipes from the x86_64 sysroot: 100% |###########################################################################################################################################| Time: 0:00:00
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.49.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "riscv32-oe-linux"
MACHINE              = "qemuriscv32"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv32"
meta                 = "work:cf10ad80a59bbeab1db0b1761bdf168d3e281474"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      = "work:759058bfb832161eb0d21f5c51a49c8361e727bc"
meta-riscv           = "work:9d10fd4f37d9f209847e79953fcc9bc018b1fe5e"

Initialising tasks: 100% |###################################################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 905 Found 0 Missed 905 Current 107 (0% match, 10% complete)
NOTE: Executing Tasks
ERROR: quilt-native-0.66-r0 do_configure: configure failed
ERROR: quilt-native-0.66-r0 do_configure: Execution of '/home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/work/x86_64-linux/quilt-native/0.66-r0/temp/run.do_configure.24446' failed with exit code 1:
configure: WARNING: unrecognized options: --disable-silent-rules, --disable-dependency-tracking
checking for a BSD-compatible install... /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/install -c
checking whether #! works in shell scripts... yes
checking for bash... (cached) /bin/bash
checking whether /bin/bash quoting works... yes
checking for gcp... no
checking for cp... /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/cp
checking for gdate... no
checking for date... /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/date
checking whether /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/date --rfc-822 works... yes
checking for perl... /usr/bin/env perl
configure: WARNING: /usr/bin/env perl does not exist
checking for grep... /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/grep
checking whether /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/grep -q works... yes
checking whether /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/grep understands (foo|bar)... yes
checking for tail... /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/tail
checking whether /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/tail -n works... yes
checking for tr... /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/tr
checking whether /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/tr understands a-z ... yes
checking for stat... /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/stat
checking whether /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/stat -c '%h' works... yes
checking for sed... /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/sed
checking whether /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/sed understands (foo|bar)... yes
checking for gawk... /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/gawk
checking whether /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/gawk supports sub... yes
checking for pod2man... no
checking for column... no
configure: WARNING: Using internal column mechanism.  Use option '--with-column' to override
checking for getopt... /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/getopt
checking for getopt --long syntax... yes
checking for mktemp... /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/mktemp
checking whether /home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/mktemp -d works... no
configure: error: 
'/home/SENSETIME/hanruobing/riscv-yocto/build/tmp-glibc/hosttools/mktemp -d' does not create temporary directories.
If you don't have a version of mktemp that can create directories, you
can specify '--without-mktemp' and quilt will use its own
[... cut ...]
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/SENSETIME/hanruobing/riscv-yocto/openembedded-core/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 490 tasks of which 457 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/SENSETIME/hanruobing/riscv-yocto/openembedded-core/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_configure
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

drcut, 2021-01-12:
> Maybe delete your `tmp` dir and then try again?

I can skip the error (the first error in this issue) by naively comment the following codes in openembedded-core/bitbake/lib/bb/process.py
```python
    #if pipe.returncode != 0:
    #    raise ExecutionError(cmd, pipe.returncode, stdout, stderr)
```
Of course this does not make sense.
I think it's some erro in ```riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/temp/run.ssh_allow_empty_password.30210```
The sed will check dropbear in etc, but the if condition only check the existings of dropbear in usr/sbin
```bash
	if [ -e /home/SENSETIME/hanruobing/tmp/riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/rootfs/usr/sbin/dropbear ] ; then
		if grep -q DROPBEAR_EXTRA_ARGS /home/SENSETIME/hanruobing/tmp/riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/rootfs/etc/default/dropbear 2>/dev/null ; then
			if ! grep -q "DROPBEAR_EXTRA_ARGS=.*-B" /home/SENSETIME/hanruobing/tmp/riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/rootfs/etc/default/dropbear ; then
				sed -i 's/^DROPBEAR_EXTRA_ARGS="*\([^"]*\)"*/DROPBEAR_EXTRA_ARGS="\1 -B"/' /home/SENSETIME/hanruobing/tmp/riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/rootfs/etc/default/dropbear
			fi
		else
			printf '\nDROPBEAR_EXTRA_ARGS="-B"\n' >> /home/SENSETIME/hanruobing/tmp/riscv-yocto/build/tmp-glibc/work/qemuriscv32-oe-linux/core-image-full-cmdline/1.0-r0/rootfs/etc/default/dropbear
		fi
	fi
```

alistair23, 2021-01-12:
Something looks really wrong. Maybe you should just remove temp, sstate and downloads.

drcut, 2021-01-13:
> Something looks really wrong. Maybe you should just remove temp, sstate and downloads.

Thanks, things go right when restart.

closed 2021-01-13 by drcut