riscv/meta-riscv #262: How to build gcc/g++ compiler in VM

drcut, 2021-01-13:
**Description**
I have succeed using your riscv32 VM. I found there isn't g++ or gcc within image. So would you please give some suggestions for how to build riscv-toolchain within it?
Thanks!

alistair23, 2021-01-13:
When you say VM I assume you mean inside the target rootFS?

Look at the `EXTRA_IMAGE_FEATURES` in the `local.conf`, it describes all the options you want.

kraj, 2021-01-13:
add following in `conf/local.conf`
```
EXTRA_IMAGE_FEATURES_append = " tools-debug dbg-pkgs tools-sdk dev-pkgs"
```
you can avoid adding `dbg-pkgs` if you don't want to debug info on target.

you might also want to add

```
IMAGE_ROOTFS_EXTRA_SPACE_append_qemuall = " + 3000000"
```

This will ensure that you have 3G of extra space added to image. In case you want to use an image out of box
to download stuff and install.

drcut, 2021-01-13:
> add following in `conf/local.conf`
>
> ```
> EXTRA_IMAGE_FEATURES_append = " tools-debug dbg-pkgs tools-sdk dev-pkgs"
> ```
>
> you can avoid adding `dbg-pkgs` if you don't want to debug info on target.
>
> you [...]

Great! It works!
Thank you so much!

closed 2021-01-13 by drcut

drcut, 2021-01-18:
Excuse me, do you have any experience of building LLVM in a riscv environments? Thanks

kraj, 2021-01-18:
@drcut add meta-clang layer along with meta-riscv or use yoe distro https://github.com/yoedistro/yoe-distro